ZerBea/hcxdumptool #101: hcxdumptools will not make- Kali 2020 OS not supported

andrewplotner, 2020-03-08:
Trying to Download hcxdumptool on kali linux 2020.
Kernel version is 5.4
i was able to successfully download hcxtools.

When I attempt to make 
i get the following output:
OS not supported
cc-03 -Wall -Wextra -std=gnu99 -o hcxpioff.c -DVERSION_TAG=\"6.0.1-46-g4d5cdf3\" -DVERSION_YEAR=\"2020\"
cc-03 -Wall -Wextra -std=gnu99 -o hcxdumptool hcxdumptool.c -lcrypto -DVERSION_TAG=\"6.0.1-46-g4d5cdf3\" -DVERSION_YEAR=\"2020\"

Any help on why this issue might be occuring would be helpful

strasharo, 2020-03-08:
What's the output of "uname -s" ?

ZerBea, 2020-03-08:
"make" retrieve information about the operating system and will only proceed if it detect Linux. We are doing this, because proprietary operating systems (MacOS and WINDOWS) doesn't support monitor mode and packet injection.
As @strasharo mentioned, uname -s will show you this information:
$ uname -s
Linux

exactly this output is tested by make:
ifeq ($(HOSTOS), Linux)

If uname -s return an empty string, you KALI installation is broken!
To compile it without os check, just remove this lines from the Makefile:
```
23 ifeq ($(HOSTOS), Linux)
24 INSTFLAGS += -D
25 endif

30 ifeq ($(HOSTOS), Linux)
34 endif

37 ifeq ($(HOSTOS), Linux)
40 else
41	$(info OS not supported)
42 endif
```

But I really suggest you to fix your installation!

ZerBea, 2020-03-12:
Some words about KALI:
KALI is a professional penetration testing platform. It is not easy to configure.  It is not designed to be used by beginners. Nearly all reported issues are related to missing dependencies, drivers that doesn't support monitor mode and full packet injection, broken installations and misconfigured systems. 
Before using KALI I suggest to read this:
https://kali.training/lessons/introduction/
https://kali.training/lessons/5-configuring-kali-linux/
https://www.tutorialspoint.com/kali_linux/kali_linux_installation_and_configuration.htm
https://www.kali.org/penetration-testing-with-kali-linux/

It is mandatory to know how to fix configure KALI, ti fix broken system and to solve KALI dependencies.

ZerBea, 2020-03-12:
Closed because issue is not related to hcxdumptool, but you can still ask your questions, here.

closed 2020-03-12 by ZerBea