ZerBea/hcxdumptool #102: --reactive_beacon

ukscone, 2020-03-08:
reactive_beacon is shown in the program help screen but is an invalid option when used (don''t see it in long_options[] )

ZerBea, 2020-03-08:
latest git hat shouldn't have --reactive_beacon. It is now together with --active_beacon

```
$ hcxdumptool -h
hcxdumptool 6.0.1-46-g4d5cdf3  (C) 2020 ZeroBeat
usage  : hcxdumptool <options>
         press the switch to terminate hcxdumptool
         hardware modification is necessary, read more:
         https://github.com/ZerBea/hcxdumptool/tree/master/docs
example: hcxdumptool -o output.pcapng -i wlp39s0f3u4u5 -t 5 --enable_status=3
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon)
         do not use hcxdumptool in combination with other 3rd party tools, which take access to the interface

short options:
-i <interface> : interface (monitor mode will be enabled by hcxdumptool)
                 can also be done manually:
                 ip link set <interface> down
                 iw dev <interface> set type monitor
                 ip link set <interface> up
                 WARNING: iw use NETLINK (libnl) and hcxdumptool will not work on pure NETLINK interfaces
-o <dump file> : output file in pcapng format
                 including radiotap header (LINKTYPE_IEEE802_11_RADIOTAP)
-f <frames     : frames to save
                 bitmask:
                  0: clear default values
                  1: MANAGEMENT frames (default)
                  2: EAP and EAPOL frames (default)
                  4: IPV4 frames
                  8: IPV6 frames
                 16: WEP encrypted frames
                 32: WPA encrypted frames
                 to clear default values use -f 0 first, followed by desired frame type (e.g. -f 0 -f 4)
-c <digit>     : set scan list (1,2,3, ...)
                 default scan list: 1...13
                 maximum entries: 127
                 allowed channels (depends on the device):
                 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14
                 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 62, 64, 68, 96
                 100, 102, 104, 106, 108, 110, 112, 114, 116, 118, 120, 122, 124, 126, 128
                 132, 134, 136, 138, 140, 142, 144, 149, 151, 153, 155, 157, 159
                 161, 165, 169, 173
-t <seconds>   : stay time on channel before hopping to the next channel
                 default 4 seconds
-m <interface> : set monitor mode by ioctl() system call and quit
-I             : show WLAN interfaces and quit
-C             : show available channels and quit
                 if no channels are available, interface is probably in use or doesn't support monitor mode

long options:
--do_rcascan                       : show radio channel assignment (scan for target access points)
                                     this can be used to test that ioctl() calls and packet injection is working
                                     if you got no HIT, packet injection is possible not working
                                     also it can be used to get information about the target
                                     and to determine that the target is in range
                                     use this mode to collect data for the filter list
                                     run this mode at least for 2 minutes
                                     to save all received raw packets use option -o
--reason_code=<digit>              : deauthentication reason code
                                      recommended codes:
                                      1 WLAN_REASON_UNSPECIFIED
                                      2 WLAN_REASON_PREV_AUTH_NOT_VALID
                                      4 WLAN_REASON_DISASSOC_DUE_TO_INACTIVITY
                                      5 WLAN_REASON_DISASSOC_AP_BUSY
                                      6 WLAN_REASON_CLASS2_FRAME_FROM_NONAUTH_STA
                                      7 WLAN_REASON_CLASS3_FRAME_FROM_NONASSOC_STA (default)
                                      9 WLAN_REASON_STA_REQ_ASSOC_WITHOUT_AUTH
--disable_client_attacks           : do not attack clients
                                     affected: ap-less (EAPOL 2/4 - M2) attack
--disable_ap_attacks               : do not attack access points
                                     affected: connected clients and client-less (PMKID) attack
[... 82 lines cut ...]
                                       16: ROGUE AP
                                       32: GPS (once a minute)
                                       64: internal status (once a minute)
                                      128: run as server
                                      256: run as client
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAP and EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
--server_port=<digit>              : define port for server status output (1...65535)
                                   : default IP: 224.0.0.255
                                   : default port: 60123
--client_port=<digit>              : define port for client status read (1...65535)
                                   : default IP: 224.0.0.255
                                   : default port: 60123
--check_driver                     : run several tests to determine that driver support all(!) required ioctl() system calls
--help                             : show this help
--version                          : show version

Run hcxdumptool -i interface --do_rcascan for at least 30 seconds, before you start an attack!
Do not edit, merge or convert this pcapng files, because it will remove optional comment fields!
It is much better to run gzip to compress the files. Wireshark, tshark and hcxpcapngtool will understand this.
If hcxdumptool captured your password from WiFi traffic, you should check all your devices immediately!
If you use GPS, make sure GPS device is inserted and has a GPS FIX, before you start hcxdumptool!

```

ZerBea, 2020-03-08:
The reactive BEACON made too much noise, so I had to remove it an favor of an active BEACON (using a fixed time interval).

ukscone, 2020-03-08:
ok thanks. pulling now

closed 2020-03-08 by ukscone

ZerBea, 2020-03-08:
I think it was a race condition between your pull and my push because latest update was very complex - hcxdumptool is complete refactored.