ZerBea/hcxdumptool #103: OS not supported Ubuntu 18.04

TanmayAgrawal, 2020-03-18:
I've made changes to hcxdumptool.c and hcxpioff.c as mentioned in readme. But still, it is showing OS not supported.
uname -s shows Linux.

ZerBea, 2020-03-18:
Please replace your Makefile with the attached one. Output should look like:
```
$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.0.2-1-ga80006d\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -DVERSION_TAG=\"6.0.2-1-ga80006d\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *
```
[Makefile.zip](https://github.com/ZerBea/hcxdumptool/files/4347657/Makefile.zip)

OS not supported means that make isn't able to detect your UBUNTU, because your environment is broken.

ZerBea, 2020-03-18:
I'm preparing a new commit. Please stay tuned.

ZerBea, 2020-03-18:
Please try latest commit:
https://github.com/ZerBea/hcxdumptool/commit/07d1d430e7b72fb606d5d7c6eb6e3d9e38b7b73c
Should be fixed, now.

TanmayAgrawal, 2020-03-18:
It works, thanks!

closed 2020-03-18 by TanmayAgrawal

ZerBea, 2020-03-18:
I have to thank you. That was an ugly "copy and paste error".