ZerBea/hcxdumptool #105: "Maximum number of errors reached" during rca_scan

thomasfsteeples, 2020-03-25:
When I run `hcxdumptool` with the `do_rcascan` option, it runs for anywhere between 15-45 seconds, before saying 

    maximum number of errors reached 

    terminating...

If I then go on and run the tool, it will work and capture PMKIDs, but it still acrrues errors (observed with `enable_status`).

Is there some way of seeing what the errors are? Any log files or anything like that? Is this a known issue?

I'm running Arch on a Dell XPS 13 with an Intel AX200 wireless card. Thank you.

ZerBea, 2020-03-25:
Running --do_rcascan the following errors are counted:
- failed to set channel (that will happen if another tool have access to the interface while hcxdumptool is running)
- failed to read packet (that will happen if another tool have access to the interface while hcxdumptool is running)
- failed to transmit PROBEREQUEST  (that will happen if the driver doesn't support full packet injection)

How did you set monitor mode?
Please post output of (latest git head):
$ sudo hcxdumptool -i interface -check_driver
$ sudo hcxdumptoll -m interface

thomasfsteeples, 2020-03-25:
I was just using `hcxdumptool -m interface` to put me in monitor mode. Running `sudo hcxdumptool -i interface --check_driver` gives me

    initialization...
    starting driver test...
    interface is already in monitor mode
    driver tests passed...
    all reuired ioctl() system calls are supported by driver

    terminating...

and `sudo hcxdumptoll -m interface` gives me

    interface is already in monitor mode
    setting interface interface to monitor mode
    
Cheers.

ZerBea, 2020-03-25:
"interface is already in monitor mode"
means that the interface is already in monitor mode.
Which tool have you used to set monitor mode?

ZerBea, 2020-03-25:
Please run hcxdumptool before you set monitor mode
$ hcxdumptool -i interface --check_driver

ZerBea, 2020-03-25:
Please post also output of
$ iw dev

ZerBea, 2020-03-25:
example:
$ iw dev
```
phy#0
	Interface wlp39s0f3u3u1u2
		ifindex 3
		wdev 0x1
		addr c8:3a:35:dc:9a:80
		type managed
		txpower 0.00 dBm
```
$ sudo hcxdumptool -i wlp39s0f3u3u1u2 --check_driver
```
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```
$ sudo hcxdumptool -m wlp39s0f3u3u1u2
`setting interface wlp39s0f3u3u1u2 to monitor mode`

$ iw dev
```
phy#0
	Interface wlp39s0f3u3u1u2
		ifindex 3
		wdev 0x1
		addr c8:3a:35:dc:9a:80
		type monitor
		channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
		txpower 20.00 dBm
```

thomasfsteeples, 2020-03-25:
I'm just using iw or hcxdumptool to put myself in monitor mode. Putting myself back in managed mode, I get:

    $ iw dev
    phy#0
            Interface wlp58s0
                    ifindex 4
                    wdev 0x6
                    addr 5c:80:b6:58:c3:94
                    type managed
                    txpower 0.00 dBm
                    multicast TXQ:
                            qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                            0       0       0       0       0       0       0       0               

    $ sudo hcxdumptool -i wlp58s0 --check_driver
    initialization...
    starting driver test...
    driver tests passed...
    all required ioctl() system calls are supported by driver

    terminating...

    $ sudo hcxdumptool -m wlp58s0
    setting interface wlp58s0 to monitor mode      

    $ iw dev
    phy#0
            Interface wlp58s0
                    ifindex 4
                    wdev 0x6
                    addr 5c:80:b6:58:c3:94
                    type monitor
                    channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
                    txpower 0.00 dBm
                    multicast TXQ:
                            qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                            0       0       0       0       0       0       0       0               0

ZerBea, 2020-03-25:
Thanks.
Now we know that we are able to set a channel.
next step is to add some printf()

here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L4342
printf("failed to set channel\n");
errorcount++;

and here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L4392
printf("fdnum error\n");
errorcount++;

and here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L4403
printf("failed to set channel\n");
errorcount++;

and here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L1683
printf("packet injection failed\n");
errorcount++;

Now compile and run --do_rcascan, again.

thomasfsteeples, 2020-03-25:
Ok, it's the channel setting which is giving me the errors. Both lines 4342 and 4403 are raising errors.

ZerBea, 2020-03-25:
I expect that the errors appear here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L1683
because there are some iwlwifi driver issues while injection packets.

ZerBea, 2020-03-25:
The status of do_rcascan will show you that packet injection is working:
 ```
BSSID         CH COUNT   HIT ESSID                 [18:05:24]
---------------------------------------------------------------
 000008ff47a0  11    29    25 test_ap_1
 000008ff47a1  11    27    24 test_ap_1
```
or not:
 ```
BSSID         CH COUNT   HIT ESSID                 [18:07:31]
---------------------------------------------------------------
 000008ff47a0  11    29    0 test_ap_1
 000008ff47a1  11    27    0 test_ap_1
```
if it isn't working, the hit column will show 0.

ZerBea, 2020-03-25:
Ok, please forget may last comments (race condition).
We are not able to set the channel.
Next step is to figure out, which channel causes the error.
Therefore please run
$ sudo hcxdumptool -i wlp58s0 -c 1 --do_rcascan
...
$ sudo hcxdumptool -i wlp58s0 -c 11 --do_rcascan
...
$ sudo hcxdumptool -i wlp58s0 -c 12 --do_rcascan
...
$ sudo hcxdumptool -i wlp58s0 -c 13 --do_rcascan

Maybe your card isn't able to set channel 12 and/or 13).
Is crda installed?

ZerBea, 2020-03-25:
As you can see here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L208
channel 12 and 13 are part of the scanlist.
To show the supported channels, just run
$ sudo hcxdumptool -i wlp58s0 -C
```
initialization...
interface is already in monitor mode
available channels:
  1 / 2412MHz (20 dBm)
  2 / 2417MHz (20 dBm)
  3 / 2422MHz (20 dBm)
  4 / 2427MHz (20 dBm)
  5 / 2432MHz (20 dBm)
  6 / 2437MHz (20 dBm)
  7 / 2442MHz (20 dBm)
  8 / 2447MHz (20 dBm)
  9 / 2452MHz (20 dBm)
 10 / 2457MHz (20 dBm)
 11 / 2462MHz (20 dBm)
 12 / 2467MHz (20 dBm)
 13 / 2472MHz (20 dBm)
 14 / 2484MHz (20 dBm)
```

ZerBea, 2020-03-25:
example:
My interface isn't able to set channel 50 (only 1..14 are supported). If I try to use this channel, ERROR count will increase:
$ sudo hcxdumptool -i wlp39s0f3u3u1u2 -c 1,3,5,7,50 --enable_status=64
```
18:22:00   1 ERROR:1 INCOMING:240 OUTGOING:80 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:23:00   1 ERROR:4 INCOMING:1179 OUTGOING:378 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
```

ZerBea, 2020-03-25:
If you can confirm this, I'll add a function to verify that crda and/or interface support channel 12/13. If not, we remove them from the scanlist before we start.

thomasfsteeples, 2020-03-25:
So these are the channels available to me - I have 1 to 13 (along with some 5Ghz channels), but not 14?

    initialization...
    interface is already in monitor mode
    available channels:
      1 / 2412MHz (22 dBm)
      2 / 2417MHz (22 dBm)
      3 / 2422MHz (22 dBm)
      4 / 2427MHz (22 dBm)
      5 / 2432MHz (22 dBm)
      6 / 2437MHz (22 dBm)
      7 / 2442MHz (22 dBm)
      8 / 2447MHz (22 dBm)
      9 / 2452MHz (22 dBm)
     10 / 2457MHz (22 dBm)
     11 / 2462MHz (22 dBm)
     12 / 2467MHz (22 dBm)
     13 / 2472MHz (22 dBm)
     40 / 5200MHz (22 dBm)
     44 / 5220MHz (22 dBm)
     48 / 5240MHz (22 dBm)
     52 / 5260MHz (22 dBm)
     56 / 5280MHz (22 dBm)
     60 / 5300MHz (22 dBm)
     64 / 5320MHz (22 dBm)  
    116 / 5580MHz (22 dBm)
    120 / 5600MHz (22 dBm)
    124 / 5620MHz (22 dBm)
    128 / 5640MHz (22 dBm)
    132 / 5660MHz (22 dBm)
    136 / 5680MHz (22 dBm)
    140 / 5700MHz (22 dBm)
    144 / 5720MHz (22 dBm)
    149 / 5745MHz (22 dBm)
    153 / 5765MHz (22 dBm)
    157 / 5785MHz (22 dBm)
    161 / 5805MHz (22 dBm)
    165 / 5825MHz (22 dBm)

ZerBea, 2020-03-25:
Ok the interface support channel 12 and 13.
What happens if you try to do a scan on channel 13:
$ sudo hcxdumptool -i wlp58s0 -c 13 --do_rcascan
Do we receive an ERROR ?

ZerBea, 2020-03-25:
Are there services that take access to the interface (iwd)?
https://wiki.archlinux.org/index.php/Iwd

thomasfsteeples, 2020-03-25:
With `sudo hcxdumptool -i wlp58s0 -c 13 --do_rcascan` I get zeros in the hit column. With `iw reg get`, I get the following output:

    global
    country 00: DFS-UNSET
            (2402 - 2472 @ 40), (6, 20), (N/A)
            (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
            (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
            (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
            (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
            (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
            (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
            (57240 - 63720 @ 2160), (N/A, 0), (N/A)

    phy#0 (self-managed)
    country 00: DFS-UNSET
            (2402 - 2437 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40MINUS, NO-80MHZ, NO-160MHZ
            (2422 - 2462 @ 40), (6, 22), (N/A), AUTO-BW, NO-80MHZ, NO-160MHZ
            (2447 - 2482 @ 40), (6, 22), (N/A), AUTO-BW, NO-HT40PLUS, NO-80MHZ, NO-160MHZ
            (5170 - 5190 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
            (5190 - 5210 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
            (5210 - 5230 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, PASSIVE-SCAN
            (5230 - 5250 @ 160), (6, 22), (N/A), NO-OUTDOOR, AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, PASSIVE-SCAN
            (5250 - 5270 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5270 - 5290 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5290 - 5310 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5310 - 5330 @ 160), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5490 - 5510 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5510 - 5530 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5530 - 5550 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5550 - 5570 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5570 - 5590 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5590 - 5610 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5610 - 5630 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, PASSIVE-SCAN
            (5630 - 5650 @ 240), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, PASSIVE-SCAN
            (5650 - 5670 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
            (5670 - 5690 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
            (5690 - 5710 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
            (5710 - 5730 @ 80), (6, 22), (0 ms), DFS, AUTO-BW, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
            (5735 - 5755 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
            (5755 - 5775 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
            (5775 - 5795 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-160MHZ, PASSIVE-SCAN
            (5795 - 5815 @ 80), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40PLUS, NO-160MHZ, PASSIVE-SCAN
            (5815 - 5835 @ 20), (6, 22), (N/A), AUTO-BW, IR-CONCURRENT, NO-HT40MINUS, NO-HT40PLUS, NO-80MHZ, NO-160MHZ, PASSIVE-SCAN

ZerBea, 2020-03-25:
Ok, we're coming closer.
Zero hits in column hits means that full packet injection isn't working by the iwlwifi driver. 
I looks like we have only a limited sniffer mode:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi#about_the_monitorsniffer_mode
https://forums.intel.com/s/question/0D50P0000490QYA/does-intel-80211ac-1x1-wifi-card-support-packet-injectioni?language=en_US
https://forum.aircrack-ng.org/index.php?topic=629.0

BTW:
Have you aireplay-ng installed? If yes, do you receive proberesponses running:
$ sudo aireplay-ng -9 interface

ZerBea, 2020-03-26:
It looks like some drivers / firmwares doesn't support full monitor mode. Instead they provide only limited sniffer capabilities.
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/7d6db577c4915a9d30d565579a7b9c7861e40706
I added a new option (--check_injection) to determine that full packet injection is supported by driver and firmware.
```
$ sudo hcxdumptool -i wlp39s0f3u3u1u2 --check_injection
initialization...
starting packet injection test (that will take up to one minute)...
packet injection is working!

terminating...
```
Please test and post result.

thomasfsteeples, 2020-03-26:
Ok, running the above, I get:
    initialization...
    interface is already in monitor mode
    starting packet injection test (that can take up to two minutes)...
    packet injection is working!

    terminating...

ZerBea, 2020-03-26:
Great. Thanks.  Looks like this is working fine, too.
At this moment, we are very restrictive if an error occurs:
#define ERROR_MAX		100
I'll add a new option to make this user defined.

ZerBea, 2020-03-26:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/c45472856e6b9b1b5a52c52d79884a788aeec0f5
we are less restrictive on errors.
```
--error_max=<digit>                : terminate hcxdumptool if error maximum reached
                                     default: 100 errors
```

ZerBea, 2020-03-26:
For some reasons, the iwlwifi driver sometimes isn't able to set the channel - maybe we are too fast: 
ioctl[SIOCSIWFREQ]: Device or resource busy
hcxdumptool count this as an ERROR and if we rech ERROR_MAX, it will terminate.
Please play around with the new option to determine the best value for your hardware.

ZerBea, 2020-03-26:
Are you running iwd (intel wireless daemon) or wicd (networkmanager)?
If not, do we receive an ERROR message in dmesg log like this ones:
https://forums.intel.com/s/question/0D70P0000068dA5SAI/command-failed-device-or-resource-busy-16?language=de
https://forums.archlinux.fr/viewtopic.php?style=9&t=14834
https://www.spinics.net/lists/linux-wireless/msg140539.html
https://archlinux.org.ru/forum/topic/13393/?page=1#post-134788
https://forums.gentoo.org/viewtopic-t-956640-start-0.html
https://forums.linuxmint.com/viewtopic.php?t=175331&start=20
http://forums.debian.net/viewtopic.php?f=7&t=122091&start=15

Maybe we can force this dmesg message running:
$ iw wlp58s0 scan

ZerBea, 2020-03-26:
Looks like we are running into a kernel issue, reported (as of  2020-03-25 19:05) here:
"This script brings up the wireless interface in monitor mode using the `iw` utility, then switches channels over and over again. No data capture is necessary to reproduce the crash. Simply toggling the channels is enough to expose what seems to be some sort of DMA bug."
https://bugzilla.kernel.org/show_bug.cgi?id=206961
hcxdumptool does exactly the same, but much faster than the script.
So the best workaround (until we receive a kernel fix) is to increase ERROR_MAX to prevent that hcxdumptool terminate. If the driver become ready again, hcxdumptool will set the new channel.

ZerBea, 2020-03-26:
...but anyway, your issue report forced me to add some nice new options to hcxdumptool.
Unfortunately there is nothing I can do within hcxdumptool code.

thomasfsteeples, 2020-03-26:
Thank you so much for all the time and effort you've put into this - it's very much appreciated! I'll follow up the kernel issue.

closed 2020-03-26 by thomasfsteeples

ZerBea, 2020-03-26:
No problem, you're welcome.
Unfortunately we have some more really nasty kernel issues at the moment:
https://bugzilla.kernel.org/show_bug.cgi?id=202541

ZerBea, 2020-03-30:
It looks like we are running into another driver issue:
https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.6-Broken-Intel-IWLWIFI