ZerBea/hcxdumptool #106: (Unknown BlockType:Zero Packet??)

MohamedSaif44, 2020-04-02:
first i shall say thank you for all your dedication and your consistent effort.
now,after i followed the instruction you mentioned in the guidelines(first_steps),also used and passed every test like:
--check_driver              
--check_injection:also verified by other tool:airplay-ng  and some other tool i can't remember
in in the end i got these results if it can even be called so:
used this:sudo hcxdumptool --silent  -i wlx503eaae72d58 -o dump.pcapng   --enable_status=1 

1.5 hour later:
[image]
[image]

OS:Ubuntu 18.04
device:tp-link tl-wn722n v3.0
driver:rtl8188eu

ZerBea, 2020-04-02:
You're dumping to dump.pcapng (hcxdumptool), but you're trying to read dump-pcapng-0 (hcxpcaptool).
hcxdumptool will store captured packets in the file, selected by option "-o". If this file exists, hcxdumptool doesn't overwrite it. Instead it append a sequential number starting with -0
Please comment output of
$ hcxpcapngtool dump-pcapng

Please attach dump.pcapng-0

BTW:
hcxpcaptool is marked as deprecated. Please use hcxpcapngtool.

ZerBea, 2020-04-02:
Are you running latest git head of hcxtools?
Please comment also output of
$ hcxpcaptool -v

The last version printing that error message "unknown blocktype x after packet x" was 5.1.6!

ZerBea, 2020-04-02:
Just released hcxtools v6.0.0.
hcxpcaptool <= v5.3.0 is deprecated and will not work in combination with hcxdumptool >= v6.0.0
Please reopen issue, if hcxpcapngtool v6.0.0 is not working on your dumpfile.

closed 2020-04-02 by ZerBea