ZerBea/hcxdumptool #107: Stupid Question. What do you actually mean by RogueXXXXX

ukscone, 2020-04-17:
Silly question but i'm a bit confuzzled as to what you mean by roguepmkid & rogueeapol. i'm assuming that they are faulty in some way so not actually usable for cracking. is this correct.

I did have a quick look at the source but I couldn't quite get my head around it

ZerBea, 2020-04-17:
No, it isn't a silly question. I changed wording:
M1M2ROGUE  (ROGUEEAPOL) = AP-LESS
PMKIDROGUE = CLIENT-LESS

Old definitions are incorrect, because in case of AP-LESS, hcxdumptool is the (ROGUE) AP
and in case of CLIENT-LESS, hcxdumptool is the (ROGUE) CLIENT.
In other words:
M1M2ROGUE (m2 from CLIENT) and PMKIDROGUE (PMKID from AP) are requested by hcxdumptool, retrieved directly from the target  and 100% valid (no packet loss).

ukscone, 2020-04-17:
Ah. Thank you. I had been misunderstanding the wording and assumed rogue == bad :) and thought I was getting fewer pmkid's since 6.x.x. 

i'm not i'm just getting pmkid's in a "different" way :)

closed 2020-04-17 by ukscone