ZerBea/hcxdumptool #109: Cant get it to work

Mangle3447, 2020-04-30:
Hi, I've done everything you said to do and then got to this part:
__________________________________________________________________________________________________________
root@kali:~# hcxdumptool -o hash -i wlan0mon --filterlist_ap=bruh.txt --filtermode=2 --enable_status=3
initialization...
warning: wlan0mon is probably a virtual monitor interface
failed to backup current interface flags, ioctl(SIOCGIFFLAGS) not supported by driver: No such device
warning: failed to init socket
try to use iw to set monitor mode
try to use ip link to bring interface up

terminating...
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device
__________________________________________________________________________________________________________

In Bruh.txt I have a bunch of BSSID's which I need to find the hash for and which I have detected.
I am in monitor mode here;
__________________________________________________________________________________________________________

root@kali:~# sudo iw dev wlan0 info
Interface wlan0
        ifindex 7
        wdev 0x200000001
        addr 02:00:00:00:00:00
        type monitor
        wiphy 2
        channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
        txpower 20.00 dBm
__________________________________________________________________________________________________________

I done the "airmon-ng check kill" or whatever command before going into monitor mode.
Please help I can't find what to do anywhere :(

ZerBea, 2020-04-30:
First of all:
Your driver must support all ioctl() system calls, otherwise hcxdumptool will not work. Unfortunately, it looks like your driver doesn't support them: ioctl(SIOCGIFFLAGS) not supported by driver
BTW:
The following drivers don't support ioctl() system calls
Broadcom (that include NEXMON on Raspberry Pi systems)
Realtek (8188 and 8812)

Second of all:
airmon-ng is designed to work with aircrack-ng suite and not with hcxdumptool! So, do not use airmon-ng in combination with hcxdumptool.
This is described in --help:
```
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon)
         do not use hcxdumptool in combination with 3rd party tools, which take access to the interface (except: tshark, wireshark, tcpdump)
-i <interface> : interface (monitor mode will be enabled by hcxdumptool)
```
Please notice:
Monitor mode will be enabled by hcxdumptool. You don't need a third party tool (airmon-ng) to do this!

Now please do the following steps (and keep in mind: do not use airmon-ng):
stop this services:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service

and post output of:
if you use an USB WiFi adapter:
$ lsusb
if you use the internal PCIe WiFi card:
$ lspci
$ sudo hcxdumptool -I
$ sudo hcxdumptool -i your_interface --check_driver
$ sudo hcxdumptool -i your_interface --check_injection

BTW:
Closed this, because it isn't an hcxdumptool issue, but you can still ask your questions here.

closed 2020-04-30 by ZerBea

Mangle3447, 2020-05-02:
Thanks for the fast reply. So do you know how i'd be able to get the correct drivers onto my computer in order to run it? (I'm on a Kali Linux virtual machine through Virtual Box).

ZerBea, 2020-05-02:
First let us see which driver is installed. Please comment output of
$ sudo hcxdumptool -I

Mangle3447, 2020-05-03:
it says 
020000000000 wlan0 (mac80211_hwsim)
020000000100 wlan1 (mac80211_hwsim)

Mangle3447, 2020-05-03:
which comes up under wlan interfaces, which I have to reinstall each time I open up the VM for some reason

ZerBea, 2020-05-04:
mac80211_hwsim is a Linux kernel module that can be used to simulate arbitrary number of IEEE 802.11 radios for mac80211. It isn't a real interface and you can't use it to run penetration tests against real networks. 

Read more here:
https://wireless.wiki.kernel.org/en/users/drivers/mac80211_hwsim

BTW:
I don't use neither a Virtual Machine (VM) nor KALI Linux, so I can't help you here. 
But in every case, running a VM isn't the best choice if you need direct access to hardware (hcxdumptool).

Mangle3447, 2020-05-04:
Oh, okay. Thank you so much for your help anyway.

shitheadexpress, 2020-09-14:
hey, running a VM and getting the same error, which I know you mention doesnt help.

Can you comment on whether this will run hcxdumptools:
Alfa AWUSO36ACH
00c0caaaf2d5 rt 88xxxau
using rt8812au-5.6.4.2.zip driver

thanks in advance!

ZerBea, 2020-09-14:
rt8812au will not work out of the box.
You can try the aircrack-ng driver and set monitor mode by hand, but you should know that there are several issues:
https://github.com/aircrack-ng/rtl8812au/issues
especially this one here:
https://github.com/aircrack-ng/rtl8812au/issues/587
```
-i <interface> : interface (monitor mode will be enabled by hcxdumptool)
                 some Realtek interfaces require NETLINK to set monitor mode
                 in this case try iw:
                 ip link set <interface> down
                 iw dev <interface> set type monitor
                 ip link set <interface> up
                 WARNING:
                  hcxdumptool may not work as expected on virtual NETLINK interfaces
                  do not report issues related to iw
                 It is mandatory that chipset and driver support monitor mode and full packet injection!
                 Running a virtual machine, it is mandatory that the hardware is looped through!
```

The second question about "the best Linux" isn't easy to answer, because there is no "best Linux".
I prefer Arch Linux, because:
- fully configurable
- no bloatware
- no unnecessary tools
- no unnecessary services
- an awesome WiKi

shitheadexpress, 2020-09-15:
Hey, thanks for the message and link.
Hmm, it might be better if I change hardware for ease of use.
If I understand correctly, driver 5.2.2.0 is best for rtl8812au and monitor mode + PI?
Have you tried Edimax EW-7722UTN V2 N300?

ZerBea, 2020-09-15:
> If I understand correctly, driver 5.2.2.0 is best for rtl8812au and monitor mode + PI?

None of the rtl88xxxx drivers is the best unless new rtw88-usb stack will support monitor mode and full packet injection and without NETLINK dependency.

We don't need a fast 802.11 adapter, because hcxdumptool slows the authentication down.
I can't recommend an adapter that doesn't support native monitor mode and full packet injection.
I can't recommend an adapter that require an additional driver. That means a driver, not supplied with the stock kernel:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/wireless?h=v5.8
I can't recommend a high power device. We are packet oriented. So it is enough when a single packet comming form hcxdumptool was received by the target and a single packet from the target was received by hcxdumptool.
I can't recommend high power adapters. Instead I use common (adapters with external antenna connector) in combination with an (expensive) antenna (mostly panels or stocked verticals).

MediaTek /Ralink chipsets are fine (supplied with the stock kernel) and support monitor mode and full packet injection. The driver code is well maintained

shitheadexpress, 2020-09-15:
Ok thanks I will keep researching, I appreciate the time. Cheers.

shitheadexpress, 2020-09-21:
Hey Zerbea, I swapped my HW to the AWUS036NH - works out of box with Kali 2020.3 
💯 
If anyone is reading, and are a newbie, use the above. The hardest part was realising the adaptor wasnt plugged in correctly for the first 10 minutes :-|

ZerBea, 2020-09-22:
Glad to hear this. The most important part is to find and an adapter which is supported by a stock kernel driver (which include full monitor mode and full packet injection). In addition to that, the driver must be well maintained.  Unfortunately not every drive will do this.
A good start point is here:
https://bugzilla.kernel.org/
https://patchwork.kernel.org/project/linux-wireless/list/
and here:
https://github.com/openwrt/mt76/issues/

Next step is to configure your system. That include that you know what services are running and how to step them, if they interfere with the adapter (e.g. NetworkManager).

Also you should have some about radio technology and 802.11 protocol.

Now you can use hcxdumptool/hcxtools. Please notice that this tools are analysis tools and they have a lot of options.