ZerBea/hcxdumptool #110: PMKID

Martha-cloud, 2020-05-08:
$ ./hcxdumptool -o test.pcapng -i wlp39s0f3u4u5 –enable_status

ZerBea, 2020-05-08:
What exactly is your issue? Commenting a single command line is not enough to help you.
Please notice:
- your driver must support monitor mode and full packet injection running AF_PACKET sockets and ioctl() system calls (not every driver will do this - especially rtl8812au, rtl8188eus, NEXMON on Broadcom drivers will not work)
- not every ACCESS POINT transmit a PMKID
- running hcxdumptool within a Virtual Machine (VM) is not recommended as long as the hard ware access to the interface is not looped through

Please comment output of:
$ hcxdumptool -I
$ hcxdumptool --check-driver
$ hcxdumptool --check-injection
additonal
$ uname -r
$ lsusb
is very helpful.

BTW:
Your command line is incomplete. The value of the status bitmap is missing here: --enable_status
examples:
$ hcxdumptool -o test.pcapng -i wlp39s0f3u4u5 --enable_status=1
will show you incoming EAP and EAPOL frames. This include PMKIDs, too.

$ hcxdumptool -o test.pcapng -i wlp39s0f3u4u5 --enable_status=9
will show you incoming BEACON, PROBERESPONSE, EAP and EAPOL frames. This include PMKIDs, too.

From hcxdumptool --help:
```
--enable_status=<digit>            : enable real-time display (waterfall)
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
                                     bitmask:
                                        0: no status (default)
                                        1: EAP and EAPOL
                                        2: ASSOCIATION and REASSOCIATION
                                        4: AUTHENTICATION
                                        8: BEACON and PROBERESPONSE
                                       16: ROGUE AP
                                       32: GPS (once a minute)
                                       64: internal status (once a minute)
                                      128: run as server
                                      256: run as client
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAP and EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```

ISimion, 2020-08-19:
Hi, so I am trying to capture PKMID's from Local Network by typing the following commands:

```
root@kali:/home/user/Desktop# ip link set wlan1 down
root@kali:/home/user/Desktop# iw dev wlan1 set type monitor
root@kali:/home/user/Desktop# ip link set wlan1 up
root@kali:/home/user/Desktop# iw dev wlan1 info
Interface wlan1
        ifindex 52
        wdev 0x2a00000001
        addr 9c:ef:d5:fd:20:f6
        type monitor
        wiphy 42
        channel 11 (2462 MHz), width: 20 MHz (no HT), center1: 2462 MHz
        txpower 20.00 dBm
root@kali:/home/user/Desktop# iw phy `iw dev wlan1 info | gawk '/wiphy/ {printf "phy" $2}'` interface add wlan1mon type monitor
root@kali:/home/user/Desktop# hcxdumptool -i wlan1mon -o galleria.pcapng --enable_status=1
initialization...
warning: wlan1mon is probably a virtual monitor interface
interface is already in monitor mode
warning: interface is not up

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1mon
INTERFACE HARDWARE MAC....: 9cefd5fd20f6
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.7.0-kali1-amd64
DRIVER FIRMWARE VERSION...: N/A
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 000dc2452750 (BROADCAST HIDDEN)
ACCESS POINT (ROGUE)......: 000dc2452751 (BROADCAST OPEN)
ACCESS POINT (ROGUE)......: 000dc2452752 (incremented on every new client)
CLIENT (ROGUE)............: c0225010f128
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 61829
ANONCE....................: 8ba2486b0f3806120226f7c0522a2720b2fe4112c552e7eb0f18cf185268782b
SNONCE....................: fb4dfd17da821e20a9adbe90f9c6b632427a6a50c9cb656a9107d82242f808ab

failed to read packet: Network is down
^C
terminating...
root@kali:/home/user/Desktop# 
```

As you can see it warns me that the interface is not up, although I turned it up. Could anyone suggest what I am doing wrong or how can I get it worked?

PS: I am using Panda Wireless PAU09 N600 Network Adapter.

ZerBea, 2020-08-19:
Your workflow doesn't work:
you set monitor mode on your interface wlan1
you bring the interface wlan1  up
now you create a virtual monitor interface wlan1mon but you don't bring it up.
at this point wlan1 (is up) and wlan1mon (is down) sharing the same hardware.
```
root@kali:/home/user/Desktop# iw phy `iw dev wlan1 info | gawk '/wiphy/ {printf "phy" $2}'` interface add wlan1mon type monitor 
```
All your commands doesn't make sense.

From --help:
```
usage  : hcxdumptool <options>
         press the switch to terminate hcxdumptool
         hardware modification is necessary, read more:
         https://github.com/ZerBea/hcxdumptool/tree/master/docs
example: hcxdumptool -o output.pcapng -i wlp39s0f3u4u5 -t 5 --enable_status=3
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon)
         do not use hcxdumptool in combination with 3rd party tools, which take access to the interface (except: tshark, wireshark, tcpdump)
```
hcxdumptool is running an own monitor mode and you received several warning because you didn't followed the instructions.

step 1:
stop all services that take access to the interface e.g:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service
step 2:
find your interface name:
$ hcxdumptool -I
output should look like this:
```
wlan interfaces:
7844767d6e93 wlp5s0f4u2 (rt2800usb)
```
step 3:
start hcxdumptool:
$ sudo hcxdumptool -i wlp5s0f4u2 -o test.pcapng --enable_status=31

Please notice:
Not all APs are transmitting a PMKID!

ZerBea, 2020-08-20:
If you follow the instructions, output should look like this:
```
$ iw dev
phy#2
	Interface wlp5s0f4u2
		ifindex 5
		wdev 0x200000001
		addr 42:d3:7e:8a:8e:2c
		type managed
		txpower 30.00 dBm

$ hcxdumptool -I
wlan interfaces:
7844767d6e93 wlp5s0f4u2 (rt2800usb)

$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service

$ sudo hcxdumptool -i wlp5s0f4u2 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$ sudo hcxdumptool -i wlp5s0f4u2 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 247 to 89 

terminating...

$ sudo hcxdumptool -i wlp5s0f4u2 -o test.pcapng --enable_status=31
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp5s0f4u2
INTERFACE HARDWARE MAC....: 7844767d6e93
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.8.1-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 000b1879b756 (BROADCAST HIDDEN)
ACCESS POINT (ROGUE)......: 000b1879b757 (BROADCAST OPEN)
ACCESS POINT (ROGUE)......: 000b1879b758 (incremented on every new client)
CLIENT (ROGUE)............: b0ece1bdc7bc
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62620
ANONCE....................: ec33449ef6381078dce816bee06a43e54a3b446539316ef21cb21ea8093ff97c
SNONCE....................: 630babd65ef550a9f1e642f697c0e6b0198edabb1dc8a9cadb32f07e134ca444

09:23:25   1 ffffffffffff 787140acf315 TEST_NETWORK [BEACON]
```
You may have noticed that, we didn't create a monitor interface (by iw- hcxdumptool doesn't use NETLINK stuff) and allow to share.
Also you may have noticed, that hcxdumptool didn't show a warning.

I strongly recommend to forget all (web) tutorials that suggest that you have to create a monitor interface. That may apply for other tools, but not for hcxdumptool!

I closed this issue, because it is not an hcxdumptool issue, but you can still ask your questions here.

closed 2020-08-20 by ZerBea