ZerBea/hcxdumptool #111: --filterlist_ap doesn't seem to be working

ukscone, 2020-05-12:
I tried to use --filterlist_ap but am getting an error 

root@kali:~/hcxdumptool# ./hcxdumptool -i wlan1 -o may12thfl.cap --filterlist_ap=fl.txt --filtermode=2                                  
initialization...                                                                                                                       
failed to read filter list line 0:                                                                           
failed to read filter list line 1:                                                                                     
failed to read filter list line 2:                                                                                         
failed to read filter list line 3:                                                                                      
failed to read filter list line 4:                                                                                          
failed to read filter list line 5:    
.
.
.                                                                                   
failed to read filter list line 13:  

[i've removed the actual macs but they came from --do_rcascan

ukscone, 2020-05-12:
oh never mind. it was because there were tons of spurious space characters all over the place. got rid of them and it seems to be ok

closed 2020-05-12 by ukscone

ZerBea, 2020-05-12:
Ok.
Format is: 112233445566 comment
additional you can comment a line (hcxdumptool will skip it): #112233445566 comment

Running protection mode, I recommend BPF code instead of fiilterlist_ap.