ZerBea/hcxdumptool #112: can you explain/give definitions for a few terms please

ukscone, 2020-05-12:
previously you have corrected my understanding of pmkidrogue & m1m2rogue but could you give the meanings of a few other things please. i've looked through the source and think I know what some mean but would like to know for certain and TIL: zeiger is German for pointer :)

KDV == ?
RC == ?
M3M4ZEROED == ? [assume it means that it's been zero'd out so no use for aircrack-ng/hashcat use

thanks

ZerBea, 2020-05-12:
No problem:
KDV = Key Descriptor Version (1= WPA1, 2 = WPA2, 3 = WPA2 key version 3)
according to Wireshark Key Information Field:
```
Key Information: 0x0109
    .... .... .... .001 = Key Descriptor Version: RC4 Cipher, HMAC-MD5 MIC (1)
    .... .... .... 1... = Key Type: Pairwise Key
    .... .... ..00 .... = Key Index: 0
    .... .... .0.. .... = Install: Not set
    .... .... 0... .... = Key ACK: Not set
    .... ...1 .... .... = Key MIC: Set
    .... ..0. .... .... = Secure: Not set
    .... .0.. .... .... = Error: Not set
    .... 0... .... .... = Request: Not set
    ...0 .... .... .... = Encrypted Key Data: Not set
    ..0. .... .... .... = SMK Message: Not set
```
RC = ReplayCount

M3M4ZEROED = Message Pair M3 (ACCESS POINT) and M4 (CLIENT) - M4 SNONCE is zeroed and can not be used to recover the PSK.

Cheers
Mike

ukscone, 2020-05-13:
thanks a lot

closed 2020-05-13 by ukscone

ZerBea, 2020-05-14:
Regarding the terminology I decided to use the same wording as Wireshark, hashcat and JtR. This makes it easier to understand.