ZerBea/hcxdumptool #113: WPA-SEC missing frames

h0wdyy, 2020-05-17:
Hey there,

I've gotten a .cap file using the following command "hcxdumptool -o capture3.cap -i wlan1 --enable_status=1 --filterlist_ap=/root/myfilter.txt --filtermode=2" and it uploads just fine. No matter what I've tried I always get the same error message saying:
 "Warning: missing frames!
This dump file contains no important frames like
authentication, association or reassociation.
That makes it hard to recover the PSK."

My network adapter is an alfa AWUS036ACH and my drivers are working fine with the newest kali for pi. Monitor mode works like a charm. Getting these .cap files aren't a problem I just don't understand what I need to do to get them up to the wpa-sec standard.

ZerBea, 2020-05-17:
That is a normal behavior. If you run filtermode 2, all frames that doesn't contain the MACs, specified in filterlist_ap are filtered out. That include some important frames.
From --help menu:
```
--filtermode=<digit>               : mode for filter list
                                     mandatory in combination with --filterlist_ap and/or --filterlist_client
                                     0: ignore filter list (default)
                                     1: use filter list as protection list
                                        do not interact with ACCESS POINTs and CLIENTs from this list
                                     2: use filter list as target list
                                        only interact with ACCESS POINTs and CLIENTs from this list
                                        not recommended, because important pre-authentication frames will be lost due to MAC randomization of the CLIENTs
```
Can you please attach the cap file (zip compressed)?

Cheers
Mike

ZerBea, 2020-05-17:
BTW:
wpa-sec is running hcxpcapngtool. So you can verify your cap/pcap/pcapng file, running:
$ hcxpcapngtool *.pcapng
Please make sure, you're on latest git head of hcxdumptool/hcxtools.

h0wdyy, 2020-05-17:
Should I upload it as a .pcapng on wpa-sec? I tried to upload a .pcap and that didn't work. I see about the filter 2. How would you go about filtering to target a single AP without losing important frames? 
[capture3.zip](https://github.com/ZerBea/hcxdumptool/files/4639988/capture3.zip)

ZerBea, 2020-05-17:
Thanks for the cap file. I'll take a closer look inside.
hcxpcapngtool accept cap, pcap, pcapng and gzip (.gz) compressed files. So everything is fine.

ZerBea, 2020-05-17:
Ok, got it:
```
timestamp minimum (GMT)..................: 17.05.2020 10:39:26
timestamp maximum (GMT)..................: 17.05.2020 10:39:47
```
The capture time was too small. 21 seconds are not enough to gather complete information. I recommend to run hcxdumptool at least for 5 minutes (ore more) to retrieve all frames from AP and CLIENTs.

For example:
```
timestamp minimum (GMT)..................: 08.04.2020 21:32:26
timestamp maximum (GMT)..................: 08.04.2020 21:41:27
```

ZerBea, 2020-05-17:
But anyway the quality of your cap file is good. I was able to recover the PSK in 3:03 minutes. 
wpa-sec will have it tomorrow (after running through all "Get Works").

```
Session..........: hashcat
Status...........: Cracked
Hash.Name........: WPA-PBKDF2-PMKID+EAPOL
Hash.Target......: test.22000
Time.Started.....: Sun May 17 11:57:43 2020 (3 mins, 1 sec)
Time.Estimated...: Sun May 17 12:00:44 2020 (0 secs)
Guess.Base.......: Pipe
Speed.#1.........:   120.1 kH/s (9.20ms) @ Accel:8 Loops:64 Thr:1024 Vec:1
Recovered........: 1/1 (100.00%) Digests
Progress.........: 21726285
Rejected.........: 1101
Restore.Point....: 0
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:0-1
Candidates.#1....: 
Hardware.Mon.#1..: Temp: 71c Fan: 49% Util:100% Core:1265MHz Mem:3004MHz Bus:16
Started: Sun May 17 11:57:42 2020
Stopped: Sun May 17 12:00:45 2020
```

ZerBea, 2020-05-17:
hcxdumptool default format is pcapng (the same is used by Wireshark, tshark).
Difference of the formats:
cap = 802.11eee frames
pcap = 802.11eee frames + radiotap header
pcapng = cap + pcap + additional comments
I recommend to use hcxdumptool -o capture.pcapng

h0wdyy, 2020-05-17:
I just ran hcxdumptool a longer amount of time and got the same error.
timestamp minimum (GMT)..................: 17.05.2020 09:55:35
timestamp maximum (GMT)..................: 17.05.2020 10:09:20

However, I used the .cap filesystem again. I'll try this once again using the .pcapng. Are you certain given the extension and time using filtermode=2 will still be an issue?

ZerBea, 2020-05-17:
It isn't an error. You are just notified that some kind of frames are not captured during the capture time.
That can happen if you're running filtermode=2. It also can happen if the AP or the CLIENT is far away.

ZerBea, 2020-05-17:
Please notice:
The AP and CLIENT running default allowed tx power. Your ALFA is running much more tx power. So it is possible that AP and/or CLIENT receive the packets from your ALFA, but you don't receive all packets from them.

h0wdyy, 2020-05-17:
I just captured it in the .pcapng file format and all of the errors are completely gone when reading the file through hcxpcapngtool, however when I try to upload it on the wpa-sec website it says: "File type not allowed
Allowed extension: cap"

ZerBea, 2020-05-17:
Extension should match to the file format.
I uploaded your capture3 using cap, pcap and pcapng and everything is fine:
```
$ wlancap2wpasec capture3.cap
connecting to https://wpa-sec.stanev.org
uploading capture3.cap to https://wpa-sec.stanev.org
This capture file was already submitted.
upload done

1 cap uploaded to https://wpa-sec.stanev.org

$ wlancap2wpasec capture3.pcap
connecting to https://wpa-sec.stanev.org
uploading capture3.pcap to https://wpa-sec.stanev.org
This capture file was already submitted.
upload done

1 cap uploaded to https://wpa-sec.stanev.org

$ wlancap2wpasec capture3.pcapng
connecting to https://wpa-sec.stanev.org
uploading capture3.pcapng to https://wpa-sec.stanev.org
This capture file was already submitted.
upload done

1 cap uploaded to https://wpa-sec.stanev.org

```

h0wdyy, 2020-05-17:
I was trying to use the browser method to upload which wasn't working but using this in terminal works well. Will this automatically link to my key when uploading this way?

ZerBea, 2020-05-17:
```
wlancap2wpasec:
options:
-k <key>           : wpa-sec user key
```

BTW:
I notifyd @RealEnder about the web interface issue.

h0wdyy, 2020-05-17:
Thanks so much for your help Mike! I'm sure the web interface and this information will be helpful to others

closed 2020-05-17 by h0wdyy

ZerBea, 2020-05-17:
Your issue report reminded me to improve hcxdumptool in that case if we have a high tx power device on the attack side. Here we must request some frames from AP and CLIENT more often.

ZerBea, 2020-05-17:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/476c66d56fd217fc837b51cb7aa5a9ce47d333de
I added an information that some useful frames are filtered out, when running filtermode = 2
That behavior is wanted to protect CLIENTs which are not in filterlist_client.
Workaround to retrieve all frames running filtermode 2:
add target AP(s) to filterlist_ap
add target CLIENT(s) to filterlist_client

ZerBea, 2020-05-17:
I added this information to hcxpcapngtool, too:
Warning: missing frames!
...
It always happens if the capture file was cleaned or
if filter options are used during capturing.
...

h0wdyy, 2020-05-17:
It's doesn't always happen when filter options are being used as long as you give it enough time and save it as a .pcapng by the way. I've used filter types 1 and 2 and got absolutely no missing frames. You might want to say filter options COULD impede capturing purity.

ZerBea, 2020-05-17:
Good point. Thanks. I'll change it to "could".

ZerBea, 2020-05-17:
@RealEnder would be nice to update to latest git head of hcxpcapngtool:
https://github.com/ZerBea/hcxtools/blob/master/hcxpcapngtool.c#L615

RealEnder, 2020-05-17:
Just updated hcxpcapngtool to latest git HEAD. As for the error with the extension, it's just some old check I forgot to fix. This should be OK now, thanks for the heads-up.