ZerBea/hcxdumptool #114: Not Getting any output

dreamhunter9m, 2020-05-23:
root@error-code:~# hcxdumptool -i wlan0mon -o naim.pcapng --enable_status=1
initialization...
warning: NetworkManager is running with pid 1007
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1116
 (possible interfering hcxdumptool)
warning: wlan0mon is probably a virtual monitor interface
interface is already in monitor mode

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0mon
INTERFACE HARDWARE MAC....: 30e37a26637b
DRIVER....................: iwlwifi
DRIVER VERSION............: 4.17.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 29.541020.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 000f096b6b31 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 000f09006b32 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 000f096b6b33 (incremented on every new client)
ROGUE (CLIENT)............: b0ece1adfb55
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 63412
ANONCE....................: c966adc2f989e49200dd3e442f9c145f733f0bdda643a67089f40e23548e77c4
SNONCE....................: 3be1c53c25bebf64b3cd3f5a9bd1385dc5228184e68c5526cdc7b8bc3fe76d6b

>>>>>>>My Pmkid Attack Showing this but not getting any Found Pmkid, Plz Help...!!!<<<<<<<<

ZerBea, 2020-05-23:
You have received a bunch of warnings:
```
warning: NetworkManager is running with pid 1007
(possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1116
(possible interfering hcxdumptool)
warning: wlan0mon is probably a virtual monitor interface
interface is already in monitor mode
```
To avoid them stop NetworkManager and wpa_supplicant:
$ systemctl stop NetworkManager.service
$ systemctl stop wpa_supplicant.service

Looks like you use airmon-ng to set monitor mode:
`warning: wlan0mon is probably a virtual monitor interface
`
Do not use airmon-ng to set monitor mode.

Additional, it looks like your kernel
4.17.0-kali1-amd64
Kali 2018.3 – 27th August, 2018 – The third 2018 Kali Rolling release. Kernel 4.17.0, GNOME 3.28.2
is outdated.

Also it looks like you are using an INTEL WiFi chipset.
Not all of them support monitor mode and full packet injection.
Example:
https://forums.kali.org/showthread.php?19248-Wifi-packet-injection-fails-on-Intel-6205-wifi-card
To test monitor mode is available and all ioctl() system calls are working, run:
$ hcxumptool -i your_interface --check_driver
and comment output

To test packet injection is working run:
$ hcxumptool -i your_interface --check_injection
and comment output

Please add also output of:
$ uname -r
and 
$ lsusb
and
$ lspci

dreamhunter9m, 2020-05-23:
root@error-code:~# ip link set wlan0 down
root@error-code:~# iw dev wlan0 set type monitor
root@error-code:~# ip link set wlan0 up
root@error-code:~# iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x6
		addr 30:e3:7a:26:63:7b
		type P2P-device
		txpower 0.00 dBm
	Interface wlan0
		ifindex 5
		wdev 0x5
		addr 26:f6:9b:f0:0a:69
		type monitor
		txpower 0.00 dBm
root@error-code:~# hcxdumptool -i wlan0 -o naimu.pcapng --enable_status=1
initialization...
warning: NetworkManager is running with pid 3136
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 3258
 (possible interfering hcxdumptool)
interface is already in monitor mode

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: 30e37a26637c
DRIVER....................: iwlwifi
DRIVER VERSION............: 4.17.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 29.541020.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 00234af14d2b (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 00234a004d2c (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 00234af14d2d (incremented on every new client)
ROGUE (CLIENT)............: f04f7c269cc0
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 63124
ANONCE....................: 441cdc4d93c734fab8d8481b002705f2dc7a0595f9dba05ce1f9783c894da5d7
SNONCE....................: 1bb075dab2c3c2ad8b214392f4d4c1c90fbac798dc2fd6d8352a94f8e30ebf03

^C
terminating...
root@error-code:~# systemctl stop NetworkManager.service
root@error-code:~# systemctl stop wpa_supplicant.service
root@error-code:~# hcxdumptool -i wlan0 -o naimu.pcapng --enable_status=1
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: 30e37a26637c
DRIVER....................: iwlwifi
DRIVER VERSION............: 4.17.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 29.541020.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 3cb87a533efe (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 3cb87a003eff (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 3cb87a533f00 (incremented on every new client)
ROGUE (CLIENT)............: e0cb1d24d548
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 63356
ANONCE....................: cae1031b5041163ff2b829edad1e5d3e65837494f2dc763fa5692ff35ebb0a63
SNONCE....................: ecda54acd41820360a0037a486de78289f689ec4d6091c6978cfd7dc91bb7a70

^C
terminating...
root@error-code:~# hcxdumptool -i wlan0 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
root@error-code:~# hcxdumptool -i wlan0 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
warning: no PROBERESPONSE received - packet injection is probably not working!

>>>>>>>Tried This but it showing packet injection is probably not working! ,what can be it's solutions, i'm also adding output below  you wanna see<<<<<<<

root@error-code:~# uname -r
4.17.0-kali1-amd64
root@error-code:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0aa7 Intel Corp. 
Bus 001 Device 004: ID 125f:db8a A-DATA Technology Co., Ltd. 
Bus 001 Device 003: ID 04f2:b56d Chicony Electronics Co., Ltd 
Bus 001 Device 002: ID 09da:2403 A4Tech Co., Ltd. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@error-code:~# lspci
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
00:04.0 Signal processing controller: Intel Corporation Skylake Processor Thermal Subsystem (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1d.3 PCI bridge: Intel Corporation Device 9d1b (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point-LP LPC Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 940MX] (rev a2)
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS522A PCI Express Card Reader (rev 01)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8101/2/6E PCI Express Fast/Gigabit Ethernet controller (rev 0a)
05:00.0 Network controller: Intel Corporation Dual Band Wireless-AC 3168NGW [Stone Peak] (rev 10)

ZerBea, 2020-05-23:
Thanks for the output. It looks like the build in AC 3168NGW doesn't support full packet injection. I saw similar issue reports here:
https://forum.aircrack-ng.org/index.php?topic=1651.0
and here:
https://forum.aircrack-ng.org/index.php?topic=1704.0
and here:
https://forum.aircrack-ng.org/index.php?topic=2024.0
and here:
"Monitor mode and packet injection are not supported with Intel® Wireless Adapters. Neither the adapters nor the drivers are designed to perform these functions."
https://forums.intel.com/s/question/0D50P0000490QYASA2/does-intel-80211ac-1x1-wifi-card-support-packet-injectioni?language=en_US
and here:
You can try to update KALI to latest release, because your kernel (4.17.0-kali1-amd64) is outdated:
https://www.kali.org/releases/kali-linux-2020-1a-release/
Latest LTS kernel is 5.4.42:
https://www.kernel.org/
But I don't think it will work, because there are too many issues with this chipset:
https://bugzilla.kernel.org/show_bug.cgi?id=206329
https://bbs.archlinux.org/viewtopic.php?id=252603
https://forums.gentoo.org/viewtopic-t-1107558-start-0.html
https://forum.manjaro.org/t/intel-ac-3168ngw-wifi-not-working-after-suspension/125564
https://forums.kali.org/showthread.php?45023-iwlwifi-broken-Kernel-5-2
https://forums.intel.com/s/question/0D50P0000490WHiSAM/intel-dual-band-wirelessac-3168ngw-ridiculously-slow-in-linux?language=de
https://forum.manjaro.org/t/intel-3168ngw-wifi-issues-on-recent-kernels/125199
and much more....

I don't see any chance that hcxdumptool will work with the Wireless-AC 3168NGW.

After reading about this issues, I decided to add a notice to README.md:
```
Not recommended WiFi chipsets:
Intel PRO/Wireless
Broadcom
```

dreamhunter9m, 2020-05-24:
Thanks a lot for this info. But i tried to crack pmkid with this hardware a few years ago in 2018. But surprisingly it worked. i Don't know why it isn’t working  i think i have to install Latest version of Kali linux 2020.2

ZerBea, 2020-05-24:
In every case, an attack is limited to the capabilities of the chipset and the driver. Both must support monitor mode and full packet injection. Additional all ioctl() system calls must be supported. 

Monitor mode (from https://en.wikipedia.org/wiki/Monitor_mode) :
Monitor mode, or RFMON (Radio Frequency MONitor) mode, allows a computer with a wireless network interface controller (WNIC) to monitor all traffic received on a wireless channel. Unlike promiscuous mode, which is also used for packet sniffing, monitor mode allows packets to be captured without having to associate with an access point or ad hoc network first.
If the chipset/driver support monitor mode, it will allow us to receive(!) packets.

Packet injection (from https://en.wikipedia.org/wiki/Monitor_mode) : 
Usually the wireless adapter is unable to transmit in monitor mode and is restricted to a single wireless channel, though this is dependent on the wireless adapter's driver, its firmware, and features of its chipset.
That means: not every chipset/driver that is able to receive packets, will be able to transmit packets.

Control:
Switching channels is done by ioctl() system calls. The driver must allow this, too. If not, we are not able to move to another channel.

To perform fast attacks, we use a RAW packet socket to control the interface, to receive packets and to transmit packets. A (virtual) NETLINK interface will not work.
"NETLINK comms is very much asynchronous, with each side receiving messages at some point after the other side sends them. ioctls are purely synchronous: “Hey kernel, WAKE UP! I need you to process my request NOW! CHOP CHOP!”"
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls

This three points are mandatory to let hcxdumptool work as expected.

Since 2018 many things changed. hcxdumptool and hcxtools are refactored several times. They have been adapted to new kernels, hardware and drivers, as soon as they are developed.
My distribution is Arch. Arch Linux is a rolling-release system that include latest kernel, latest firmware, latest drivers and latest software (for example gcc 10.1). From this point of view, a 2018 distribution is very, very old.
I give you and example:
Here, we discovered an driver issue (iwlwifi driver)
https://github.com/ZerBea/hcxdumptool/issues/105
An issue report was made here: 
https://bugzilla.kernel.org/show_bug.cgi?id=206961
and it received a kernel fix. 

Your old distribution will never get a kernel fix!

BTW:
How about receiving packets:
$ hcxdumptool -i wlan0 --enable_status=31
Do you receive anything, running this command?

dreamhunter9m, 2020-05-24:
>>>>>>>>>>>>>>>>>>Here it is

root@error-code:~# hcxdumptool -i wlan0 --enable_status=31
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: 30e37a26637b
DRIVER....................: iwlwifi
DRIVER VERSION............: 4.17.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 29.541020.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 906f18289071 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 906f18009072 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 906f18289073 (incremented on every new client)
ROGUE (CLIENT)............: f04f7ce1e06e
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 61813
ANONCE....................: d921d14ea5050a102455cf9556d1c81d5cb5bef707fb6856c43f455be76c6cd3
SNONCE....................: 1e1200ca8217ea4539331a461251bea33fad05657d118b8c2b15d41bacb08a48

17:42:36   6 ffffffffffff 04c80765b177 Redmi [BEACON]
17:42:44   3 ffffffffffff 0c8063da67c6 Shilpi [BEACON]
17:42:44   3 ffffffffffff 007263616786 oroni [BEACON]
17:43:04   2 ffffffffffff 7c8bca7ef088 SOBNOM [BEACON]
17:43:28   9 00082297fe9c 906f18289073 taste fast [ROGUE PROBERESPONSE]
17:44:44   2 ea8d1a40c863 0c8063da67c6 Shilpi [PROBERESPONSE]

ZerBea, 2020-05-24:
Thanks. Looking not so bad. I would say that 50 percent are working. You should be able to capture handshakes and PMKIDs even without having full packet injection capabilities.
Is dmesg showing a driver related warning?
$ dmesg

ZerBea, 2020-07-07:
Closed, because the issue is related to the driver. Work on the iwlwifi driver is in progress:
https://bugzilla.kernel.org/buglist.cgi?quicksearch=iwlwifi

closed 2020-07-07 by ZerBea