ZerBea/hcxdumptool #115: Is it (still) too noisy?

uniformsierra, 2020-06-10:
Thank you for this useful tool. 

I am trying to test capture M1/M2 from clients which i don't know their MAC address (ap-less) but i do know their SSID (I am using an essidlist file).
I do not want to interact with any other AP/client other than the clients listed in the essidlist file.

I am noticing (by using airodump on another PC) that there are many more beacons being sent from hcxdumptool, that are not in the essidlist. I'm guessing this is due to the --active-beacon switch, but if I don't use that hcxdumptool does not broadcast my essidlist...

I have tested with different wifi cards and architectures (x86 and ARM) with hcxdumptool versions from 5.x to 6.0.6.
I have tried lots of command switch combinations, but most commonly;
hxcdumptool -i wlan0 -c 9 --disable_ap_attacks --active-beacon --essidlist=essidlist.txt -o capture.pcap

Is there a way to make hcxdumptool broadcast ONLY beacons from the essidlist file and nothing else? I would like to be able to broacast every 1 second for reliable testing.

ZerBea, 2020-06-10:
Running active-beacon hcxdumptool will add every received ESSID its internal beacon list. I the user select --essidlist=xxx, this beacons will added to the list. The beacons from Beacons from this list are transmitted in a special time interval. This interval depend on the traffic on the channel and not on a fixed time (e.g. 100 ms = default beacon intervall).
Let me think about a switch to transmit only beacons from the essidlist.

ZerBea, 2020-06-10:
Ok, that may work:
on option  --essidlist= we transmit beacons from this list only
on option --active-beacon we transmit beacons from all received ESSIDs
on option  --essidlist=  and --active-beacon we transmit beacons from both

I'll check how to code it....