ZerBea/hcxdumptool #117: Probably out of scope but would be a nice addition

ukscone, 2020-06-19:
This is almost certain out of scope for the program and would add a bit of bloat but it would be nice if we could "test" for several weak keys. maybe from the command line by allowing a small list (3 to 5) separated by commas or from a file with 1 per line say upto 256 (too many?)

ZerBea, 2020-06-19:
A good idea, but impossible to realize. I added this feature last year and had to remove it:
https://github.com/ZerBea/hcxdumptool/commit/2bf1604b576943f94985862a8726b48c31493d25
We have only one try between two EAPOL frames. If we we'll try more, we will loose EAPOL frames.

ukscone, 2020-06-19:
ok. didn't think it'd be that doable but would have been nice and we have hashcat and aircrack-ng etc. to get the passwords anyway even if personally I have nothing that runs hashcat or rather nothing that runs hashcat well.

EAPOL frames are more important than weak keys

ZerBea, 2020-06-19:
Correct. We need every CPU cycle to perform the attack vector.

adrastee, 2020-08-18:
Add a crontab job that will convert your *.pcang files into hashes, or that will send them to websites (I only know 2 websites that have APIs for WPA). Then you can feed john/hashcat with 22000 hashes and/or wait for the websites results.

ZerBea, 2020-08-18:
Additional you can do this via a simple bash script.
Run hcxdumptool and if it terminates, start hcxpcapngtool to convert the dumpfile.
hcxdumptool has a switch for that (--tot).
Pleas notice, bash scripting and cron jobs are not part of hcxdumptool and hcxtools. If you need this features, you have to code them by your own.

ukscone, 2020-08-19:
> Add a crontab job that will convert your *.pcang files into hashes, or that will send them to websites (I only know 2 websites that have APIs for WPA). Then you can feed john/hashcat with 22000 hashes and/or wait for the websites results.

i already do that. well i don't do the cronjob thing as no point the way i use hcxdumptool. i just meant for things where i know a partial pw for 1 or more ap just to do a quick check which they are using e.g. i know an ap in my area has nearly always used babypanda+a digit but they change the digit randomly every few months so being able to quickly run through them just using hcxdumptool would have been nice rather than capture, convert or upload and then wait or run another program to see which it is. also lots of transient ap use 012345678,123456789,987654321,... so the ability to just check a small number of possibilties while hcxdumptool was running would have been nice but i agree more frames are better than a few weak password checks

closed 2020-08-30 by ukscone