ZerBea/hcxdumptool #119: channel switch (-c) not working

Czechball, 2020-06-29:
# Shortly
When I run hcxdumptool with the -c parameter followed by a single number or more numbers separated by commas (x,x,x), hcxdumptool still works on all available channels
## More info
Exact commands used:
```
hcxdumptool -o outputfile -c 8,9,10,11,12,13,14 --enable_status=9 -i interface_name --filtermode=1 --filterlist_ap=whitelist.txt
```
```
hcxdumptool -o outputfile -c 1 --enable_status=9 -i interface_name --filtermode=1 --filterlist_ap=whitelist.txt
```
Result: hcxdumptool found networks from all 2,4 GHz channels.

hcxdumptool --check_driver output was `all required ioctl() system calls are supported by this driver`
hcxdumptool -C output was all 14 channels.
## System info
Debian 10 i686, kernel 4.19.0-9-686-pae
Wireless interface used is a TP-Link TL-WN722N with the Atheros AR9271 chipset (ath9k_htc)
hcxdumptool version 6.0.7-1-g486d40e

ZerBea, 2020-06-30:
I can't reproduce that. Running  hcxdumptool -c 1

test device:  TP-Link TL-WN722N

$ lsusb
`Bus 001 Device 007: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n
`

$ dmesg
```
[   44.265898] ath9k_htc 1-3:1.0: ath9k_htc: FW Version: 1.4
[   44.265907] ath9k_htc 1-3:1.0: FW RMW support: On
[   44.265911] ath: EEPROM regdomain: 0x809c
[   44.265914] ath: EEPROM indicates we should expect a country code
[   44.265917] ath: doing EEPROM country->regdmn map search
[   44.265920] ath: country maps to regdmn code: 0x52
[   44.265924] ath: Country alpha2 being used: CN
[   44.265926] ath: Regpair used: 0x52
[   44.273906] ieee80211 phy1: Atheros AR9271 Rev:1
[   44.412526] ath9k_htc 1-3:1.0 wlp0s29f7u3: renamed from wlan0
```

hcxdumptool stays on channel 1:
```
$ sudo hcxdumptool -o test.pcapng --enable_status=9 -c 1 -i wlp0s29f7u3
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp0s29f7u3
INTERFACE HARDWARE MAC....: f81a67077d0e
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.4.39-1.0-lts
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 8ce74897d303 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 8ce74800d304 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 8ce74897d305 (incremented on every new client)
ROGUE (CLIENT)............: acde483c9319
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62756
ANONCE....................: b38697b784fd87a28f22db803beb83b088fe90197d00b874c117e20939052b73
SNONCE....................: d148d979476ba0ee9fcfcb6a21e6b6b63720b6151706a2c7969e979d74bcb5b3

08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
08:23:44   1 ...
```

And the Radiotap Header of every captured packet confirm this (Channel frequency: 2412 [BG 1]):
```
Radiotap Header v0, Length 36
    Header revision: 0
    Header pad: 0
    Header length: 36
    Present flags
    MAC timestamp: 108752905
    Flags: 0x10
    Data Rate: 1,0 Mb/s
    Channel frequency: 2412 [BG 1]
    Channel flags: 0x00a0, Complementary Code Keying (CCK), 2 GHz spectrum
    Antenna signal: -80dBm
    RX flags: 0x0000
    Antenna signal: -81dBm
    Antenna: 0
```

But, you should know that a CLIENT from a network, located on channel e.g. 12, will scan all possible channels, too. If he reached hcxdumptool on channel 1, he will connect to us, even though if it is not his AP channel. He will also connect to us, if his AP channel is within 5GHz! That is described in README.md:
```
* hcxdumptool is able to capture handshakes from 5GHz clients on 2.4GHz (only one single M2 from the client is required)
  (use hcxpcapngtool to save them to file)
```

Please attach an example pcapng file (only a few packets) running hcxdumptool -c 1, that we can take a look at the Radiotap Header.

ZerBea, 2020-06-30:
Finished some more tests running hcxdumptool -c 1 and -c 1,6,11 and verified the pcapng file running tshark:

```
$ tshark -r test.pcapng -T fields -E header=y -e frame.number -e frame.time -e wlan_radio.channel 
frame.number	frame.time	wlan_radio.channel
1	Jun 30, 2020 10:15:51.379575000 CEST	1
2	Jun 30, 2020 10:15:51.379716000 CEST	1
3	Jun 30, 2020 10:15:51.379810000 CEST	1
4	Jun 30, 2020 10:15:51.379877000 CEST	1
5	Jun 30, 2020 10:15:51.379979000 CEST	1
6	Jun 30, 2020 10:15:51.380955000 CEST	1
7	Jun 30, 2020 10:15:51.390094000 CEST	1
8	Jun 30, 2020 10:15:51.392464000 CEST	1
9	Jun 30, 2020 10:15:51.394134000 CEST	1
10	Jun 30, 2020 10:15:51.398374000 CEST	1
...
600	Jun 30, 2020 10:15:58.765442000 CEST	1
601	Jun 30, 2020 10:15:58.766958000 CEST	1
602	Jun 30, 2020 10:15:58.771304000 CEST	1
603	Jun 30, 2020 10:15:58.773699000 CEST	1
604	Jun 30, 2020 10:15:58.777655000 CEST	1
605	Jun 30, 2020 10:15:58.783777000 CEST	1
606	Jun 30, 2020 10:15:58.792194000 CEST	1
607	Jun 30, 2020 10:15:58.800814000 CEST	1
608	Jun 30, 2020 10:15:58.804691000 CEST	1
609	Jun 30, 2020 10:15:58.806353000 CEST	1
610	Jun 30, 2020 10:15:58.808324000 CEST	1
```

test -c 1,6,11 - hcxdumptool stayed on channel 1, 6, 11:
```
$ tshark -r test2.pcapng -T fields -E header=y -e frame.number -e frame.time -e wlan_radio.channel 
frame.number	frame.time	wlan_radio.channel
1	Jun 30, 2020 10:20:51.663377000 CEST	1
2	Jun 30, 2020 10:20:51.674227000 CEST	1
3	Jun 30, 2020 10:20:51.676730000 CEST	1
4	Jun 30, 2020 10:20:51.681756000 CEST	1
5	Jun 30, 2020 10:20:51.683598000 CEST	1
6	Jun 30, 2020 10:20:51.689327000 CEST	1
7	Jun 30, 2020 10:20:51.693720000 CEST	1
8	Jun 30, 2020 10:20:51.703986000 CEST	1
9	Jun 30, 2020 10:20:51.712054000 CEST	1
10	Jun 30, 2020 10:20:51.734566000 CEST	1
11	Jun 30, 2020 10:20:51.743907000 CEST	1
12	Jun 30, 2020 10:20:51.748964000 CEST	1
13	Jun 30, 2020 10:20:51.751552000 CEST	1
14	Jun 30, 2020 10:20:51.753180000 CEST	1
15	Jun 30, 2020 10:20:51.755196000 CEST	1
16	Jun 30, 2020 10:20:51.759440000 CEST	1
17	Jun 30, 2020 10:20:51.779397000 CEST	1
18	Jun 30, 2020 10:20:51.786796000 CEST	1
19	Jun 30, 2020 10:20:51.793928000 CEST	1
20	Jun 30, 2020 10:20:51.795781000 CEST	1
21	Jun 30, 2020 10:20:51.799808000 CEST	1
22	Jun 30, 2020 10:20:51.844352000 CEST	1
23	Jun 30, 2020 10:20:51.847280000 CEST	1
24	Jun 30, 2020 10:20:51.859557000 CEST	1
25	Jun 30, 2020 10:20:51.934253000 CEST	1
26	Jun 30, 2020 10:20:51.939612000 CEST	1
27	Jun 30, 2020 10:20:51.940740000 CEST	1
28	Jun 30, 2020 10:20:51.943839000 CEST	1
29	Jun 30, 2020 10:20:51.945858000 CEST	1
30	Jun 30, 2020 10:20:51.962354000 CEST	1
31	Jun 30, 2020 10:20:51.964712000 CEST	1
32	Jun 30, 2020 10:20:51.977438000 CEST	1
33	Jun 30, 2020 10:20:51.979460000 CEST	1
34	Jun 30, 2020 10:20:51.983766000 CEST	1
35	Jun 30, 2020 10:20:51.986183000 CEST	1
36	Jun 30, 2020 10:20:51.988358000 CEST	1
37	Jun 30, 2020 10:20:52.773081000 CEST	6
[... 199 lines cut ...]
237	Jun 30, 2020 10:20:56.878699000 CEST	11
238	Jun 30, 2020 10:20:56.878912000 CEST	11
239	Jun 30, 2020 10:20:56.880224000 CEST	11
240	Jun 30, 2020 10:20:56.880899000 CEST	11
241	Jun 30, 2020 10:20:56.881543000 CEST	11
242	Jun 30, 2020 10:20:56.900212000 CEST	11
243	Jun 30, 2020 10:20:56.908665000 CEST	11
244	Jun 30, 2020 10:20:56.914427000 CEST	11
245	Jun 30, 2020 10:20:56.915803000 CEST	11
246	Jun 30, 2020 10:20:56.917280000 CEST	11
247	Jun 30, 2020 10:20:56.918573000 CEST	11
248	Jun 30, 2020 10:20:56.919957000 CEST	11
249	Jun 30, 2020 10:20:56.931382000 CEST	11
250	Jun 30, 2020 10:20:56.935406000 CEST	11
251	Jun 30, 2020 10:20:56.936708000 CEST	11
252	Jun 30, 2020 10:20:56.944220000 CEST	11
253	Jun 30, 2020 10:20:56.945534000 CEST	11
254	Jun 30, 2020 10:20:56.946833000 CEST	11
255	Jun 30, 2020 10:20:56.948324000 CEST	11
256	Jun 30, 2020 10:20:57.024302000 CEST	11
257	Jun 30, 2020 10:20:57.029284000 CEST	11
258	Jun 30, 2020 10:20:57.038248000 CEST	11
259	Jun 30, 2020 10:20:57.159715000 CEST	11
260	Jun 30, 2020 10:20:57.201183000 CEST	11
```

ZerBea, 2020-06-30:
BTW:
Don't upgrade your kernel because ath9k driver is broken on kernel >= 5.4.7:
https://bugzilla.kernel.org/show_bug.cgi?id=208251

Czechball, 2020-07-01:
Sorry for taking so long with my response, but I wasn't able to replicate the issue anymore... I have no idea what fixed it, but the -c switch works for me now, with the exact same command as before... I even tried my other rt2800usb interface and it worked as well (my ultimate goal was to run 2 instances of hcxdumptool with 2 separate interfaces, each having half of 2,4 GHz channels assigned to it to maximize efficiency). I have no idea what caused this problem and what solved it, but it doesn't happen anymore. Oh and thanks for the warning about the kernel bug. Let me know if you still want my pcap files, but I'll close this issue for now. Thanks for your help.

closed 2020-07-01 by Czechball