ZerBea/hcxdumptool #120: How can I use 5GHz

fa1rid, 2020-07-06:
The tool works with 2.4GHz by default, how can I use 5GHz?

ZerBea, 2020-07-06:
To use 5GHz, you must leave the automatic and add the channels by hand using the -c switch. 5GHz channels are not included, because the range is poor (very poor). The device must support 5GHz and wirelessregdb must allow transmitting on that channels.
Option -C will give you an overview about the channels that the device support:
$ hcxdumptool -i wlp39s0f3u1u1u2 -C
```
initialization...
available channels:
  1 / 2412MHz (14 dBm)
  2 / 2417MHz (14 dBm)
  3 / 2422MHz (14 dBm)
  4 / 2427MHz (14 dBm)
  5 / 2432MHz (14 dBm)
  6 / 2437MHz (14 dBm)
  7 / 2442MHz (14 dBm)
  8 / 2447MHz (14 dBm)
  9 / 2452MHz (14 dBm)
 10 / 2457MHz (14 dBm)
 11 / 2462MHz (14 dBm)
 12 / 2467MHz (14 dBm)
 13 / 2472MHz (14 dBm)
 14 / 2484MHz (14 dBm)
 36 / 5180MHz (17 dBm)
 40 / 5200MHz (17 dBm)
 44 / 5220MHz (17 dBm)
 48 / 5240MHz (17 dBm)
 52 / 5260MHz (17 dBm)
 56 / 5280MHz (17 dBm)
 60 / 5300MHz (17 dBm)
 64 / 5320MHz (17 dBm)
100 / 5500MHz (17 dBm)
104 / 5520MHz (17 dBm)
108 / 5540MHz (17 dBm)
112 / 5560MHz (17 dBm)
116 / 5580MHz (17 dBm)
120 / 5600MHz (17 dBm)
124 / 5620MHz (17 dBm)
128 / 5640MHz (17 dBm)
132 / 5660MHz (17 dBm)
136 / 5680MHz (17 dBm)
140 / 5700MHz (17 dBm)
149 / 5745MHz (17 dBm)
153 / 5765MHz (17 dBm)
157 / 5785MHz (17 dBm)
161 / 5805MHz (17 dBm)
165 / 5825MHz (17 dBm)
```

Than select the channels by -c option:
$ hcxdumptool -i wlp39s0f3u1u1u2 -c 1,6,11,56,100 --enable_status=31 -o dump.pcapng

That is described in --help:
```
-c <digit>     : set scan list (1,2,3, ...)
                 default scan list: 1...13
                 maximum entries: 127
                 allowed channels (depends on the device):
                 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14
                 32, 34, 36, 38, 40, 42, 44, 46, 48, 50, 52, 54, 56, 58, 60, 62, 64, 68, 96
                 100, 102, 104, 106, 108, 110, 112, 114, 116, 118, 120, 122, 124, 126, 128
                 132, 134, 136, 138, 140, 142, 144, 149, 151, 153, 155, 157, 159
                 161, 165, 169, 173
```

fa1rid, 2020-07-06:
I wanted an option to automatically scan all channels above 14 (Which I believe is the 5Ghz band) I don't want to manually select channels as it will make command very long a difficult to type by hand. What do you  think?

ZerBea, 2020-07-06:
There are to much issues regarding 5GHz adapters and 5GHz supported channels. If we do this by automatic we will receive too many errors and hcxdumptool will terminate.

ZerBea, 2020-07-06:
I'm doing some more tests. If I'll find a driver without issues on 5GHz, I'll add 5GHz support by default.
At this moment neither my TP-LINK Archer T2UH nor my ASUS USB-AC51 are working without issues on 5GHz.

ZerBea, 2020-07-06:
But your idea is not so bad. Let me think about some predefined scanlists (--scanlist=<digit>)
--scanlist=1  ->  channel 1...13
--scanlist=2  ->  channel  36..165
--scanlist=3  ->  channel 1...165
--scanlist=4  ->  channel 1,6,11
...

ZerBea, 2020-07-06:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/775c3366aaf6f4053a1efbefaab88d3cb3f28158
I added a new option -s to set predefined scanlists. We started with this 3 lists:
```
-s <digit>     : set scan predefined scanlist
                 0 = 1,6,11,3,5,1,6,11,2,4,1,6,11,7,9,1,6,11,8,10,1,6,11,12,13 (default)
                 1 = 1,2,3,4,5,6,7,8,9,10,11,12,13
                 2 = 36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136,140,149,153,157,161,165
                 3 = 1,2,3,4,5,6,7,8,9,10,11,12,13,36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136,140,149,153,157,161,165
```
Please notice:
Option -s will override option -c

Please test and close issue report if it works.
Please notice that not all drivers support full packet injection on 5GHz!

fa1rid, 2020-07-06:
Alright thank you for doing that so fast. You are awesome! 
I will test and close the issue.

fa1rid, 2020-07-07:
I tried `-s 2` and it's still showing channels under 13

ZerBea, 2020-07-07:
Thanks for reporting it. I reversed s2 and s3. I'll fix that.
Also I noticed another issue in driver behavior. In every case it will start scanning on channel 1, even though if we set channel 30. I have to find a workaround for this driver behavior.

ZerBea, 2020-07-07:
I pushed another commit
https://github.com/ZerBea/hcxdumptool/commit/db03140fe9f336efae94b7172b03897983b54bfc
that include the -s2 -s3 fix and the workaround if driver fall back to channel 1 on option -s 2.
Please test.

ZerBea, 2020-07-07:
The mt76x0u 5GHz issue is reported on kernel.org, too. You can follow the progress, here:
https://bugzilla.kernel.org/show_bug.cgi?id=208483

ZerBea, 2020-07-07:
I closed that issue report, because the new scan lists are implemented.
The driver issues are reporter on kernel.org and we have to wait for a fix.
You can still ask your questions here.

closed 2020-07-07 by ZerBea

fa1rid, 2020-08-25:
Did you try 
Realtek RTL8812AU Chipset
Alfa AWUS036ACH

ZerBea, 2020-08-26:
Yes, but Realtek RTL8812AU isn't working out of the box on the Linux kernel. You have to compile a third party driver and you have to disable the stock driver, supplied with the kernel.
This driver from aircrack-ng (kimocoder)
https://github.com/aircrack-ng/rtl8812au
is partly working, but you'll run into several issues:
https://github.com/aircrack-ng/rtl8812au/issues
In addition to this issues you'll run into more issues  due to the NETLINK dependency of the driver. The driver doesn't accept hcxdumptool ioctl() system calls to set monitor mode and iw must be used to do this. hcxdumptool doesn't use NETLINK.
Also you must(!) install crda and configure the correct country code (wireless-regdb) to allow active scanning and the whole frequency range.