ZerBea/hcxdumptool #124: Failed to read packets

zonalimitatore, 2020-07-12:
Hi there.
I run Kali on a virtual machine.
I use Alfa rtl8187. I set right the monitor mode,but when I run the line command,after about 120seconds (sometimes more), I get that error.
It's very frustrating,because I can't understand why I got that.

I try to mount a heatskin on alfa's chip (thinking was an error by heat),but no.

Someone can help me,or have this issue?

ZerBea, 2020-07-13:
Failed to read packets usually means that hcxdumptool doesn't have full access to the interface. But to help you, some more information is required. Please comment your command line and the error message.
Stop all services (on host and VM) that take access to the interface.
Than comment output of:
$ hcxdumptool -I
$ hcxdumptool -m "your interface"
$ hcxdumptool -i "your interface" --check_driver
$ hcxdumptool -i "your interface" --check_injection

Run the commands exactly in the sequence mentioned above.
Your output should look like this:
```
$ hcxdumptool -I
wlan interfaces:
00c0ca3124eb wlp5s0f4u2 (rtl8187)

$ hcxdumptool -m wlp5s0f4u2
setting interface wlp5s0f4u2 to monitor mode

$ hcxdumptool -i wlp5s0f4u2 --check_driver
initialization...
starting driver test...
interface is already in monitor mode
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$  hcxdumptool -i wlp5s0f4u2 --check_injection
initialization...
interface is already in monitor mode
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 343 to 227 

terminating...
```

BTW:
hcxdumptool is running an own monitor mode. How did you set your monitor mode?

zonalimitatore, 2020-07-13:
I've try 2 ways.One let the program set monitor mode (after manually stop wpa_supplicant and network_manager)
Second way i try,by do:

ip link set wlan0 down
iw dev wlan0 set type monitor
ip link set wlan0up

wlan interfaces:
$ hcxdumptool -I  >>> 00c0ca728a13 wlan0 (rtl8187)

$ hcxdumptool -m wlan0  >>> setting interface wlan0 to monitor mode

$ hcxdumptool -i "your interface" >>> initialization...
starting driver test...
interface is already in monitor mode
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$ hcxdumptool -i "your interface" --check_injection >>> initialization...
interface is already in monitor mode
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 150 to 99

ZerBea, 2020-07-13:
Everything looking fine and hcxdumptool is able to control the device (set channel, receive packets and transmit packets).
The ration 150:99 looking fine, too.
When you receive the error, what does dmesg show?

zonalimitatore, 2020-07-13:
I never try to do a dmesg.
How I can do it? Simply do dmesg when message appear?

ZerBea, 2020-07-13:
When hcxdumptool show the error message, just stop it. Then type dmesg and check the output (usually the last 20 lines) for errors and or warnings.

ZerBea, 2020-07-13:
dmesg output for an ALFA AWUS036H should look like this:
```
[ 1353.130964] usb 3-2: new high-speed USB device number 4 using xhci_hcd
[ 1353.279898] usb 3-2: New USB device found, idVendor=0bda, idProduct=8187, bcdDevice= 1.00
[ 1353.279904] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1353.279907] usb 3-2: Product: RTL8187_Wireless
[ 1353.279910] usb 3-2: Manufacturer: Manufacturer_Realtek_RTL8187_
[ 1353.279912] usb 3-2: SerialNumber: 00C0CA3124EB
[ 1353.642714] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[ 1353.643467] ieee80211 phy1: hwaddr 00:c0:ca:31:24:eb, RTL8187vB (default) V1 + rtl8225z2, rfkill mask 2
[ 1353.667464] rtl8187: Customer ID is 0xFF
[ 1353.668659] rtl8187: wireless switch is on
[ 1353.668700] usbcore: registered new interface driver rtl8187
[ 1353.682085] rtl8187 3-2:1.0 wlp5s0f4u2: renamed from wlan0
...
[ 1420.017494] device wlp5s0f4u2 entered promiscuous mode
...
[ 1427.428410] device wlp5s0f4u2 left promiscuous mode
```
Especially after "device wlp5s0f4u2 entered promiscuous mode" no warning or error message should appear.
BTW:
Are you running latest KALI (Kali 2020.2)?

zonalimitatore, 2020-07-13:
Ok thanks.Yes i'm in last Kali.
At the first stop i do that ;)

zonalimitatore, 2020-07-13:
voilà:

[   34.430536] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   34.430805] ieee80211 phy0: hwaddr 00:c0:ca:72:8a:13, RTL8187vB (default) V1 + rtl8225z2, rfkill mask 2
[   34.906461] rtl8187: Customer ID is 0xFF
[   34.921463] rtl8187: wireless switch is on
[   34.921516] usbcore: registered new interface driver rtl8187
[   36.551168] e1000: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   36.552002] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  104.885091] wlan0: authenticate with bc:e2:65:19:35:fa
[  105.756964] wlan0: send auth to bc:e2:65:19:35:fa (try 1/3)
[  105.964043] wlan0: send auth to bc:e2:65:19:35:fa (try 2/3)
[  106.166756] wlan0: send auth to bc:e2:65:19:35:fa (try 3/3)
[  106.371233] wlan0: authentication with bc:e2:65:19:35:fa timed out
[  162.986432] device wlan0 entered promiscuous mode
[  168.600715] rtl8187: wireless radio switch turned off
[  174.037729] rtl8187: wireless radio switch turned on
[  195.948612] device wlan0 left promiscuous mode

zonalimitatore, 2020-07-13:
Mmmmm...Seems someone turnoff my card 📶  😥

ZerBea, 2020-07-14:
Great, thanks. We located the issue:
The device entered promiscuous - that is expected
`[ 905.909148] device wlan0 entered promiscuous mode`
But than, something switch the device off - that is not ecpected:
`[ 1460.720478] rtl8187: wireless radio switch turned off`
hcxdumptool is not able to read packets any longer (the device is off) and you will receive the error message.

hcxdumptool doesn't switch the wireless radio off. There is no code for this inside.
Unfortunately, now you have to find out, why the radio switch off.
You are not alone with this issue:
https://askubuntu.com/questions/260464/wifi-randomly-turning-off-via-hardware-switch
https://www.raspberrypi.org/forums/viewtopic.php?t=61790

For me, it looks like the device can't get enough power from the USB port, and turns off. Also it could be possible that power management of the device is misconfigured or the host shut down the device.

zonalimitatore, 2020-07-14:
Yep,i think it so.
In fact is not a code problem.
I've to find why switch off.For leave any doubt i can install it on a "true o.s." in a portable pc,and not in a virtual machine.

Anyway,thank a lot,for the support and for the program 😋

closed 2020-07-14 by zonalimitatore

ZerBea, 2020-07-14:
No problem. You're welcome.