ZerBea/hcxdumptool #125: Plainmasterkey was transmitted unencrypted.

vido89, 2020-07-18:
How can I check/know if wlan-key or plainmasterkey was transmitted unencrypted ? I can find anything abut this on web ?

vido89, 2020-07-19:
```
./hcxhashtool --pmk="01234567" ../hcxdumptool/output.pcapng 
wrong PMK length 
```

ZerBea, 2020-07-19:
You received a correct warning, because you typed a PreShardKey (PSK) instead of a PlainMasterKey (PMK).
A PMK is 32 bytes long. A PSK is 8 to 63 characketers long
To find PMKs. run hcxpcapngtool -E option and search for 32 bytes HEX strings.
To find PSKs. run hcxpcapngtool -E option and search for for possible PSKs.

example:
$ wlangenpmk -e TESTNET -p 01234567

essid (networkname)....: TESTNET
password...............: 01234567
plainmasterkey (SHA1)..: 8e24fe34120f488f4bd793637d078bebb7fa636c9c55dbd19502afa508fe1218
plainmasterkey (SHA256): fdefeaf07a9ab1b459c6bdf56d8e3713678af6d038b870f3920a0f863a114ebf

BTW:
Searching for PMKs is only useful on Authentication Management Key systems (AKM) or EAP based systems.

I closed this, because it isn't an hcxtools issue, but you can still ask your questions here.

closed 2020-07-19 by ZerBea

vido89, 2020-07-19:
where I can get wlangenpmk I dont have it in hcxtools

ZerBea, 2020-07-19:
It's from my key generator playground:
https://github.com/ZerBea/hcxkeys
There is an OpenCL version inside, too.

vido89, 2020-07-19:
When I run `./hcxpcaptool -E ../hcxdumptool/output.pcapng` I got nothing as output

ZerBea, 2020-07-19:
hcxpcaptool -E wordlist output.pcapng
cat wordlist

vido89, 2020-07-19:
```
$ ./hcxpcaptool -E /media/slobodan/Podaci/Lista-rijeci/8ivise ../hcxdumptool/output.pcapng 

reading from output.pcapng
                                                
summary capture file:                           
---------------------
file name........................: output.pcapng
file type........................: pcapng 1.0
file hardware information........: x86_64
capture device vendor information: 3c3300
file os information..............: Linux 5.3.0-62-generic
file application information.....: hcxdumptool 6.0.7-22-g2f82e84 (custom options)
network type.....................: DLT_IEEE802_11_RADIO (127)
endianness.......................: little endian
read errors......................: flawless
minimum time stamp...............: 18.07.2020 13:07:34 (GMT)
maximum time stamp...............: 18.07.2020 13:15:01 (GMT)
packets inside...................: 2088
skipped damaged packets..........: 0
packets with GPS NMEA data.......: 0
packets with GPS data (JSON old).: 0
packets with FCS.................: 0
beacons (total)..................: 50
beacons (WPS info inside)........: 6
probe requests...................: 36
probe responses..................: 24
association requests.............: 7
association responses............: 13
reassociation requests...........: 3
reassociation responses..........: 8
authentications (OPEN SYSTEM)....: 28
authentications (BROADCOM).......: 6
EAPOL packets (total)............: 1919
EAPOL packets (WPA2).............: 1919
PMKIDs (zeroed and useless)......: 1160
PMKIDs (not zeroed - total)......: 17
PMKIDs (WPA2)....................: 1255
PMKIDs from access points........: 6
PMKIDs from stations.............: 11
best handshakes (total)..........: 12 (ap-less: 7)
best PMKIDs (total)..............: 8

summary output file(s):
-----------------------
```

ZerBea, 2020-07-19:
My fault. I didn't realize that you're running hcxpcaptool.
hcxpcaptool is deprecated and will be removed, soon.
Please use hcxpcapngtool v6 in combination with hcxdumptool v6

From README.md:
```
Detailed description
--------------

| Tool           | Description                                                                                                            |
| -------------- | ---------------------------------------------------------------------------------------------------------------------- |
| hcxpcapngtool  | Provide new hashcat format 22000                                                                                       |
| hcxhashtool    | Provide various filter operations on new PMKID/EAPOL hash line                                                         |
| hcxpsktool     | Calculates candidates for hashcat and john based on based on hcxpcapngtool output (-c -o -z- -U) or commandline input  |
| hcxwltool      | Calculates candidates for hashcat and john based on hcxpcaptool output (-E -I- -U)                                     |
| hcxhash2cap    | Converts hash file (PMKID&EAPOL, PMKID, EAPOL-hccapx, EAPOL-hccap, WPAPSK-john) to cap                                 |
| wlancap2wpasec | Upload multiple (gzip compressed) pcapng, pcap and cap files to https://wpa-sec.stanev.org                             |
| whoismac       | Show vendor information and/or download oui reference list                                                             |

| deprecated     | obsolete when hashcat and JtR moved to new PMKID/EAPOL hash line - no longer under maintenance                         |
| -------------- | ---------------------------------------------------------------------------------------------------------------------- |
| hcxmactool     | Various MAC based filter operations on HCCAPX and PMKID files - convert hccapx and/or PMKID to new hashline format     |
| hcxpcaptool    | Shows info of pcap/pcapng file and convert it to other hashformats accepted by hashcat and John the Ripper             |
| hcxpmkidtool   | CPU based tools to verfiy a PMKID                                                                                      |
| hcxessidtool   | Various ESSID based filter operations on HCCAPX and PMKID files                                                        |
| hcxhashcattool | Convert old hashcat (<= 5.1.0) separate potfile (2500 and/or 16800) to new potfile format                              |
| wlanwkp2hcx    | Converts wpk (ELMCOMSOFT EWSA projectfile) to hccapx                                                                   |
| wlanhcx2ssid   | Strips BSSID, ESSID, OUI                                                                                               |
| wlanhcxinfo    | Shows detailed info from contents of hccapxfile                                                                        |
| wlanhcxcat     | Simple password recovery tool for WPA/WPA2/WPA2 SHA256 AES-128-CMAC (hash-modes 2500, 2501)                            |
| wlanpmk2hcx    | Converts plainmasterkey and ESSID for use with hashcat hash-mode 12000 or john PBKDF2-HMAC-SHA1                        |
| wlanjohn2hcx   | Converts john wpapsk hashfiles for use with hashcat hash-modes 2500, 2501                                              |
| wlanhcx2john   | Converts hccapx to format expected by John the Ripper                                                                  |
```

vido89, 2020-07-19:
```
./hcxpcapngtool -E /media/slobodan/Podaci/Lista-rijeci/8ivise ../hcxdumptool/output.pcapng 
reading from output.pcapng...

summary capture file
--------------------
file name................................: output.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.3.0-62-generic
application..............................: hcxdumptool 6.0.7-22-g2f82e84
interface name...........................: wlx3c3300200748
interface vendor.........................: 3c3300
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 10ae60ca47ce (incremented on every new client)
MAC CLIENT...............................: f0a22563d753
REPLAYCOUNT..............................: 65225
ANONCE...................................: d618b0300abf05d105cda1993e7a9f583cc6fb7276286c316a2243aed63c31ec
SNONCE...................................: b3d00c84fea1b93670b144a39ab909f5efed7707327bf9f07f8b3c81a8f5146e
timestamp minimum (GMT)..................: 18.07.2020 15:07:34
timestamp maximum (GMT)..................: 18.07.2020 15:15:01
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 2088
BEACON (total)...........................: 50
PROBEREQUEST.............................: 31
PROBEREQUEST (directed)..................: 5
PROBERESPONSE............................: 24
AUTHENTICATION (total)...................: 28
AUTHENTICATION (OPEN SYSTEM).............: 28
ASSOCIATIONREQUEST (total)...............: 7
ASSOCIATIONREQUEST (PSK).................: 7
REASSOCIATIONREQUEST (total).............: 3
REASSOCIATIONREQUEST (PSK)...............: 3
EAPOL messages (total)...................: 1919
EAPOL RSN messages.......................: 1919
ESSID (total unique).....................: 68
EAPOLTIME gap (measured maximum usec)....: 21234303
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (suggested NC)...........: 4
EAPOL M1 messages (total)................: 1798
EAPOL M2 messages (total)................: 89
EAPOL M3 messages (total)................: 21
EAPOL M4 messages (total)................: 11
EAPOL pairs (total)......................: 864
EAPOL pairs (best).......................: 12
EAPOL ROGUE pairs........................: 7
EAPOL M12E2 (challenge)..................: 8
EAPOL M32E2 (authorized).................: 4
PMKID (total)............................: 106
PMKID (useless)..........................: 1160
PMKID (best).............................: 8
PMKID ROGUE..............................: 5
```

ZerBea, 2020-07-19:
cat /media/slobodan/Podaci/Lista-rijeci/8ivise
should show you ESSIDs, PSKs in the clear (if transmitted by some CLIENTs) and PMKs (if transmitted by some CLIENTs)

BTW:
timestamp minimum (GMT)..................: 18.07.2020 15:07:34
timestamp maximum (GMT)..................: 18.07.2020 15:15:01
the time period, running hcxdumptool is too short! Running over a longer time will give you more results.

ZerBea, 2020-07-19:
```
$ hcxpcapngtool -E wordlist 202007111207.pcapng
reading from 202007111207.pcapng...

summary capture file
--------------------
file name................................: 202007111207.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.7.7-arch1-1
application..............................: hcxdumptool 6.0.7-17-g69ff356
interface name...........................: wlp3s0f0u2
interface vendor.........................: c83a35
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 00234a3c1f72 (incremented on every new client)
MAC CLIENT...............................: fcc233c8321c
REPLAYCOUNT..............................: 64939
ANONCE...................................: af02e2570508cfcb31805c48acbe7971d822b827eda33231b72949f95a10ed7d
SNONCE...................................: a378c4eb74e582f3c84618e4e799b42973bb79bd2f3d6f633115717638d30146
timestamp minimum (GMT)..................: 11.07.2020 12:07:37
timestamp maximum (GMT)..................: 11.07.2020 12:07:47
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 422
BEACON (total)...........................: 21
PROBEREQUEST.............................: 7
PROBERESPONSE............................: 9
AUTHENTICATION (total)...................: 16
AUTHENTICATION (OPEN SYSTEM).............: 16
ASSOCIATIONREQUEST (total)...............: 1
ASSOCIATIONREQUEST (PSK).................: 1
REASSOCIATIONREQUEST (total).............: 2
REASSOCIATIONREQUEST (PSK)...............: 2
EAPOL messages (total)...................: 352
EAPOL RSN messages.......................: 352
ESSID (total unique).....................: 20
EAPOLTIME gap (measured maximum usec)....: 20304
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (recommended NC).........: 8
EAPOL M1 messages (total)................: 324
EAPOL M2 messages (total)................: 21
EAPOL M3 messages (total)................: 5
EAPOL M4 messages (total)................: 2
EAPOL pairs (total)......................: 279
EAPOL pairs (best).......................: 3
EAPOL ROGUE pairs........................: 3
EAPOL M12E2 (challenge)..................: 3
PMKID (total)............................: 104
PMKID (best).............................: 4
PMKID ROGUE..............................: 3

$ wc wordlist
 21  29 279 wordlist
```
working as expected.....

vido89, 2020-07-19:
Im getting a word list how can I know which password go to which AP. My wpa password is TestTest but in list its test test ?

vido89, 2020-07-19:
How did you captured that list which tool did you used ? hcxpcapngtool ?

ZerBea, 2020-07-19:
Just run hcxdumptool for some hours.
`$ hcxdumptool -i interface -o test.pcapng --enable_status=31`

Than convert file
```
$ hcxpcapngtool -E wordlist -o test.22000 test.pcapng
now run hashcat:
$ hashcat -m 22000 test.pcapng wordlist
or in combination with rules:
$ hashcat --stdout -r rockyou-30000.rule  wordlist | hashcat -m 22000 test.pcapng
```
The same with john:
```
$ hcxpcapngtool -E wordlist --john=test.john test.pcapng
$ john -w=wordlist --format=wpapsk-opencl test.john
$ john --single --format=wpapsk-opencl test.john
```

ZerBea, 2020-07-19:
As an alternative, you can use hcxwltool, too:

```
$ hcxwltool -i wordlist --straight > tmp.list
$ hcxwltool -i wordlist --digit >> tmp.list
$ hcxwltool -i wordlist --xdigit >> tmp.list
$ hcxwltool -i wordlist --digit --length=8 >> tmp.list
$ hcxwltool -i wordlist --digit --length=9 >> tmp.list
$ hcxwltool -i wordlist --digit --length=10 >> tmp.list
$ hcxwltool -i wordlist --digit --length=11 >> tmp.list
$ hcxwltool -i wordlist --digit --length=12 >> tmp.list
$ cat tmp.list | sort | uniq | hashcat -m 22000 test.pcapng
```
or in combination with rules:
```
$ cat wordlist | grep -v "\$HEX\[" > wordlist.tmp
$ hcxwltool -i wordlist.tmp --digit >> tmp.list
$ hcxwltool -i wordlist.tmp --xdigit >> tmp.list
$ hcxwltool -i wordlist.tmp --digit --length=8 >> tmp.list
$ hcxwltool -i wordlist.tmp --digit --length=9 >> tmp.list
$ hcxwltool -i wordlist.tmp --digit --length=10 >> tmp.list
$ hcxwltool -i wordlist.tmp --digit --length=11 >> tmp.list
$ hcxwltool -i wordlist.tmp --digit --length=12 >> tmp.list
$ cat tmp.list | tr -d [:punct:] >> tmp1.list
$ cat tmp.list | sed 's/[^0-9]*//g' >> tmp1.list
$ cat tmp.list | sed 's/[^0-9a-fA-F]*//g' >> tmp1.list
$ hashcat --stdout -r "rockyou-30000..rule" tmp1.list > tmp3.list
$ hcxwltool -i tmp.list --straight >> tmp1.list
$  cat tmp.list | sed 's/[-_.: @()]/\n/g' | sed 's/[^a-zA-Z]*//g' | sort | uniq > tmp2.list
$ cat tmp.list | sed 's/[^a-zA-Z]*//g' | sort | uniq >> tmp2.list
$ hashcat --stdout -r "$HOME/WLAN/Rules/hcxappend.rule" tmp2.list >> tmp3.list
$  hashcat --stdout -r "$HOME/WLAN/Rules/hcxcase.rule" tmp2.list >> tmp3.list
$ cat wordlist.tmp wordlist tmp.list tmp1.list tmp2.list tmp3.list | awk 'length($0) >7' | sort | uniq > test.list
$ hashcat -m 22000 test.22000 test.list
```

This lists are generated from WiFi traffic - the hit rate is very good.
Try also other combinations and rules.

vido89, 2020-07-20:
Ok tnx, but how wifi password end up as plain text in wifi traffic ?

ZerBea, 2020-07-20:
That can be caused by several reasons:
user fault: user typed PSK as ESSID and ESSID as PSK
system fault: NVRAM overflow
configuration error: wpa-supplicant.conf mismatch
and more
in this thread, you can find a nice example:
https://hashcat.net/forum/thread-6661-post-35891.html#pid35891
Download test.cap.zip and decompress it
Now run
$ hcxpcapngtool -E wordlist -o test.2000 test.cap
$ hashcat -m 22000 test.22000 wordlist

BTW:
Nearly everything is explained in this thread:
https://hashcat.net/forum/thread-6661.html

vido89, 2020-07-20:
How can see recovered passwords -o for some reason does not work for me

```
$ ../hashcat/hashcat -m 22000 test.22000 -o "temp_temp" wordlist2 
hashcat (v6.0.0-36-g7cba2257) starting...

* Device #1: WARNING! Kernel exec timeout is not disabled.
             This may cause "CL_OUT_OF_RESOURCES" or related errors.
             To disable the timeout, see: https://hashcat.net/q/timeoutpatch
* Device #2: WARNING! Kernel exec timeout is not disabled.
             This may cause "CL_OUT_OF_RESOURCES" or related errors.
             To disable the timeout, see: https://hashcat.net/q/timeoutpatch
CUDA API (CUDA 10.2)
====================
* Device #1: GeForce GTX 750 Ti, 1760/2000 MB, 5MCU

OpenCL API (OpenCL 1.2 CUDA 10.2.185) - Platform #1 [NVIDIA Corporation]
========================================================================
* Device #2: GeForce GTX 750 Ti, skipped

Minimum password length supported by kernel: 8
Maximum password length supported by kernel: 63

Hashes: 45 digests; 45 unique digests, 25 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates
Rules: 1

Applicable optimizers applied:
* Zero-Byte
* Slow-Hash-SIMD-LOOP

Watchdog: Temperature abort trigger set to 90c

INFO: Removed 6 hashes found in potfile.

Host memory required for this attack: 151 MB

Dictionary cache hit:
* Filename..: wordlist2
* Passwords.: 81
* Bytes.....: 743
* Keyspace..: 81

The wordlist or mask that you are using is too small.
This means that hashcat cannot use the full parallel power of your device(s).
Unless you supply more work, your cracking speed will drop.
For tips on supplying more work, see: https://hashcat.net/faq/morework

Approaching final keyspace - workload adjusted.  

                                                 
Session..........: hashcat
Status...........: Exhausted
Hash.Name........: WPA-PBKDF2-PMKID+EAPOL
Hash.Target......: test.22000
Time.Started.....: Mon Jul 20 18:53:39 2020 (1 sec)
Time.Estimated...: Mon Jul 20 18:53:40 2020 (0 secs)
Guess.Base.......: File (wordlist2)
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:     1174 H/s (0.21ms) @ Accel:16 Loops:32 Thr:1024 Vec:1
Recovered........: 6/45 (13.33%) Digests, 3/25 (12.00%) Salts
Progress.........: 2025/2025 (100.00%)
Rejected.........: 1175/2025 (58.02%)
Restore.Point....: 81/81 (100.00%)
Restore.Sub.#1...: Salt:24 Amplifier:0-1 Iteration:0-1
Candidates.#1....: 12345678 -> zuriosam
Hardware.Mon.#1..: Temp: 37c Fan: 34% Util: 96% Core:1176MHz Mem:2700MHz Bus:16

Started: Mon Jul 20 18:53:38 2020
Stopped: Mon Jul 20 18:53:40 2020
```

ZerBea, 2020-07-20:
First of all, start with the example, I mentioned before. Than take a look inside the test.cap file and identify the PSK transmitted by the CLIENT. Compare it with the result of hcxpcapngtool -E wordlist.
hcxtools are analysis tools. It is not like pixi dust or wep attack that present you a PSK. Running hcxdumptool for a few minutes is not enough. You have to monitor the traffic of as many CLIENTs as possible.
Run hcxdumptool for several hours. Than take a look inside the -E output and search for possible PSKs.
Read the whole thread:
https://hashcat.net/forum/thread-6661.html
especially the parts on how to create a basic environment.

vido89, 2020-07-20:
I run it for whole night and now I'm trying to crack passwords, as you can se I have Recovered........: 6/45 but its not writeen in file with -o "temp_temp" for some reason, it worked before

vido89, 2020-07-20:
Why hashcat wont to write results in -o "file_name" ? what Im doing wrong ?

ZerBea, 2020-07-20:
Because the PSKs are already recovered. hashcat told you that:
INFO: Removed 6 hashes found in potfile.

To see the PSKs use --show option
```
$ hashcat --help | grep show
     --show                     |      | Compare hashlist with potfile; show cracked hashes   |
     --left                     |      | Compare hashlist with potfile; show uncracked hashes |
```
https://hashcat.net/wiki/doku.php?id=frequently_asked_questions

vido89, 2020-07-21:
Great tnx @ZerBea In my AP list i have 

`WPA*2*6b7b7cb71388cac71dbe12695f110153*e4beed7976ef*a89675f9e979*44616d6a616 .....000001630140100000fac020100000fac040100000fac020000*10` what is this ?

ZerBea, 2020-07-21:
That is the converted hash in new hashcat v.6 format -m 22000. In your case it is an EAPOL MESSAGE PAIR.
WPA*01* = PMKID (old -m 16800)
WPA*02* = EAPOL MESSAGE PAIR (old -m 2500 hccapx)
Please read more here:
https://github.com/hashcat/hashcat/issues/1816