ZerBea/hcxdumptool #128: Libressl

avalonv, 2020-07-23:
Would it be possible to support libressl as an alternative openssl?

ZerBea, 2020-07-23:
Why?
As long as Arch doesn't replace openSSL with librSSL and nearly every other tool is using openSSL, there is no need for me to support it.
https://bbs.archlinux.org/viewtopic.php?id=180502
https://bbs.archlinux.org/viewtopic.php?id=246756

avalonv, 2020-07-23:
I think lots of non-arch users want to use this software. Void is a relatively well known distribution that uses libressl, and that prevents us from building hcxdumptool.

ZerBea, 2020-07-23:
https://github.com/void-linux/void-packages/issues/20935

And I'm not sure that libressl support all HMAC/OMAC functions, used by hcxdumptool/hcxtools.
BTW:
There is a similar request for openWRT:
https://github.com/ZerBea/hcxtools/issues/106

avalonv, 2020-07-23:
Understandable. I hadn't seen that issue before, I personally don't have a huge preference for either library, but I hope they do switch to OpenSSL if it makes things easier.

closed 2020-07-23 by avalonv

ZerBea, 2020-07-23:
You may have noticed that I haven't closed this request:
https://github.com/ZerBea/hcxtools/issues/106

mbedTLS is really lightweight and I think about it to switch over to this lib - if all functions can be realized with this lib, inclusive Elliptic Curve Diffie-Hellman (ECDH)-​Exchange and Elliptic Curve Digital Signature Algorithm (ECDSA).