ZerBea/hcxdumptool #129: gpio options not working

Czechball, 2020-07-24:
I wired up my led and button as shown in https://github.com/ZerBea/hcxdumptool/blob/master/docs/gpiowait.odg, but when I start hcxdumptool with ``hcxdumptool -i wlan1  --enable_status=19 --gpio_button=4 --gpio_statusled=17``, hcxdumptool instantly terminates after startup. When I omit the --gpio_button option, hcxdumptool works but the connected LED still doesn't light up.

hcxdumptool output:
```
start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1
INTERFACE HARDWARE MAC....: 940c6d896175
DRIVER....................: ath9k_htc
DRIVER VERSION............: 4.19.118-Re4son-v7+
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 00bb3ae6be05 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 00bb3ae6be06 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 00bb3ae6be07 (incremented on every new client)
ROGUE (CLIENT)............: b0ece17ff7f7
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 61448
ANONCE....................: 440ca3a8bada6205a2f6986ad721d5b9e10e0150bf01dd18a3b4540010360380
SNONCE....................: 8073bb4b89c5b2759eb3c444f08ae11a4585363b2ba285808e3c1faba1e66fe4

terminating...
```
My hcxdumptool version is `6.0.7-33-g570cca4`

ZerBea, 2020-07-24:
Double check you circut. On startup hcxdumptool check the status of the pull down resistors (330 Ohm and 10 KOhm). This resistors are mandatory to ground the line. Otherwise hcxdumptool assume the line is high and terminates.

ZerBea, 2020-07-24:
What Raspberry Pi model do you use?

Czechball, 2020-07-24:
Raspberry Pi 3B+

Czechball, 2020-07-24:
[image]
here's a picture of my circuit

Czechball, 2020-07-24:
nevermind, I'm an idiot. I'll rewire the entire thing

ZerBea, 2020-07-24:
Let's do another test to make sure the RPI 3 is detected correctly:

Please add a printf() here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5774
```
gpiobasemem = getrpirev();
printf("\n\ndebug  %08x\n\n", gpiobasemem);
if(gpiobasemem == 0)
```
we expect to see value of GPIO_PERI_BASE_NEW = 0x3F000000

ZerBea, 2020-07-24:
BTW:
It happened very often to me, that the wires are connected the wrong way. The RPIU doesn't break, because the lines are secured by resistors, but it will not work as expected.
[image]

Czechball, 2020-07-24:
What I did wrong in my photo was that I assumed that the main lines in that breadboard are interconnected vertically, when they're actually connected horizontally...
[image]

ZerBea, 2020-07-24:
Correct, I have a similar kind of a board.

Czechball, 2020-07-24:
Alright, I did the change in hcxdumptool.c and now I get this output
```
kali@kali:~/hcxdumptool$ sudo hcxdumptool -i wlan1 --enable_status=19 --gpio_button=4 gpio_statusled=17
initialization...

debug  3f000000

warning: NetworkManager is running with pid 485
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 578
 (possible interfering hcxdumptool)

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1
INTERFACE HARDWARE MAC....: 940c6d896175
DRIVER....................: ath9k_htc
DRIVER VERSION............: 4.19.118-Re4son-v7+
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 18421d7f2343 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 18421d7f2344 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 18421d7f2345 (incremented on every new client)
ROGUE (CLIENT)............: b025aa7426ff
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 64539
ANONCE....................: 287f33bad44643f069753c780c2894586c954fd6cd831bf6dc157a72bf252d90
SNONCE....................: d958f7b67edf7e6567100d5a329f923c571fcf249c1274046c988d069ad3fc1a

terminating...
```

Czechball, 2020-07-24:
My new circuit looks like this:
[image]
(red = 3.3v, black = ground, yellow = 7 (gpio 4), orange = 11 (gpio 17)

ZerBea, 2020-07-24:
Great. The RPI 3 is detected and teh GPIO is initialized as expceted.

warning: NetworkManager is running with pid 485
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 578
 (possible interfering hcxdumptool)

Both services must be stopped.

But there is still something wrong with the wiring.

Please try without --gpio_button=4 and check if the LED is flashsing.
Also please use --enable_status=95 to get more warnings and status messages.

Czechball, 2020-07-24:
NetworkManager and wpa_supplicant aren't interfering, I sucesfully used hcxdumptool before with both of these running and no issued occured.
However, I stopped wpa_supplicant service and marked wlan1 as unmanaged in nmcli.

Test without button:
```
kali@kali:~/hcxdumptool$ sudo hcxdumptool -i wlan1 --enable_status=95 gpio_statusled=17
initialization...
warning: NetworkManager is running with pid 485
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1613
 (possible interfering hcxdumptool)

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1
INTERFACE HARDWARE MAC....: 940c6d896175
DRIVER....................: ath9k_htc
DRIVER VERSION............: 4.19.118-Re4son-v7+
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 00a0546e6b08 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 00a0546e6b09 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 00a0546e6b0a (incremented on every new client)
ROGUE (CLIENT)............: b025aa186414
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 63025
ANONCE....................: 2a2883f393fa281c32584b8f848b7652ea6a1fb3a0b2e997c9fce44407cda29d
SNONCE....................: 174aa5c442803c4ff96174aff964d3a0f9ee5eb22e48cf1cef158c694ffc9f0e

... normal hcxdumptool activity
```
LED is NOT blinking.

Test with button:
```
kali@kali:~/hcxdumptool$ sudo hcxdumptool -i wlan1 --enable_status=95 --gpio_button=4 gpio_statusled=17
initialization...

debug  3f000000

warning: NetworkManager is running with pid 485
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1613
 (possible interfering hcxdumptool)

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1
INTERFACE HARDWARE MAC....: 940c6d896175
DRIVER....................: ath9k_htc
DRIVER VERSION............: 4.19.118-Re4son-v7+
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 005079411834 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 005079411835 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 005079411836 (incremented on every new client)
ROGUE (CLIENT)............: dc70147d8cd5
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62787
ANONCE....................: 3b19d3e8d2a3026fa3255f446b3b4d15bee434eb6f226bb80c7d35983ab8a010
SNONCE....................: 72dc85d8211cda47f9eed6adf8984c08ae93ce868f71920f5e6bc1a053084ede

terminating...
```
ACT LED on Rpi blinks once

ZerBea, 2020-07-24:
To stop the services, was only a notice. That doesn't affect the GPIO.
Let's double check that the besemamory is ok:
$ xxd -c 4 -g 4 /proc/device-tree/soc/ranges

should show something like this:
00000000: 7e000000 ~...
00000004: 3f000000 ?..
00000008: 02000000 ....
...

where this
3f000000
is the base mamory.

ZerBea, 2020-07-24:
Please check also the value of the pull down resistors (should be 330 Ohm and 10KOhm).

ZerBea, 2020-07-24:
Additional you can use wiringpi to verify that the pus butto is really working:
https://raspberrypihq.com/use-a-push-button-with-raspberry-pi-gpio/
http://wiringpi.com/examples/quick2wire-and-wiringpi/install-and-testing/

And you can use the "blink" program to test your LED.

Czechball, 2020-07-24:
``xxd -c 4 -g 4 /proc/device-tree/soc/ranges`` output:
```
00000000: 7e000000  ~...
00000004: 3f000000  ?...
00000008: 01000000  ....
0000000c: 40000000  @...
00000010: 40000000  @...
00000014: 00001000  ....
```
I checked resistor values with my multimeter and confirmed that they're correct.
While probing the circuit, I noticed that the button was in a wrong rotation so it wasn't working - I had to rotate it 90 degrees clockwise so the pins are actually working... 

After this fix, it seems that the led and button are indeed working correctly. At the end, it turned out that the only problem here was my wiring and not hcxdumptool. Thank you very much for all your help and time.

closed 2020-07-24 by Czechball

ZerBea, 2020-07-24:
No problem. Wiring the circuit is a little bit tricky. After assembling a new system, I always had to double check the circuit. And every time I checked it, I found some wiring issues (due to finger trouble).

ZerBea, 2020-07-24:
Push button and LED are not hard coded, so you can use every GPIO pin you like.

Czechball, 2020-07-24:
Yeah, I'm aware of that. I'll see if I can design a case for Rpi3b+ that will also house an RTC and the whole setup will run both Kismet on internal WiFi and hcxdumptool on external.

ZerBea, 2020-07-25:
Correct. Additional hcxdumptool has a big advantage. Neither kismet, nor aircrack-ng nor Wireshark/tshark provide this:
hcxdumptool request missing packets from the target! So we have no(!) packet loss.