ZerBea/hcxdumptool #130: Allow server socket to send messages to 127.0.0.1 without MCAST

DaniloNC, 2020-07-30:
I am writing a python script to parse the status_messages sent to port 60123 and display some information on an Adafruit miniPiTFT.

[image]

Everything is working if I have wlan0 from my RPi0w connected to the wifi. the problem is when I lose the connection or when I try to start hcxdumptool in server mode without any connectivity.

```
...
server socket failed: Network is unreachable.

terminating...
...
```

Please create an option allowing to send the traffic to the loopback IP or create a configuration allowing to change the IP to something other than the Multicast 224.0.0.255

DaniloNC, 2020-07-30:
Another way to trigger the problem is trying to use hcxdumptool as client when all interfaces are down (but the one hcxdumptool is using as monitor interface)

```
➜  ~ hcxdumptool --enable_status=256
initialization...
setsockopt() IP_ADD_MEMBERSHIP failed: No such device
waiting for hcxdumptool server...
```

ZerBea, 2020-07-30:
Please try latest git head (new option --ip)

DaniloNC, 2020-07-30:
It is working great, thanks!
I will upload the code to the GitHub once I clean-up the code.

closed 2020-07-30 by DaniloNC

ZerBea, 2020-08-02:
Nice to hear that. Allowing local host seems to be a good idea.
Cheers
Mike