ZerBea/hcxdumptool #131: Error make install

sinx002, 2020-07-31:
Install: 'hcxdumptool' and '/usr/local/bin/hcxdumptool' are the same archiver
Make *** [Makefile: 41: install] Error 1

but when I install in the home directory it installs without problems

sorry for my english if something went wrong

ZerBea, 2020-08-02:
My english isn't good, too - so no problem, here.
But I need some more information about the Linux distribution:
$ uname -r
Also make sure, you have the correct date/time:
$ date
if not, set date and time.

Frist we can try the following procedure:
remove hcxdumptool directory complete.
Now do a fresh git clone:
$ git clone https://github.com/ZerBea/hcxdumptool
output should look like this:
```
Cloning into 'hcxdumptool'...
remote: Enumerating objects: 80, done.
remote: Counting objects: 100% (80/80), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 2620 (delta 48), reused 56 (delta 29), pack-reused 2540
Receiving objects: 100% (2620/2620), 966.76 KiB | 407.00 KiB/s, done.
Resolving deltas: 100% (1781/1781), done.
```
cd into hcxdumptool directory:
$ cd hcxdumptool
now run make
$ make
output should look like this:
```
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
```
than install:
$ sudo make install
```
output should look like this:
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~
```
test if hcxdumptool is installed:
$ hcxdumptool -v
output should look like this
`hcxdumptool 6.1.0-3-gb47fbe9 (C) 2020 ZeroBeat
`

sinx002, 2020-08-03:
Okay

the exit to uname -r is below

5.6.0-kali1-686-pae

regarding the time it is correct, without errors

I removed the folder and clone the hcxdumptool again

the make command came out correct but the make install continues with the same error

ZerBea, 2020-08-03:
Ok. Some information about this ERROR:
`Make *** [Makefile: 41: install] Error 1`

In line 41 make tried to install hcxdumptool in  /usr/local/bin/:
`install $(INSTFLAGS) hcxdumptool $(INSTALLDIR)/hcxdumptool
`
That is not possible and make returned with Error 1

That is new in KALI:
https://securityboulevard.com/2020/02/kali-2020-1-default-username-password-kali-kali/

Do you have the permissions to install programs?
If you don't have the permissions, you're running into an issue when trying "make install":
```
$ make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install: cannot remove '/usr/local/bin/hcxpioff': Permission denied
make: *** [Makefile:40: install] Error 1
```
Now the same procedure, running sudo:
```
$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~
```
hcxdumptool is installed as expected.

As an alternative you can install hcxdumptool as super user:
```
$ su
Password: 
[root@tux1 git_hcxdumptool]# make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.0-3-gb47fbe9\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~
```
hcxdumptool is installed as expected, too

I closed this issue because it is related to KALI and missing privileges, not to hcxdumptool.
But you can still ask questions here.

closed 2020-08-03 by ZerBea

sinx002, 2020-08-03:
Whenever I open my terminal I give it a sudo -i and put my password, which is the same as sudo su

But I noticed that your permission error was denied in the installation and mine is giving the following message

[image]

My terminal is in Portuguese but the meaning is that hcxdumptool and the hcxdumptool that is inside the hcxdumptool folder are the same file

[image]

I took a picture where you can repair the file that it claims a conflict

But what I find stranger is that if I install it in another folder it will go without error as you can see in the image below

[image]

I don't know if it helps but I installed hcxtools in usr / local / bin and it was without error, only in hcxdumptool that gave an error

ZerBea, 2020-08-03:
Sem problemas, eu entendo um pouco de portugues.

Your complete installation procedure is wrong.
Your source code must be located in home directory and not in /usr/local/bin - that is important.
You have created a directory in /usr/local/bin/ ( /usr/local/bin/ hcxdumptool).
That will not work, because the directory name is the same as hcxdumptool.

Let's start at the beginning:
remove all hcxdumptool related files and directories complete.
cd into your home directory:
$ cd /home
$ git clone https://github.com/ZerBea/hcxdumptool
$ cd hcxdumptool
$ make
$ sudo make install

ZerBea, 2020-08-04:
Before running Linux, especially KALI, some basic knowledge about Linux is mandatory:
Where to store source code, where to store files, where to store binaries:
https://www.howtogeek.com/117435/htg-explains-the-linux-directory-structure-explained/
How to install and configure KALI:
https://www.tutorialspoint.com/kali_linux/kali_linux_installation_and_configuration.htm
How Linux works:
https://linuxnewbieguide.org/

For a Linux beginner is KALI not(!) the best choice! It is designed to be used by experienced penteration testers with Linux knowledge.

sinx002, 2020-08-04:
thank you very much for the help but it didn't work.

 I do not know what happened because it is not the first time that I install the application in this directory, and the other times it worked

I managed to solve it in another way, installed it in another directory and gave a mv in all folders associated with hcxdumptool to / usr / local / bin, and only after that it worked

but thank you very much for your help

ZerBea, 2020-08-04:
You cloned the git source tree into the binary directory: /usr/local/bin/hcxdumptool
That caused that make failed because a directory with the same name exists.
make doesn't have the permission to overwrite this directory and failed.

Adeus
Mike

SuryaN03, 2022-12-01:
> Install: 'hcxdumptool' and '/usr/local/bin/hcxdumptool' are the same archiver Make *** [Makefile: 41: install] Error 1
> 
> but when I install in the home directory it installs without problems
> 
> sorry for my english if something went wrong

You forgot something.
and you have to do this installation, before doing "make && sudo make install" . commond  :
**sudo apt-get install -y  libcurl4-openssl-dev libssl-dev pkg-config**