ZerBea/hcxdumptool #132: Hcxdumptool starts but wont capture anything

HappyCampero76, 2020-08-05:
Hi guys a month ago more or less i installed hcxdumptool with all the libs required and hcxpcaptool "NOT hcxtools,but i can remember where i found hcxpcaptool",i also installed john-the-ripper,macchanger,pyrit  "the last 2 are needed by wifite",and it was working just fine capturing.messed up a lil the OS and i had to clean up and reinstall pretty much everything i had before.
now hcxdumptool starts recognize the wifi device and all the ports on which the tool scans,but since i reinstalled it wont capture a thing after i launched the command or give me the following:
putting wlan in monitor mode:
# airmon-ng start wilp1s0

Found 4 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

    PID Name
    724 avahi-daemon
    729 NetworkManager
    772 wpa_supplicant
    791 avahi-daemon

Requested device "wilp1s0" does not exist.
Run /usr/sbin/airmon-ng without any arguments to see available interfaces.

# hcxdumptool -i wlp1so --enable_status --filterlist=targets.lst --filtermode=2 -o alpha.cap
initialization...
warning: NetworkManager is running with pid 729
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 772
 (possible interfering hcxdumptool)
failed to backup current interface flags, ioctl(SIOCGIFFLAGS) not supported by driver: No such device
warning: failed to init socket
try to use iw to set monitor mode
try to use ip link to bring interface up

terminating...
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device

recognize the interface:
# hcxdumptool -I
wlan interfaces:
b8868748b6a5 wlp1s0mon (ath9k)  warning: probably a monitor interface!

recognize all the channel:
# hcxdumptool -i wlp1s0mon -C
initialization...
warning: NetworkManager is running with pid 729
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 772
 (possible interfering hcxdumptool)
warning: wlp1s0mon is probably a virtual monitor interface
interface is already in monitor mode
available channels:
  1 / 2412MHz (15 dBm)
  2 / 2417MHz (15 dBm)
  3 / 2422MHz (15 dBm)
  4 / 2427MHz (15 dBm)
  5 / 2432MHz (15 dBm)
  6 / 2437MHz (15 dBm)
  7 / 2442MHz (15 dBm)
  8 / 2447MHz (15 dBm)
  9 / 2452MHz (15 dBm)
 10 / 2457MHz (15 dBm)
 11 / 2462MHz (15 dBm)
 12 / 2467MHz (16 dBm)
 13 / 2472MHz (16 dBm)
 14 / 2484MHz (18 dBm)

terminating...

wireless adapter:
# lspci | grep -i wireless
01:00.0 Network controller: Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)

what am i missing to install or what can i do to get hcxdumptool work again?
many thanks in advance.

ZerBea, 2020-08-05:
It is not a hcxdumptool issue:
hcxdumptool isn't working, because the driver is completely broken:
https://bugzilla.kernel.org/show_bug.cgi?id=208251
https://lkml.org/lkml/2020/6/25/13
https://labs.parabola.nu/issues/2824
airmon-ng tols you that, too:
```
Requested device "wilp1s0" does not exist.
Run /usr/sbin/airmon-ng without any arguments to see available interfaces.
```

BTW:
Do not use airmon-ng in combination with hcxdumptool! airmon-ng is designed to run with aircrack-ng suite.
hcxdumptool is running an own monitor mode. You only need to stop all services that take access to the device.

You're running old versions of hcxtools and hcxdumptool.
hcxpcaptool is outdated and will no work with hcxdumptool v6
Please update hcxdumptool and hcxtools to latest version 6.1.0 or latest git head.

Update your OS to receive the Kernel fix.

Closed, because it is not an hcxdumptool issue, but you can ask your questions here.

closed 2020-08-05 by ZerBea

ZerBea, 2020-08-05:
I'm interested in the version of hcxtools/hcxdumptool and OS, you're running.
Please comment output of
$ uname -r
$ hcxdumptool -v
$ hcxpcangtool -v

ZerBea, 2020-08-05:
In addition to the mentioned driver bug, there are several more bugs:
https://bugzilla.kernel.org/buglist.cgi?bug_status=NEW&bug_status=ASSIGNED&bug_status=NEEDINFO&bug_status=REOPENED&field0-0-0=product&field0-0-1=component&field0-0-2=alias&field0-0-3=short_desc&field0-0-4=status_whiteboard&field0-0-5=content&no_redirect=1&order=changeddate%20DESC%2Cbug_status%2Cpriority%2Cassigned_to%2Cbug_id&query_format=advanced&type0-0-0=substring&type0-0-1=substring&type0-0-2=substring&type0-0-3=substring&type0-0-4=substring&type0-0-5=matches&value0-0-0=ath9k&value0-0-1=ath9k&value0-0-2=ath9k&value0-0-3=ath9k&value0-0-4=ath9k&value0-0-5=%22ath9k%22

HappyCampero76, 2020-08-05:
> I'm interested in the version of hcxtools/hcxdumptool and OS, you're running.
> Please comment output of
> $ uname -r
> $ hcxdumptool -v
> $ hcxpcangtool -v

# uname -r
5.4.0-42-generic
# hcxdumptool -v
hcxdumptool 6.1.0-3-gb47fbe9 (C) 2020 ZeroBeat
# hcxpcapngtool -v
hcxpcapngtool 6.1.0-1-g2a36e2b (C) 2020 ZeroBeat

OS version BackBox 7.0 released 2 3 months ago..no sure

# airmon-ng

PHY	Interface	Driver		Chipset

phy0	wlp1s0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)

well i believe that all are up to date since every times i installed it i took the commands to install hcxdumptool and hcxtools on the page you posted along with all the libraries you posted along.
thanks for the help and i will check it out in the hope of fixing my issue.

ZerBea, 2020-08-05:
Your command line will disable status messages:
$ hcxdumptool -i wlp1so --enable_status --filterlist=targets.lst --filtermode=2 -o alpha.cap
enable_status with no value is the same as --enable_status=0
```
--enable_status=<digit>            : enable real-time display (waterfall)
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
                                     bitmask:
                                        0: no status (default)
                                        1: EAP and EAPOL
                                        2: ASSOCIATION and REASSOCIATION
                                        4: AUTHENTICATION
                                        8: BEACON and PROBERESPONSE
                                       16: ROGUE AP
                                       32: GPS (once a minute)
                                       64: internal status (once a minute)
                                      128: run as server
                                      256: run as client
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAP and EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```

filterlist is missing for what kind of target it will be used:
```
--filterlist_ap=<file>             : ACCESS POINT MAC filter list
                                     format: 112233445566, 11:22:33:44:55:66, 11-22-33-44-55-66 # comment
                                     maximum entries 256
                                     run first --do_rcascan to retrieve information about the target
--filterlist_client=<file>         : CLIENT MAC filter list
                                     format: 112233445566, 11:22:33:44:55:66, 11-22-33-44-55-66 # comment
                                     maximum entries 256
                                     due to MAC randomization of the CLIENT, it does not always work!
```
If kind of target is not added (--filterlist), the list will be ignored:
```
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: attack
```

ZerBea, 2020-08-05:
But anyway, thanks for your issue report, because I found an issue in command line handling. I fixed it and pushed an update.
Now your "wrong command line option" will cause an ERROR message:
```
$ hcxdumptool -i wlp39s0f3u1u1u2 --enable_status --filterlist_ap=targets.lst --filtermode=2 -o alpha.pccapng
status must be a digit
```

ZerBea, 2020-08-05:
The issue should be fixed on latest Linux kernels:
https://bugzilla.kernel.org/show_bug.cgi?id=208251#c30
Please update your OS to kernel >= 5.4.54.

The command line switch issue is fixed by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/c7d484eeba00c8ddc1c54dd026366f84d28eec81
Please update to latest git head.

After the updates are finished, everything should work as expected.

HappyCampero76, 2020-08-05:
about the kernel i installed the ukuu but unfortuntely there is none of the kernels listed on https://bugzilla.kernel.org/show_bug.cgi?id=208251#c30.
about the fix on hcxdumptool i will update now,but i was curious will it be best to install those kind of application directly to the home root????
coz i installed them on my home...
about the command i've show before to run hcxdumptool,i found it on a site,and up until i reinstalled the whole system were working just fine hcxdumptool..

1st set of command of which i ran hcxdumptool:
https://www.shellvoide.com/wifi/how-to-crack-wpa2-password-without-handshake-newly-discovered-method/

2nd set of command which i ran hcxdumptool:
https://codibyte.com/blog/2018/11/28/cracking-wpa2-passwords-using-the-new-pmkid-hashcat-attack/

now i am not sure which at that time was the right command to run,but both sets were working just fine at that time,
whom wrote the articles may have been wrong or not,a fact stand still i am a noob on using hacking tools and linux...
not that noob but i am still at the beginning..
anyway thanks for your time and the tips about the kernel,very appreciated..
sure is when i find the kernel you mentioned and being able to install it..
i will for sure leave wifite and other  tools to use mostly hcxdumptool,for what i remember was working just fine on capturing almost all the hanshake i could find in range

ZerBea, 2020-08-05:
Both instructions from the your links are from 2018 and completely outdated. They may work on an old version of hcxdumptool/hcxtools, but not on the last one. Please check --help or -h for correct syntax.

BTW:
What do you mean here:
`about the fix on hcxdumptool i will update now,but i was curious will it be best to install those kind of application directly to the home root????`

You can run them in your home directory (make and run with  ./hcxdumptool...)
or you can install them (make, sudo make install, and run them with hcxdumptool

Default install directory by make (sudo make install) for hcxdumptool and hcxtools is /usr/local/bin.

As an alternative, hcxtools and hcxdumptool are availbale by your default package manager.

Everything else will break your system! This also applies to hashcat and all other tools cloned from git.

Here is a good (negative) example, what happened if you clone a git repository to binary file folders:
https://github.com/ZerBea/hcxdumptool/issues/131

ZerBea, 2020-08-05:
BTW:
No need to leave wifite, because we are working together:
https://github.com/kimocoder/wifite2
Please read more here:
https://github.com/kimocoder/wifite2/issues/1

ZerBea, 2020-08-05:
As far as I know, BackBox 7.0 is based on Ubuntu 20.04 LTS. Running sudo apt-get update followed by sudo-get dist-upgrade should update your system to latest available Ubuntu kernel.

HappyCampero76, 2020-08-05:
> Both instructions from the your links are from 2018 and completely outdated. They may work on an old version of hcxdumptool/hcxtools, but not on the last one. Please check --help or -h for correct syntax.
> 
> BTW:
> What do you mean here:
> `about the fix on hcxdumptool i will update now,but i was curious will it be best to install those kind of application directly to the home root????`
> 
> You can run them in your home directory (make and run with ./hcxdumptool...)
> or you can install them (make, sudo make install, and run them with hcxdumptool)
> 
> Default install directory by make (sudo make install) for hcxdumptool and hcxtools is /usr/local/bin.
> 
> As an alternative, hcxtools and hcxdumptool are available by your default package manager.
> 
> Everything else will break your system! This also applies to hashcat and all other tools cloned from git.
> 
> Here is a good (negative) example, what happened if you clone a git repository to binary file folders:
> #131

i will install the update now....
about what you said about cloning i guess i'll download it and leave the cloning the hell alone...
whats the difference between installing  an application in my home folder from installing an application in the home root folder?
if there is any difference at all...
about what version is backbox based on ubuntu i have no idea and i actually don't really care,but still i found and installed the suggested kernel 5.4.54ran the command apt-get dist-upgrade and for what i could see finished the installation of the kernel included the driver r8168 which i presume is the driver needed to allow hcxdumptool to run...
now going to install the hcxdumptool update..in the hope my problem is fixed

ZerBea, 2020-08-05:
Copying/installing a tool in $HOME makes it available only for the current user.
Installing it to /usr/bin (/usr/sbin) or /usr/local/bin (/usr/local/sbin) makes it available for every user on the system.

r8168 is the Realtek Ethernet driver that make your LAN working.
hcxdumptool use wireless drivers.
This one is yours:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/ath/ath9k?h=v5.4.54

ZerBea, 2020-08-05:
If everything is fine, output should look like this:
```
$ uname -r
5.7.12-arch1-1

$ hcxdumptool -I
wlan interfaces:
f81a67077d0e wlp39s0f3u1u1u2 (ath9k_htc)

$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 125 to 64 

terminating...
```
That should show you networks nearby:
$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --do_rcascan

That should start your attack (with full status display):
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --enable_status=95 --filterlist_ap=targets.lst --filtermode=2 -o alpha.pccapng
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u1u2
INTERFACE HARDWARE MAC....: f81a67077d0e
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.7.12-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 4 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: attack
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 000f09e12dcd (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 000f09e12dce (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 000f09e12dcf (incremented on every new client)
ROGUE (CLIENT)............: acde48bb144f
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62471
ANONCE....................: 137decc6bd69d10714b2fe14acd62a0a06cd7ba60562c131e2d18b36cabc0bb1
SNONCE....................: 88117334f0a832daf97c27d4e3847dbc8872143910d6e8d56461e6b78df4c690

14:40:44   5 18421d0cae03 000f09e12dcf TEST_NETWORK [ROGUE PROBERESPONSE]
```

HappyCampero76, 2020-08-05:
well i had a small problem installing the firmware from command line,so i decided to attempt and find the firmware on synaptic...
well i know many don't use synaptic anymore but i do,so i found the firmware and installed it on synaptic,rebooted and worked like a charm!!.
now i know if something is wrong with hacking tools i have to go and find firmware.
thank you for your help.
since i have a second wifi but its usb and won't work with hcxdumptool,later i'll attempting to find the firmware.
OUT OF TOPIC help,i can remember that in backbox 4 or 5 there was a script that allowed me to start stop restart network manager...i wonder is there anybody that can point me where to find that kind of script?
as i said earlyer i am slightly better than a noob not sure if i can make such a script...
once again thanks for the help with hcxdumptool and for any help that will come for the network manager script.

ZerBea, 2020-08-05:
synaptic is a fine gui for apt and working as expected.

stop network script:
```
#!/bin/sh

sudo systemctl stop NetworkManager.service
sudo systemctl stop wpa_supplicant.service
```

start network script:
```
#!/bin/sh

sudo systemctl start NetworkManager.service
sudo systemctl start wpa_supplicant.service
```