ZerBea/hcxdumptool #133: Signal strength for surrounding wifis

RaduNico, 2020-08-06:
Is there a way to display the signal strength for surrounding wifis during do_rcascan?

I am currently trying to make a tool similar to wifite to automate attacks on wifis and I am trying to remove the need for aircrack (in this case airodump-ng) as much as possible.

ZerBea, 2020-08-06:
No. And it is not planed to add this, because RSSI is a relative, one way value calculated by the firmware.
We are packet oriented. The important value is the HIT count.
count = packets received from the AP
hit = responded packets from the AP
This works in both directions (hcxdumptool -> AP, AP -> hcxdumptool) while RSSI shows only one direction (AP -> airodump-ng).
In other words:
We can have a good RSSI of -70dBm, but the AP doesn't hear us.
We can have a poor RSSI of -90dBm and the AP hear us perfect.

Read more here:
https://www.metageek.com/training/resources/understanding-rssi.html
https://en.wikipedia.org/wiki/Fresnel_zone
https://en.wikipedia.org/wiki/Reflection_phase_change

If you really need RSSI you can use tshark on hcxdumptool's pcapng file (hcxdumptool is compatibel to tshark and Wireshark.
):
`$ tshark -r dump.pcapng -T fields -e wlan.sa -e wlan_radio.signal_dbm
`

RaduNico, 2020-08-06:
Thank you for taking your time to explain your perspective and for the alternative solution.
This solves my issue

closed 2020-08-06 by RaduNico

ZerBea, 2020-08-06:
airodump-ng is a passive dumper. From that point it is fine to sort by the highest RSSI.
hcxdumptool is active.  If we don't reach the target, the attack is useless.
You can calculate the ratio counts/hits to get an average value. hcxdumptool --check_injection is doing that, too
```
$ sudo hcxdumptool -i wlp3s0f0u2 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 286 to 153 

terminating...
```
Round about 46,5% packet loss. Not so bad, because hcxdumptool is able to request lost packets.

ZerBea, 2021-01-03:
By latest commit, I added some new features to rcascan.
Now the ACCESS POINT with the highest hit rate (PROBERESPONSE) is on top of the ranking:
https://github.com/ZerBea/hcxdumptool/commit/789110116cdc42d0845baa021752422e3d5bd45a

A new option (rcascan_max) will limit the status of rcascan to n  ACCESS POINTs with the highest hit rate:
https://github.com/ZerBea/hcxdumptool/commit/e1bacb9b6aed8029ea3830d8c268e651989450b6

ZerBea, 2021-01-16:
I decided to add RSSI information to rcascan output:
```
 BSSID          CH RSSI COUNT   HIT ESSID           injection ratio:  45% [18:11:28]
-------------------------------------------------------------------------------------
```
Sometimes you'll see that you have a low RSSI but many HITs or a high RSSI and less HITs.
I strongly recommend to direct the antenna by HITs, because you can't trust RSSI.