ZerBea/hcxdumptool #135: bash_profile & root privileges

adrastee, 2020-08-18:
`this program requires root privileges`
when rebooting my RPi with your [bash_profile](https://github.com/ZerBea/hcxdumptool/blob/master/usefulscripts/bash_profile)  file

Should you add 'sudo' before hcxdumptool?

ZerBea, 2020-08-18:
That depend on your configuration. The bash_profile is designed to run on a Raspberry Pi, running Arch Linux with no(!) user configured. That makes life easier on a headless system and you don't need sudo. 
`--filterlist_ap=~/filter.txt --filterlist_client=~/filter.txt`
will not work as expected, because you use the same list (filtet.txt) for APs and CLIENTs. You need 2 different lists. One with the MACs of APs and one with the MACs of  CLIENTS.
```
failed to open filter list ~/filter.txt
failed to open filter list ~/filter.txt
```
means that the filter lists are not in the folder (HOME directory of the current user) where hcxdumptool expect to load them.
example for my configuration:
filterlistap.txt is in $HOME:
$ pwd
/home

$ ls -all
drwxr-xr-x  2 root  100 4096 18. Aug 15:56 .
drwxr-xr-x 17 root     root 4096 29. Mai 16:17 ..
-rw-r--r--  1 root     root  248 25. Jul 15:20 filterlistap
-rw-r--r--  1 root     root  153 25. Jul 15:20 filterlistclient

what is your output of pwd and ls -all?

If you want to run hcxdumptool as background task, just add & to your commandline
to stop hcxdumptool use
$ kill hcxdumptool.

ZerBea, 2020-08-18:
If you added a user and you run hcxdumptool as this user you must modify your bash_profile.
Additional you must use sudo to allow hcxdumptool tu use the interface.

closed 2020-08-18 by ZerBea

ZerBea, 2020-08-18:
Why haven't you installed hcxdumptool into its default folder: /user/local/bin
$ make
$ make install
Instead you are using ~/hcxdumptool/hcxdumptool
You store your pcapng file in ~/hcxdumptool/
but you try to read you filter lists from $HOME (~/) of you current user.
That and the combination of your "User and Group Management" will lead you into ugly issues. You may have noticed that.
I strongly recommend to clean up your system and to install hcxdumptool as recommended and not to use it from a user folder.

If you want to start hcxdumptool during boot from a user account without typing a password, read more here:
https://www.tecmint.com/run-sudo-command-without-password-linux/

adrastee, 2020-08-19:
Thanks for your kind help.

- I replaced ~/filter.txt  to /home/pi/filter.txt and it works.
- hcxdumptool is now in user/local/bin
- I will have a look at users mangement and autoruns of sudo program for Raspbian.

ZerBea, 2020-08-19:
Nice to hear that.