ZerBea/hcxdumptool #137: Unable to attack particular AP providers.

TheGuyWithTheIssue, 2020-09-11:
Hello ZeroBeat,

I hope you are well. First, I want to thank you for your work, it is outstanding tool, and has very high accuracy in terms of attack and capture. I have hacked almost all neighbourhood with it. Thank you so much for coding it, I am python scripter but this is another level of tool.

Using hcxdumptool I have been largely successful but with only a limited type of APs. I live in the United Kingdom.
The Access Points I have been able to attack with 90% accuracy (and crack the hash with hashcat) are the Sky routers. I have obtained PMKID's from Talk Talk provider too, but haven't worked on them with custom masks.

I wanted to ask, if you have time, why I have been unsuccessful in capturing PMKID from BTHub and Virgin Media ISP.

The exchanges are taking place as such:

00:13:35   6 6231451a2d98 e45740cd9090 VM6573660 [EAPOL:M3M4ZEROED EAPOLTIME:14706 RC:2 KDV:2]
00:13:35   6 b05cdab43ecc e45740cd9090 VM6573660 [EAPOL:M2M3 EAPOLTIME:10449 RC:2 KDV:2]
00:13:35   6 b05cdab43ecc e45740cd9090 VM6573660 [EAPOL:M3M4ZEROED EAPOLTIME:5077 RC:2 KDV:2]
00:13:55   6 b05cdab43ecc e45740cd9090 VM6573660 [AUTHENTICATION]
00:14:12   6 6231451a2d98 e45740cd9090 VM6573660 [EAPOL:M1M2 EAPOLTIME:16100 RC:1 KDV:2]
00:14:12   6 6231451a2d98 e45740cd9090 VM6573660 [EAPOL:M2M3 EAPOLTIME:12791 RC:2 KDV:2]
00:14:12   6 6231451a2d98 e45740cd9090 VM6573660 [EAPOL:M3M4ZEROED EAPOLTIME:10624 RC:2 KDV:2]

However, I have not had success with obtaining a valid Identifier. Is this because:

1) These routers do not exchange information like other routers, and authentication processes are different.
2) I am too far from the APs and the attack is unable to complete efficiently.
3) Other obscure reasons, such as driver incompatibility or some feature not working (I don't think so).

I use Kali operating system (yes, for everyday use, browsing, I even run Plex Server on it) and it appears everything is working apart from only some models. My monitor interface is ALFA Network AWUS036NH. 

Thank you again, I hope to get an answer.

ZerBea, 2020-09-12:
> These routers do not exchange information like other routers, and authentication processes are different.
BTHub3-xxxx, BTHub4-xxxx, BTHub5-xxxx, BTHub3-xxxx router transmit a PMKID and you should receive it (PMKIDROQUE)
VMxxxxxxx usually don't transmit a PMKID (except repeater systems).
Additional you should receive some M1M2ROGUE on all captures.
The authentication process is not different.

> I am too far from the APs and the attack is unable to complete efficiently.
Just run
$ sudo hcxdumptool -i interface --do_rcascan
and check the HIT count. It should increase on APs in range. If you got no HITs, either no AP is in range or packet injection is not working.
output should look like this:
```
$ sudo hcxdumptool -i wlp5s0f4u2 --do_rcascan
 BSSID         CH COUNT   HIT ESSID                 [08:08:31]
---------------------------------------------------------------
140708b68f9a  11    14    11 TEST_NETWORK
```

> Other obscure reasons, such as driver incompatibility or some feature not working (I don't think so).
Do a driver test and an injection test:
$ sudo  hcxdumptool -i interface --check-driver
output should look like this:
```
$ sudo hcxdumptool -i wlp5s0f4u2 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

$ sudo hcxdumptool -i interface --check-injection
output should look like this:
```
$ sudo hcxdumptool -i wlp5s0f4u2 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 141 to 94 

terminating...
```
A ratio above 100 to 25 is is workable, but there is room to improve your antenna.
A ratio above 100 to 50 is is standard.
All above - your equipment is nearly perfect.

If one (or both) of the two tests failed or you received a warning, you're running into driver issues and/or your system is misconfigured.

TheGuyWithTheIssue, 2020-09-12:
Thank you for your time. The tests undertaken have all passed with flying colours it seems:

[image]
[image]
[image]

I also have captured a PMKID from BThub-3. I have not been successfull with BT-Hub6 with your application, and I have instead used EAPOL cap (cleaned with wpaclean) from deauthentication attack with aircrack suite, then converted into hccapx with hcap2hccapx utility.

Also you mention that VM does not send PMKID, so I guess it is not possible to attack this router clientless. Is it possible with standard deauth attack? I do not know of antoher way to attack this router, as its a popular ISP here in the UK.
I wonder if you have any advice on this, meanwhile thank you for the detailed reply.

ZerBea, 2020-09-13:
It looks like every thing is fine. I can see at least 4 BT-Hubs in your list. Two of them are out of range (0 hits).
If hcxdumptool requested a PMKID from an AP, you should see PMKIDROGUE.
If it captured the PMKID from an authentication sequence between an ACCESS POINT and a CLIENT you should see PMKID.

BTW:
Do not use wpaclean. It will remove important frames from your dumpfile and has several unfixed issues in timestamp handling and handshake detection.
https://github.com/aircrack-ng/aircrack-ng/issues/1958
https://github.com/aircrack-ng/aircrack-ng/issues/1993
https://github.com/aircrack-ng/aircrack-ng/issues/2079
https://github.com/aircrack-ng/aircrack-ng/issues/2067
There is also an issue in PMKID detection:
https://github.com/aircrack-ng/aircrack-ng/issues/2175

hccapx is deprecated and replaced by new hash mode 22000 which covers old hash mode 16800 and old hash mode 2500:
https://github.com/hashcat/hashcat/issues/1816
https://hashcat.net/forum/thread-6661-post-47931.html#pid47931
The format is no longer binary and you can use simple bash tools or hcxhashtool to prepare them for hashcat.
https://hashcat.net/forum/thread-6661-post-48055.html#pid48055
In every case, I recommend to use hcxpcapngtool to convert hashes for hashcat and JtR.

>Also you mention that VM does not send PMKID, so I guess it is not possible to attack this router clientless. Is it possible with standard deauth attack? I do not know of antoher way to attack this router, as its a popular ISP here in the UK.
I wonder if you have any advice on this, meanwhile thank you for the detailed reply.

A deauthentication attack is working as long as Management Frame Protection (MFP) isn't activated. 
hcxdumptool detect this and use a reassociation attack to get the handshake.
In addition it will attack the CLIENT to request his M2. In that case you should see M1M2ROGUE.
In other words, hcxdumptool will use all attack modes (PMKID, deauthentication, disassociation, M4 fail, reassociation, and M2 request).
To make less noise, an attack is stopped after n trials. We don't want to spam the channel.
You can change that behavior with --stop_ap_attacks=n --resume_ap_attacks=n

example:
```
$ hcxdumptool -i interface -f 79 -o dump.pcapng  --stop_ap_attacks=6000 --resume_ap_attacks=12000 --essidlist=beaconlist --active_beacon
that is a very, very aggressive mode against APs and CLIENTs

$ hcxdumptool -i interface -o dump.pcapng --disable_deauthentication --disable_ap_attacks --active_beacon
that is a less noisy mode against CLIENTs
```
than convert the results:
```
$ hcxpcapngtool -o hash.22000 -E list1 -I list2 -U list3 dump.pcapng
For the first hashcat task try:
$ cat list1 list2 list3 | sort | uniq > list
$ hashcat -m 22000 hash.22000 list
```
Please notice: That will only convert the best hashes.
You can use --all to convert them all!
`$ hcxpcapngtool -o hash.22000 --all -E list1 -I list2 -U list3 dump.pcapng
`
Additional you can select single hashes by hcxhashtool (by ESSID, MAC, ESSID length and more.....)

This will give you an overview of the converted hashes:
`$ hcxhashtool -i hash.22000 --info=stdout
`

TheGuyWithTheIssue, 2020-09-15:
Hello ZeroBeat,

I wanna say.... that was beautiful. Initially had some difficulty with getting any response from the AP, but setting the channels did the trick:

It seemed like there was no response, possibly due to incorrect settings:

[image]

adjusting the channel and with proper bssid list all four EAPOL exchanges were made (ANONCE, SONCE+MIC etc)

[image]

cleaning things up with your commands made it beautiful:

[image]

and everything was written to file for hashcat to work on. 

There are things which I still don't understand, like setting the packet frames to be saved to "79." Maybe there are frames which are needed beyond the "64" value instructed in "hcxdumptool -h". Or maybe it is not necessary and all one really needs to save are EAPOL exchanges, or settings "-f 2".

In the initial attack I inputed the "stop_ap_attacks" and "resume" values as instructed, but I felt that this was not necessary as exchanges were made without "aggressive mode." I also removed the option to transmit beacon frames more frequent.

Of course, there is the slight matter of obtaining valid passkey, which can take days, but everything is very clear on at least how the program works on a superficial level (i.e. options and commands). It is all much simpler than airmon-ng and the captures seems cleaner. I am happy with it.

ZerBea, 2020-09-15:
That is fine, but please do not use airmon-ng to set monitor mode for hcxdumptool.
airmon-ng is designed to run with aircrack-ng tools and attack modes and not with hcxdumptool and its attack modes.
airmon-ng run internally iw and iw will add(!) a virtual monitor interface (by NETLINK messages) named wlanxmon and leave the physical interface untouched.
hcxdumptool need a physical interface (not shared, not virtual) and set an own(!) monitor mode. You don't need airmon-ng to do this and you received a clear warning "wlan0mon is probably a virtual monitor interface". That means, some active high preformance - high speed attack modes will not(!) work on this interface.

Will explain the difference between NETLINK messages and ioctl() system calls:
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls
```
Control: ioctl should be your first choice, unless there’s an overriding reason, due to its immediacy and reliable delivery.
Data: For occasional data passing, ioctl should work fine. For bulk data, and especially if you’re look at asynchronous operation, Netlink is preferred.
```
We control - we don't wan't to share social media cat videos!

f 79 option will dump TCP and UDP traffic, too and hcxpcapngtool can look for additional passwords in the dumpfile. You can remove this option safely (and run defaults - dump all necessary and important frames) if you don't need that feature (usernames from WiFi traffic: hcxpcapngtool -U).

Transmitting BEACONs will force a CLIENT to connect to hcxdumptool. You can receive his M2 without being in range of the CLIENTs ACCESS POINT.

BTW: If an AP is far away (only a few HITs in do_rcascan) you can that channel option (-c) to stay on that channel. You have noticed that this will increase your chances to get PMKID and or EAPOL 4Way handshake.) 

I think we can close this report, because it isn't an issue of hcxdumptool, but you can still ask your questions here.

closed 2020-09-15 by ZerBea

awesomedonald, 2021-06-23:
Ok. I know something has got to be wrong here. Please teach me. 
[image]

ZerBea, 2021-06-23:
Answer is here:
https://github.com/ZerBea/hcxdumptool/issues/178