ZerBea/hcxdumptool #138: PMKIDROUGE

renonet, 2020-09-13:
Hi ZerBea,

I got this from [PMKIDROGUE:55bcf627349fd188abda13fefd997461 KDV:2] hcxdumptool.

What does PMKIDROUGE mean?

I never get PMKID Found message, but only show this [PMKID:a5c839f2b94e044a32d61759c83f2930 KDV:2]. Is it considered as PMKID found?

What different it from PMKIDROUGE?

ZerBea, 2020-09-13:
PMKIDROGUE:
the PMKID is requested by hcxdumptool and not by a CLIENT. You can recover the password from both of them.
M1M2ROGUE :
EAPOL M2 is requested from a CLIENT by hcxdumptool and not from an ACCESS POINT. You can recover a valid password from it.
PMKID:a5c839f2b94e044a32d61759c83f2930 KDV:2
You captured a PMKID requested from a CLIENT. WPA version (Key Descriptor Version) is WPA 2.
So everything is fine.

Closed, because it is not an issue, but you can still ask your questions here.

closed 2020-09-13 by ZerBea

ZerBea, 2020-09-14:
By latest commit:
https://github.com/ZerBea/hcxdumptool/commit/6e752e629389dbed9c39873f7b801df9602fb989
I added an explanation to --help