ZerBea/hcxdumptool #143: Installation make error

tomasaquiles, 2020-10-18:
hi, I'm trying to install on kali linux 2020.3 and when trying the command: "make" I have the following error:

root@kali:/opt/hcxdumptool# make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"
In file included from /usr/include/openssl/bioerr.h:13,
                 from hcxdumptool.c:55:
/usr/include/openssl/macros.h:109:4: error: #error "OPENSSL_API_COMPAT expresses an impossible API compatibility level"
  109 | #  error "OPENSSL_API_COMPAT expresses an impossible API compatibility level"
      |    ^~~~~
In file included from hcxdumptool.c:55:
/usr/include/openssl/bioerr.h:22:5: error: conflicting types for ‘ERR_load_BIO_strings’
   22 | int ERR_load_BIO_strings(void);
      |     ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/openssl/evp.h:75,
                 from hcxdumptool.c:49:
/usr/include/openssl/bio.h:809:6: note: previous declaration of ‘ERR_load_BIO_strings’ was here
  809 | void ERR_load_BIO_strings(void);
      |      ^~~~~~~~~~~~~~~~~~~~
hcxdumptool.c: In function ‘tlsinit’:
hcxdumptool.c:6889:1: warning: implicit declaration of function ‘SSL_CTX_set_max_proto_version’; did you mean ‘SSL_CTX_set_ssl_version’? [-Wimplicit-function-declaration]
 6889 | SSL_CTX_set_max_proto_version(tlsctx, TLS1_2_VERSION);
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      | SSL_CTX_set_ssl_version
make: *** [Makefile:32: build] Error 1

Any suggestions to help me solve? thanks for your attention

ZerBea, 2020-10-18:
This is a KALI issue:
https://github.com/curl/curl/issues/4097
https://manpages.debian.org/experimental/libssl-doc/SSL_CTX_set_options.3ssl.en.html
```
These options turn off the SSLv3, TLSv1, TLSv1.1, TLSv1.2 or TLSv1.3 protocol versions
with TLS or the DTLSv1, DTLSv1.2 versions with DTLS, respectively.
As of OpenSSL 1.1.0, these options are deprecated, use SSL_CTX_set_min_proto_version(3)
and SSL_CTX_set_max_proto_version(3) instead.
```
maybe due to a broken package management system:
https://github.com/ZerBea/hcxtools/issues/140#issuecomment-706021372

You received a warning that your old version doesn't support this functions:
`warning: implicit declaration of function ‘SSL_CTX_set_max_proto_version’
`
Unfortunately there is nothing I can do, but you can try to update your system in the hope that a newer version will be installed.
https://linuxconfig.org/how-to-update-kali-linux

Compilation an Arch Linux works as expected:
```
$ make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"

$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.3-5-g755405a\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~
```

Closed this issue, because it is related to KALI, but you can still ask your questions here.

BTW:
What is the output of
$ uname -a
and
$ sudo apt update

closed 2020-10-18 by ZerBea

n0w1re, 2020-10-19:
@tomasaquiles Could you please do also a `openssl version -a`

ZerBea, 2020-10-19:
KALI claims to be a rolling release, so the version should be at least:
```
$ openssl version -a
OpenSSL 1.1.1h  22 Sep 2020
built on: Tue Sep 22 14:59:44 2020 UTC
platform: linux-x86_64
options:  bn(64,64) rc4(8x,int) des(int) idea(int) blowfish(ptr) 
compiler: gcc -fPIC -pthread -m64 -Wa,--noexecstack -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wa,--noexecstack -D_FORTIFY_SOURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_CPUID_OBJ -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DKECCAK1600_ASM -DRC4_ASM -DMD5_ASM -DAESNI_ASM -DVPAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DX25519_ASM -DPOLY1305_ASM -DNDEBUG -D_FORTIFY_SOURCE=2
OPENSSLDIR: "/etc/ssl"
ENGINESDIR: "/usr/lib/engines-1.1"
Seeding source: os-specific
```

n0w1re, 2020-10-19:
It's 1.1.1g. Rolling a little behind but it does compile here.

tomasaquiles, 2020-10-19:
I reinstalled kali linux 2020.3 and after installation I installed via apt-get the following packages:
build-essential
python-dev
libpcap-dev
libssl-dev

After this, the compilation works fine.

Thanks to all for atention

ZerBea, 2020-10-20:
Glad to hear this.

BTW;
You don't need to install libpcap-dev, because neither hcxdumptool nor hcxtools nor hcxkeys use this library.