ZerBea/hcxdumptool #145: make error

piirios, 2020-10-23:


piirios, 2020-10-23:
hello, first sorry for my English, i have this error: 

> cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxhashtool.d -o hcxhashtool hcxhashtool.c -lcrypto -lssl -lcurl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
> hcxhashtool.c:19:10: fatal error: curl/curl.h: Aucun fichier ou dossier de ce type
>    19 | #include <curl/curl.h>
>       |          ^~~~~~~~~~~~~
> compilation terminated.
> make: *** [Makefile:101 : hcxhashtool] Erreur 1

i've already upgrade my kali and install with apt-get build-essential, python-dev, libpcap-dev, libssl-dev

ZerBea, 2020-10-23:
No problem, I'm also not an English native speaker.

From README.md:
```
You need to install missing dependencies before running make:
sudo apt-get install libcurl4-openssl-dev
```
it is part of KALI  repository:
http://pkg.kali.org/pkg/curl

and for hcxtools additional:
`zlib and zlib-dev
`
also part of KALI repository:
http://pkg.kali.org/pkg/zlib

Than it should compile.

Closed this issue report, because it is related to KALI, but you can still ask your questions here.

Please read more about the KALI related problems here:
https://github.com/ZerBea/hcxtools/issues/140
https://github.com/ZerBea/hcxtools/issues/153
https://github.com/ZerBea/hcxtools/issues/132
https://github.com/ZerBea/hcxtools/issues/102

BTW:
You don't need libpcap-dev, because hcxdumptool and hcxpcapngtool run an own parser for cap/pcap/pcapng (inclusive gzip compression).

closed 2020-10-23 by ZerBea

ZerBea, 2020-10-23:
BTW:
99% of the KALI issues are related to missing dependencies (KALI don't install them as default), broken packages, a broken system (KALI doesn't upgrade as expected) or a misconfigured system ( or all four in combination).

ZerBea, 2020-10-23:
hcxdumptool:
```
$ make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.3-7-g8787ea6\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.3-7-g8787ea6\" -DVERSION_YEAR=\"2020\"

$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c  -DVERSION_TAG=\"6.1.3-7-g8787ea6\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto -lssl -DVERSION_TAG=\"6.1.3-7-g8787ea6\" -DVERSION_YEAR=\"2020\"
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~
```

and hcxtools:
```
$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxpcapngtool.d -o hcxpcapngtool hcxpcapngtool.c -lz -lcrypto -lssl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxhashtool.d -o hcxhashtool hcxhashtool.c -lcrypto -lssl -lcurl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxpsktool.d -o hcxpsktool hcxpsktool.c -lcrypto -lssl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxwltool.d -o hcxwltool hcxwltool.c   -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxhash2cap.d -o hcxhash2cap hcxhash2cap.c   -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/wlancap2wpasec.d -o wlancap2wpasec wlancap2wpasec.c -lcrypto -lssl -lcurl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/whoismac.d -o whoismac whoismac.c -lcrypto -lssl -lcurl  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxpmkidtool.d -o hcxpmkidtool hcxpmkidtool.c -lcrypto -lssl -lpthread  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxhashcattool.d -o hcxhashcattool hcxhashcattool.c -lcrypto -lssl -lpthread  -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxmactool.d -o hcxmactool hcxmactool.c   -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"
cc -O3 -Wall -Wextra -std=gnu99  -MMD -MF .deps/hcxessidtool.d -o hcxessidtool hcxessidtool.c   -DVERSION_TAG=\"6.1.3-36-g8a6f5c9\" -DVERSION_YEAR=\"2020\"

$ sudo make install
install -D -m 0644 manpages/hcxpcapngtool.1 /usr/local/share/man/man1/hcxpcapngtool.1
install -D -m 0755 hcxpcapngtool /usr/local/bin/hcxpcapngtool
install -D -m 0644 manpages/hcxhashtool.1 /usr/local/share/man/man1/hcxhashtool.1
install -D -m 0755 hcxhashtool /usr/local/bin/hcxhashtool
install -D -m 0644 manpages/hcxpsktool.1 /usr/local/share/man/man1/hcxpsktool.1
install -D -m 0755 hcxpsktool /usr/local/bin/hcxpsktool
install -D -m 0755 hcxwltool /usr/local/bin/hcxwltool
install -D -m 0644 manpages/hcxhash2cap.1 /usr/local/share/man/man1/hcxhash2cap.1
install -D -m 0755 hcxhash2cap /usr/local/bin/hcxhash2cap
install -D -m 0644 manpages/wlancap2wpasec.1 /usr/local/share/man/man1/wlancap2wpasec.1
install -D -m 0755 wlancap2wpasec /usr/local/bin/wlancap2wpasec
install -D -m 0755 whoismac /usr/local/bin/whoismac
install -D -m 0755 hcxpmkidtool /usr/local/bin/hcxpmkidtool
install -D -m 0755 hcxhashcattool /usr/local/bin/hcxhashcattool
install -D -m 0755 hcxmactool /usr/local/bin/hcxmactool
install -D -m 0755 hcxessidtool /usr/local/bin/hcxessidtool
```