ZerBea/hcxdumptool #146: filterlist_ap gets ignored

slothhy, 2020-10-27:
I have added my ap bssid from --do_rcascan to a filterlist.txt and run it with filterlist_ap=filterlist.txt and --filtermode=2. I am still getting results from other bssid in the scan.

ZerBea, 2020-10-27:
That is a wanted behavior. The rca_scan should show unfiltered traffic on a channel to detect interferences between the routers. That include own routers, too.
If you really want to filter them out, you have to use the BPF.
I'll add a small notice about this behavior to --help.

BTW:
You should use old filter style only for testing purpose. Running a BPF is much faster.

ZerBea, 2020-10-27:
Do you need this user space filter to be added to rca_scan?

slothhy, 2020-10-27:
Oh the output that I pasted above is not from the rca_scan. I only used rca_scan to get my router's bssid. I am testing the tool on my router to see if I can get the PMKID.

ZerBea, 2020-10-27:
Ok, now I understand. Please let me explain how hcxdumptool works:

Filter lists are attack and protect lists. it is not a display filter.
Adding an entry to one of the lists, prevent a target to be attacked or a target to be protected.
In every case hcxdumptool is a passive dumper and will show you what's going on - regardless of filter mode and filter list

BPF is a kernel space filter and works as display filter, too.

slothhy, 2020-10-27:
Ok I think I understand now. So running a filterlist will display unwanted results but inside the pcap it will only contain the bssid that I specify?

Let me try out BPF and see how it goes.

ZerBea, 2020-10-27:
Thanks for the issue report. By latest commit, I added this notice to help:
```

-bpfc=<file>                      : input kernel space Berkeley Packet Filter (BPF) code
                                    notice: this is working as a protect, an attack and a display filter

--filtermode=<digit>               : user space filter mode for filter list
                                     notice: running this filter will let hcxdumptool protect or attack a target - it is not a display filter
 ```
That should make it clear.

slothhy, 2020-10-27:
Create BPF to attack a MAC
$ tcpdump -i <interface> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 -ddd > attack.bpf

Do I have to specify 2 MACs or 1 will do?

ZerBea, 2020-10-27:
> Ok I think I understand now. So running a filterlist will display unwanted results but inside the pcap it will only contain the bssid that I specify?

No.
The specified target will be protected or attacked. That is the only affect.
Everything else is not affected.

Addr1 = to
Addr2 = fm
Addr3 = AP
addr4 = additional in WDS systems

$ tcpdump -i wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 -ddd > attack.bpf
Should do the trick...

ZerBea, 2020-10-27:
Please notice:
Running your BPF will prevent that you see and capture undirected proberequests from CLIENTS using MAC randomization.
You will loose this information:
https://github.com/evilsocket/pwnagotchi/issues/835#issuecomment-598597214

slothhy, 2020-10-27:
Would you still recommend running BPF then? I am using to retrieve PMKID. Maybe I should just stick to the filterlist?

ZerBea, 2020-10-27:
I recommend to use the BPF as protection list to protect own APs, CLIENTs.
run hcxdumptool -c of your target - if you don't receive a PMKID after 5 minutes, the AP isn't vulnerable (or out of range)
run hcxpcapngtool (mode 22000)
run hcxhashtool to get your target (or use bash tools)
run hashcat -m 22000 on your target

closed 2020-10-27 by slothhy