ZerBea/hcxdumptool #149: Failed to open Berkeley Packet Filter list

conf11s3d, 2020-11-28:
Hello congratulations for the hard work, I was asking if you can help, some time ago I found your post where you published two methods of attack, when I run them I get this error:

Failed to open Berkeley Packet Filter list own.bpfc

would you be kind to explain to me where I am wrong with the commands ?
below are the commands that fail:

hcxdumptool -i wlan0 -o vector1.pcapng --stop_ap_attacks=6000 --resume_ap_attacks=12000 --bpfc=own.bpfc --active_beacon --tot=10 --enable_status=1

initialization...
failed to open Berkeley Packet Filter list own.bpfc

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: 3a6bb7cc7acd
DRIVER....................: iwlwifi
DRIVER VERSION............: 5.9.0-kali2-amd64
DRIVER FIRMWARE VERSION...: 46.8902351f.0 9000-pu-b0-jf-b0-
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: a468bca17d7c (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: a468bc007d7d (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: a468bca17d7e (incremented on every new client)
ROGUE (CLIENT)............: c02250ba7efd
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 64809
ANONCE....................: 223a55602bbfb2f93a4d8ddeb58cda7325d183a028ec8e907424996f61156738
SNONCE....................: b95555a576beae360d5a3ced5f466b7c2dfa322e60da66c85d0282cefb3c080d

^C
terminating...

hcxdumptool -i wlan0 -o vector2.pcapng --tot=1440 --bpfc=own.bpfc --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,9,6,3,11, -t 3600 --enable_status=1

initialization...
failed to open Berkeley Packet Filter list own.bpfc

Same mistake...!

Thanks for your kind attention.

ZerBea, 2020-11-28:
To use a BERKELEY PACKET FILTER, you have to create the BERKELEY PACKET FILTER CODE first as described in --help:
```
--bpfc=<file>                      : input kernel space Berkeley Packet Filter (BPF) code
                                     affected: incoming and outgoing traffic - that include rca scan
                                     steps to create a BPF (it only has to be done once):
                                      set hcxdumptool monitormode
                                       $ hcxumptool -m <interface>
                                      create BPF to protect a MAC
                                       $ tcpdump -i <interface> not wlan addr1 11:22:33:44:55:66 and not wlan addr2 11:22:33:44:55:66 -ddd > protect.bpf
                                       recommended to protect own devices
                                      or create BPF to attack a MAC
                                       $ tcpdump -i <interface> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 -ddd > attack.bpf
                                       not recommended, because important pre-authentication frames will be lost due to MAC randomization of the CLIENTs
                                      use the BPF code
                                       $ hcxumptool -i <interface> --bpfc=attack.bpf ...
                                     notice: this is a protect/attack, a capture and a display filter
                                     see man pcap-filter for a list of all filter options
```

General  information about BPF is here:
https://en.wikipedia.org/wiki/Berkeley_Packet_Filter

ZerBea, 2020-11-28:
example:
protect AP (BSSID MAC:11:22:33:44:55:66)
```
$ hcxdumptool -m interface
$ tcpdump -i <interface> not wlan addr1 11:22:33:44:55:66 and not wlan addr2 11:22:33:44:55:66 -ddd > protect.bpf
```
Now you can add the filter code to hcxdumptool to protect this AP:
```
$ hcxdumptool -i wlan0 -o vector2.pcapng --tot=1440 --bpfc=protect.bpfc --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,9,6,3,11, -t 3600 --enable_status=1
```

ZerBea, 2020-11-28:
Closed this report, because it is not an issue, but you can still ask your questions here.

closed 2020-11-28 by ZerBea

conf11s3d, 2020-11-28:
Heartfelt thanks for the help now all clear .... a hug

ZerBea, 2020-11-28:
Nice to hear that.
The example is an advanced command line.

conf11s3d, 2020-11-29:
Hi Zerbea another help, always if possible, I would like both the attack and the client mode to be directed on a single Access Point target chosen by me, I do not want to attack or acquire all the traffic of all the Access Points or Clients, you would be kind to help me better filter both the attack and the monitor mode of customers directed towards a single Access Point ?

ATTACK:
hcxdumptool -i wlan0 -o vector1.pcapng --stop_ap_attacks=6000 --resume_ap_attacks=12000 --bpfc=own.bpfc --active_beacon --tot=10 --enable_status=1

Which filters should I add for a single Access Point attack ?

CLIENTS:
hcxdumptool -i wlan0 -o vector2.pcapng --tot=1440 --bpfc=own.bpfc --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,9,6,3,11, -t 3600 --enable_status=1

Which filters should I add to filter clients traffic directed to a single Access Point ?

Thanks always for your kind attention.

ZerBea, 2020-11-29:
First of all you should know that filtermode (in combination with filterlists) is not(!) a display filter.
Adding APs or CLIENTs to the lists will allow/prevent to attack them, but they are still shown in the waterfall status.

The BPF is working as a display filter, too.

Both methods have pros and cons, especially if the CLIENT use MAC RANDOMIZATION.
https://www.centurylink.com/home/help/internet/secure-wifi/What-is-MAC-randomization.html

If you knwo the AP MAC and the CLIENT MAC you can use a BPF. But be warned, all additional information from CLIENTs running MAC RANDOMIZATION is filtered out:
```
$ sudo hcxdumptool -m wlan0
$ sudo tcpdump -i wlan0  wlan addr1 11:11:11:11:11:11 or wlan addr2 11:11:11:11:11:11 or  wlan addr1 22:22:22:22:22:22 or wlan addr2 22:22:22:22:22:22 -ddd > attack.bpf
$ sudo hcxdumptool -i wlan0 -o vector1.pcapng --stop_ap_attacks=6000 --resume_ap_attacks=12000 --bpfc=attack.bpfc --enable_status=1
```
where
11:11:11:11:11:11 is your MAC AP
and
22:22:22:22:22:22 is your MAC CLIENT

Now wait until you got an EAPOL handshake or a PMKID.

If you have more CLIENTs, just add them to the tcpdump BP command line, separated by "or".

conf11s3d, 2020-11-29:
Perfect just tried thanks for the help....

ZerBea, 2020-11-29:
Nice to hear that.

BTW:
You should know that hcxdumptool/hcxtools are designed as analysis tools. They have lots of options and are not easy to use unless you understand the whole 802.11 stuff.