ZerBea/hcxdumptool #150: View connected clients ?

conf11s3d, 2020-12-02:
Hi Zerbea I was lucky with your good advice and I will never stop thanking you for all the hard work you do, I ask if possible I have a way to view the connected clients ?

    --do_rcascan                               I have no way to view clients but only AP
    --do_targetscan = <MAC_AP>    the same here too

it would be nice to see the connected clients too.
Thanks always for your good hugs advice ....

ZerBea, 2020-12-02:
Showing CLIENTs is not the purpose of the RCA scan.

If you would like to see CLIENTS, enable this in realtime display:
```
-enable_status=<digit>            : enable real-time display (waterfall)
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
                                     bitmask:
                                         0: no status (default)
                                         1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
                                        32: GPS (once a minute)
                                        64: internal status (once a minute)
                                       128: run as server
                                       256: run as client
                                       512: EAP
                                      1024: EAP NAK
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```
This will show you all CLIENT requests, too:
`$ sudo hcxdumptool -i <interface> --enable status=15
`
You'll see the kind of the connection between a CLIENT and an AP:
 ```
18:02:02   1 dc70146ce5e3 0025df125c99 WLAN-TEST [AUTHENTICATION]

18:02:02                  = time
1                              = channel
dc70146ce5e3        = CLIENT
0025df125c99         = AP
WLAN-TEST            = ESSID
AUTHENTICATION  = kind of connection
```

conf11s3d, 2020-12-02:
Thanks Zerbea I had read in the guide -enable_status = <digit> ok for everything these days I have had excellent results on tests carried out thanks for your advice.

ZerBea, 2020-12-02:
In combination with BPF code, you'll see only traffic related to the AP:
`
$ sudo tcpdump -i <interface> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 -ddd > attack.bpf
`
where 11:22:33:44:55:66 is your target AP.

ZerBea, 2020-12-02:
As I wrote before, hcxtools are not easy to use, but there is an option for nearly every purpose

closed 2020-12-02 by ZerBea