ZerBea/hcxdumptool #151: failed to read packet network is down

nxfxcom, 2020-12-02:
Hello ZerBea,

First of all - Thank you for the amazing support and product! I am running the latest version on Kali. However, after running for a random amount of time I get the "failed to read packet network is down" error.

I am using the following adapter:
https://www.amazon.com/Panda-Wireless-PAU06-300Mbps-Adapter/dp/B00JDVRCI0

I am using:
`rt2800usb       Ralink Technology, Corp. RT5372`

When I check the driver I get:
`all required ioctl() system calls are supported by driver`

When I check injection I get:
```
packet injection is working!
ratio: 193 to 122 
```

Thank you

ZerBea, 2020-12-02:
This error message is located here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5143
It is caused by the driver and it is not a hcxdumptool issue. If you receive it, hcxdumptool doesn't receive packets from the driver any longer, because the system disconnected the device or a third party tool takes access to the device.
Please attach output of dmesg if that happened. We don't need the complete dmesg log.
Comment dmesg output after you have plugged in the device until hcxdumptool terminate with the warning.
It will look like this
```
[52711.247717] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[52711.491915] usb 1-2: New USB device found, idVendor=148f, idProduct=3070, bcdDevice= 1.01
[52711.491920] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[52711.491923] usb 1-2: Product: 802.11 n WLAN
[52711.491925] usb 1-2: Manufacturer: Ralink
[52711.491927] usb 1-2: SerialNumber: 1.0
[52711.639265] usb 1-2: reset high-speed USB device number 5 using xhci_hcd
[52711.872442] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[52712.550416] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 0005 detected
[52712.556082] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[52712.573617] rt2800usb 1-2:1.0 wlp3s0f0u2: renamed from wlan0
[52939.870814] ieee80211 phy1: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[52939.870845] ieee80211 phy1: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
``` 

First run this commands:
$ hcxdumptool -i interface --check_driver
and comment complete output
```
$ sudo hcxdumptool -i wlp3s0f0u2 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

Than run hcxdumptool using your comaind line parameters and wait until the error occurs.
 dmesg will show you that the interface entered monitor mode:
``` 
[52944.104616] device wlp3s0f0u2 entered promiscuous mode
``` 
If the error occurs, you will find the error message of dmesg here.

When hcxdumptool teminate, dmesg will show you this message:
``` 
[52944.197999] device wlp3s0f0u2 left promiscuous mode
``` 

Here is an example for a broken connection (damaged USB connector):
```
[53926.616755] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[53926.793790] usb 1-2: New USB device found, idVendor=148f, idProduct=3070, bcdDevice= 1.01
[53926.793794] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[53926.793797] usb 1-2: Product: 802.11 n WLAN
[53926.793799] usb 1-2: Manufacturer: Ralink
[53926.793801] usb 1-2: SerialNumber: 1.0
[53927.004900] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[53927.237725] ieee80211 phy2: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[53927.915849] ieee80211 phy2: rt2x00_set_rf: Info - RF chipset 0005 detected
[53927.921854] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
[53927.939143] rt2800usb 1-2:1.0 wlp3s0f0u2: renamed from wlan0
```
$  sudo hcxdumptool -i wlp3s0f0u2 --do_rcascan
```
[53970.172774] ieee80211 phy2: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[53970.172816] ieee80211 phy2: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[53974.429564] device wlp3s0f0u2 entered promiscuous mode
```
hcxdumptool report an error:
```
failed to read packet: Network is down

maximum number of errors is reached
```
and dmesg show what caused it:
```
[53987.679975] usb 1-2: USB disconnect, device number 6
[53987.680371] ieee80211 phy2: rt2x00usb_vendor_request: Error - Vendor Request 0x07 failed for offset 0x101c with error -19
[53987.680543] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[53987.777161] device wlp3s0f0u2 left promiscuous mode
```
device was disconnected from the system due to a faulty USB connector.

nxfxcom, 2020-12-24:
Hello,

this was super helpful. Thank you for the reply and please excuse the delay in response. After doing some upgrades and modifications I got it to work! It was mostly a power issue! 

Thank you so much and happy holidays!

closed 2020-12-24 by nxfxcom

ZerBea, 2020-12-24:
Glad to hear that, now.
Merry Christmas and a happy new year.
Cheers
Mike