ZerBea/hcxdumptool #152: False "Positives" PMKID (...with repeaters routers)

vincenzogianfelice, 2020-12-03:
Hi. Today I did some tests and I noticed how (after converting the file generated by hcxdumptool to pcap) **aircrack-ng** informed me the presence of the **_PMKID_** an _AP_ when in reality there was not. This thing only happens between two _AP's_ where at least one is a repeater (and which hooks to the main _AP_). Why does this happen? thanks and sorry for my english

ZerBea, 2020-12-03:
Hi. Good investigation.
You should know that not every AP calculate a PMKID and not every CLIENT use a PMKID.
Sometimes the initial PMKID can be zeroed, too. Mostly that happens on the first connection attempt of a new CLIENT.
Sometimes the PMKID is calculated from a zeroed PMK. Mostly that happens if the AP or the CLEINT is misconfigured.
Transmitting to many deauthentication packets will destroy the EAPOL timer of an AP. That happens if you run a stupid deauthenticator (mdk3, aireplay-ng). That result in a useless PMKID and/or EAPOL message.
This are  known issues in aircrack-ng
https://github.com/aircrack-ng/aircrack-ng/issues/2197
https://github.com/aircrack-ng/aircrack-ng/issues/2175
https://github.com/aircrack-ng/aircrack-ng/issues/2079

Please notice:
hcxdumptool will show you every received PMKID. That include PMKIDs calculated from a zeroed PMK as well as non PBKDF2 based PMKIDs. The reason is simple: On a Raspberry Pi, we are in real time and have not time (CPU cycle) to verify this.
Later on, when converting, we have every time, wee need. So hcxpcapngtool will do several tests and convert only useful PMKIDs and EAPOL message pairs.

Closed this, because it isn't an issue, but you can still ask your questions, here.

closed 2020-12-03 by ZerBea

vincenzogianfelice, 2020-12-03:
Thank you so much!
So, how can I "fix it" with **hcxpcapngtool**?
Thanks

ZerBea, 2020-12-03:
hcxpcapngtool can't fix it, because its purpose is to convert hashes from WiFi traffic to a format hascat and/or john understand.
How do you captured the traffic?

vincenzogianfelice, 2020-12-03:
With **hcxdumptool** (`hcxdumptool --enable_status=1 -o pmkid -i wlan1mon`)

ZerBea, 2020-12-03:
In this case, there is nothing to fix.
hcxdumptool capture the traffic and hcxpcapngtool convert the hashes to hashcat format 22000. 
Usless PMKIDs and/or useless EAPOL messages are not converted.

BTW:
You shouldn't use hcxdumptool on a monitor interface created by airmon-ng!

vincenzogianfelice, 2020-12-03:
> "Non dovresti usare hcxdumptool su un'interfaccia monitor creata da airmon-ng!"

Why?

ZerBea, 2020-12-03:
airmon-ng is designed to be used with aircrack-ng and not with hcxdumptool. It create a virtual monitor NETLINK interface that could interfere with hcxdumptool.
hcxdumptool is running an own monitor mode. It doesn't use NETLINK. You should have received a warning if a NETLINK monitor interface is detected.
Correct procedure:
stop all services that take access to the device
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service
run hcxdumptool
when teminated, start your restart you services
sudo systemctl start NetworkManager.service
sudo systemctl start wpa_supplicant.service

vincenzogianfelice, 2020-12-04:
> hcxpcapngtool convert the hashes to hashcat format 22000.
> Usless PMKIDs and/or useless EAPOL messages are not converted.

Its possibile convert **pcapng** in **_hashcat format_** (removing unnecessary pmkid), then convert the file to **pcap**?

ZerBea, 2020-12-04:
convert pcapng to hashcat format:
`$ hcxpcapngtool -o test.22000 *.pcapng
`
get info about hash format
`$ hcxhastool -i test.22000 --info=stdout
`
convert hashcat 22000 format to cap (pcap is not possible because 22000 doesn't contain radio information)
`$ hcxhash2cap --pmkid-eapol=test.22000 -c test.cap
`

Please notice:
The conversion from 22000 to cap is not lossless! You will loose important information from the origin pcapng file.

hcxtools provide tools to work on pcapng/pcap/cap and hash files for every purpose. --help will give you an information about the command line options.