ZerBea/hcxdumptool #153: Test analysis my router

conf11s3d, 2020-12-06:
Hi Zerbea I hope everything goes well your advice is always precious to me, in these days I have carried out several tests for study and analysis, but I have noticed that in some tests I have not received PMKID but I still received the four-way handshake

**Question:** in the handshake I acquired can I continue with hashcat or do I have to trash it ?

In the acquired file obviously I have the final **WARNING** message but I still have a handshake

Warning: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

**hcxpcapngtool vector1.pcapng**

reading from vector1.pcapng...

summary capture file
--------------------
file name................................     : vector1.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.9.0-kali2-amd64
application..............................     : hcxdumptool 6.1.4-1-g9c03887
interface name...........................  : wlan1
interface vendor.........................  : 20f41b
openSSL version..........................: 1.1
weak candidate........................... : 12345678
MAC ACCESS POINT..................: 7ce4aabd3b0a (incremented on every new client)
MAC CLIENT...............................: e804106e21f4
REPLAYCOUNT...........................: 62670
ANONCE..................................   .: 896350968c5b2e60ba4cf3f2e337b25b4caff1833ed8990e4ba53edfb4b096a3
SNONCE...................................   : cdf333e678d93b2292d063219c0ffe3c3e9c467e5e6cb8545b3f982218670d62
timestamp minimum (GMT)..........: 05.12.2020 20:31:52
timestamp maximum (GMT)..........: 05.12.2020 20:35:35
used capture interfaces.................: 1
link layer header type................... : DLT_IEEE802_11_RADIO (127)
endianess (capture system)......... : little endian
packets inside...........................     : 300
BEACON (total)...........................   : 1
PROBERESPONSE...................... : 1
AUTHENTICATION (total)..............: 3
AUTHENTICATION (OPEN SYSTEM)......: 3
ASSOCIATIONREQUEST (total)...............: 2
ASSOCIATIONREQUEST (PSK).............  : 2
EAPOL messages (total)...................        : 291
EAPOL RSN messages.......................      : 291
ESSID (total unique)....................              .: 1
EAPOLTIME gap (measured maximum usec)....: 81437
EAPOL ANONCE error corrections (NC)......: not detected
REPLAYCOUNT gap (measured maximum).......: 18
EAPOL M1 messages (total)................: 279
EAPOL M2 messages (total)................: 4
EAPOL M3 messages (total)................: 4
EAPOL M4 messages (total)................: 4
EAPOL pairs (total)......................: 10
EAPOL pairs (best).......................: 2
EAPOL M32E2 (authorized).................: 2

Information: no hashes written to hash files

Warning: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.

session summary
---------------
processed pcapng files................: 1

Thanks Zerbea always a hug for your advice ....

ZerBea, 2020-12-06:
I assume you used filter options when capturing. So the warning is absolutely normal, because some frames are filtered out by options. Some APs doesn't calculate a PMKID. In this case, you have to use the EAPOL message pair.
Running
`$ hcxpcapngtool -o test.22000 vector1.pcapng
`
will convert convert the EAPOL message to hashcat 22000 format.

Now run
`$ hashcat -m 22000 test.22000 your_wordlist
`
to recover the PSK.

To get some information about the converted hash file run
`$ hcxhashtool -i test.22000 --info=stdout
`

Closed this, because it isn't an issue, but you can still ask your questions, here.

closed 2020-12-06 by ZerBea

conf11s3d, 2020-12-07:
Thanks Zerbea for the answer, yes I used filter options same commands as the messages posted in days back using  **--bpfc=<file>** I needed to understand and now I have confirmation thanks to your answers I continue my study analysis test thanks a hug

ZerBea, 2020-12-07:
You should know that most of the APs are hardened and most of the CLIENTs are weak. Regarding this, you will get much more information from a CLIENT than from an AP. That include EAPOL messages (unencrypted EAPOL M2 from a CLIENT), a PMKID (if used by the CLIENT) and possible a PSK in  the clear (in PROBEREQUEST and or EAP-ID).
In other words: to recover the PSK for a network you don't need to attack an AP. You don't need to be in range of an AP. You can retrieve all information (and more) from a CLIENT. That is called an AP-LESS attack and hcxdumptool will show you M1M2ROGUE.
Get an example from here:
https://github.com/evilsocket/pwnagotchi/issues/835#issuecomment-598597214 

From this EAPOL messages we are able to recover the PSK:
M1M2ROGUE = hcxdumptool request M2 (CHALLENGE) from a CLIENT
PMKIDROGUE = hcxdumptool REQUEST PMKID from an AP
M1M2 = hcxdumptool captured M1 from the AP and M2 (CHALLENGE) from the CLIENT 
M2M3 = hcxdumptool captured M3 the AP and M3 from the CLIENT
M3M4 = hcxdumptool captured M3 the AP and M4 from the CLIENT
M1M4 = hcxdumptool captured M1 from the AP and M4 from the CLIENT
EAPOL M1 and M2 are unencrypted, while EAPOL M3 and M4 are encrypted.

From this messages we are not able to recover the PSK, because the SNONCE of the CLIENT is zeroed
M3M4ZEROED = hcxdumptool captured M3 the AP and M4 from the CLIENT
M1M4ZEROED = hcxdumptool captured M1 from the AP and M4 from the CLIENT

conf11s3d, 2020-12-07:
Hi Zerbea sorry if I still answer now due to work shifts, yes now I am focusing only on client-side acquisition using two modes

**General client:**   (with bpf protect)
hcxdumptool -i wlan1 -o client.pcapng --tot=1440 --bpfc=protect.bpf --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,2,3,4,5,6,7,8,9,10,11,12,13 -t 3600 --enable_status=15

**Single AP client:**  (with bpf attack)
hcxdumptool -i wlan1 -o client.pcapng --tot=1440 --bpfc=attack.bpf --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,2,3,4,5,6,7,8,9,10,11,12,13 -t 3600 --enable_status=15

I hope that with these commands I should have the highest number of packet capture, if you have to suggest or refine these commands help me to add more filters.

Now a question if possible in your opinion attacking a single client with the same command as above putting in bpf MAC_Client and not MAC_AP in bpf ATTACK mode I have the same acquisition result as when I attack AP with bpf ATTACK I explain better with the commands:

**Single AP client:** client acquisition on AP (with bpf attack MAC_AP) 
hcxdumptool -i wlan1 -o client.pcapng --tot=1440 --bpfc=attack.bpf --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,2,3,4,5,6,7,8,9,10,11,12,13 -t 3600 --enable_status=15

**Single client:**  acquisition directly on the client (with bpf attack MAC_Client)
hcxdumptool -i wlan1 -o client.pcapng --tot=1440 --bpfc=attack.bpf --disable_deauthentication --disable_ap_attacks --active_beacon -c 1,2,3,4,5,6,7,8,9,10,11,12,13 -t 3600 --enable_status=15

of course on MAC_Client not random but with static MAC
Thanks always for your great advice I owe my results to you thanks ..

ZerBea, 2020-12-07:
If you are targeting CLIENTs, I recommend a smaller channel list (-c 1,9,6,3,11). 
You can be sure that  every CLIENT will find hcxdumptool.

conf11s3d, 2020-12-07:
I was convinced to enter all channels to have a total acquisition on all channels thanks for this suggestion ...

ZerBea, 2020-12-07:
You commands doing nearly the same. In your case, both of them are targeting a connection between the CLIENT and the AP.

conf11s3d, 2020-12-07:
so if i want a single client with static MAC i shouldn't put it in bpf attack

ZerBea, 2020-12-07:
A CLIENT will scan all channels for "his" AP. So you can simply wait for him until he arrived on hcxdumptool channel. We use three crowded channels (1,6,11) and two less crowded channels (3 and 9) to give him the chance to connect to us.

ZerBea, 2020-12-07:
Most (nearly all) of the CLIENTs use MAC randomization, when sending PROBEREQUESTs. Adding a static MAC of a CLIENT to an attack filter will reduce the chance that the CLIENT find hcxdumptool by 100%!

conf11s3d, 2020-12-07:
ok on the channels now I understand well, for the rest I think that filters are fine or should I add other filters to have an accurate acquisition ?

ZerBea, 2020-12-07:
You should notic, that hcxdumptool stops to respond to a CLIENT if it retrieved his M2 or his PMKID. You can control this behavior with stop_client_m2_attacks. 
```
--stop_client_m2_attacks=<digit>   : stop attacks against CLIENTS after 10 M2 frames received
                                     affected: ap-less (EAPOL 2/4 - M2) attack
                                     require hcxpcangtool --all option
```

Please notice:
A CLIENT can have more than one AP in his wpa-supplicant.conf!

conf11s3d, 2020-12-07:
well thanks for this last tip i will go with general client bpf protect

ZerBea, 2020-12-07:
Filters make sense if you attack a single AP. They always make sense to protect your own devices.
But they never make sense when attacking a CLIENT!

conf11s3d, 2020-12-07:
the day before yesterday I was trying to understand well in the guide --stop_client_m2_attacks = <digit>
in fact it is excellent as an addition

conf11s3d, 2020-12-07:
thanks now I understand the client side, my in-depth study is largely due to your teaching thank you very much

conf11s3d, 2020-12-07:
now I disconnect and continue with the tests thanks again for the help .... a hug

ZerBea, 2020-12-07:
Glad that I could help.

JuniorJPDJ, 2020-12-07:
BTW. why bpf and not whitelist/blacklist?

ZerBea, 2020-12-07:
The BPF is written in a special code which is based on logical and comparison operators.
So, working as whitelist or blacklist depend on the code (eg.: and, not, or, ...). 
Read more here:
https://biot.com/capstats/bpf.html
BTW:
hcxdumptool doesn't have a BPF compiler build in and therefor it use the BPF compiler from tcpdump.

conf11s3d, 2020-12-08:
Hello Zerbea, a question about a test.pcapng captured file if you would like to review the waterfall screen at a later time you can review this screen example:

8:10:28   6 ffffffffffff 101xxxxfc479 AP [BEACON]
8:10:43   5 507xxxx1d6dd 008xxxx35459 AP [ROGUE PROBERESPONSE]
8:11:28   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M1M2 EAPOLTIME:3320 RC:4 KDV:2]
8:11:28   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M2M3 EAPOLTIME:8947 RC:5 KDV:2]
8:11:28   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M3M4ZEROED EAPOLTIME:2836 RC:5 KDV:2]
8:11:29   6 4c1xxxxdd479 008xxxx3545a AP [ROGUE PROBERESPONSE]
8:11:29   6 4c1xxxxdd479 101xxxxfc479 AP [PROBERESPONSE]
8:11:30   6 4c1xxxxdd479 101xxxxfc479 AP [AUTHENTICATION]]
8:12:08   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M1M2 EAPOLTIME:3807 RC:3 KDV:2]
8:12:08   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M2M3 EAPOLTIME:7494 RC:4 KDV:2]
8:12:08   6 4c1xxxxdd479 101xxxxfc479 AP [EAPOL:M3M4ZEROED EAPOLTIME:3408 RC:4 KDV:2]

I don't remember the command for redisplaying on an already acquired file
thank you always for your patience on me ....

ZerBea, 2020-12-08:
There is no command to redisplay the waterfall output in hcxtools, because tshark and Wireshark will do the job.
This is the simplest example to display the waterfall:
`$ tshark -r test.pcapng
`
There are many, many filter options to filter the waterfall on what ever you want, e.g.:
```
$ tshark -r test.pcapng -T fields -e frame.number -e frame.time -e wlan.sa -e wlan.da
$ tshark -r test.pcapng -T fields -e wlan.ta -e wlan.ssid -e wps.device_name -e wps.serial_number
```
Please read more here:
https://www.wireshark.org/docs/man-pages/tshark.html

To view the converted content in hcxtools, convert it using hcxpcapngtool -o -e -i -u options and take a look inside the files.
For the hash file (-o output of hcxpcapngtool) use hcxhashtool or a text editor (e.g. geany)
For the other files (-e -i-u output of hcxpcapngtool) use text editor (e.g. geany).

conf11s3d, 2020-12-08:
Thank you very much Zerbea I was looking for these commands thanks ....

JuniorJPDJ, 2020-12-08:
> The BPF is written in a special code which is based on logical and comparison operators.
> So, working as whitelist or blacklist depend on the code (eg.: and, not, or, ...).
> Read more here:
> `$ man bpf`
> or here:
> https://biot.com/capstats/bpf.html
> BTW:
> hcxdumptool doesn't have a BPF compiler build in and therefor it use the BPF compiler from tcpdump.

I mean.. I know what's BPF, I'm asking why should we use BPF filtering and not hcxdump's whitelist/blacklist feature itself.

ZerBea, 2020-12-08:
The BPF is part of the kernel and extreme fast, while the filter list is running in user space. It will take some CPU cycles.
Additional, the BPF is working as a display filter, too.

conf11s3d, 2020-12-10:
Hi Zerbea I just wanted to wish you a good day before I go to work, the study and analysis continue with excellent results I will never stop thanking you for the great development work for these tools with attention to the smallest details ... hugs.

ZerBea, 2020-12-10:
You're welcome.
I worked in this fields (signals analysis and crypto analysis) for more than 35 years and immensely enjoyed this work.
And I'm still learning.

conf11s3d, 2020-12-10:
Hi Zerbea, as always I am reading the info manual and I would say well done with attention to detail, but reading I focused on a note that you make if you use the **--all** command while converting the file just acquired  *** .pcapng** I prefer the command **--all** all hashes, but you noted this

--all  : convert all possible hashes instead of only the 
         best one that can lead to much overhead hashes
         need hashcat --nonce-error-corrections >= 8

so advice for those who use the **--all** command to use in hashcat  **--nonce-error-corrections=8**

the question is: do I have to use **--nonce-error-corrections=8** every time I use **--all** or you refer to conversions of scanned files from other programs not belonging to your toolkit and advice in that case to use --nonce -error-corrections=8 ?
and then how do I understand that in the converted file example **test.22000** there are overhead hashes where I should see the errors in this command **hcxhashtool -i test.22000** and here it should show any errors ?
Thanks as always...

ZerBea, 2020-12-10:
First you have to understand how nonce-error-corrections is working.
Usually an AP increment replaycount on evvery EAPOL 4way handshake. That looks like this:
M1 RC 0
M2 RC 0
M3 RC 1
M4 RC 1

If the EAPOL timer is not resetted, next 4 way looks like this:
M1 RC 2
M2 RC 2
M3 RC 3
M4 RC 4

Some APs don't do this. Instead the increment their ANONCE:
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb7
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb8
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb9
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbba

To calculate a valid message pair, we need this ANONCE (comming from an AP) and matching SNONCE (comming from a CLIENT). Running a passive dumper, it can happen that some of the EAPOL packets get lost and we are not able to find a matching pair.
nonce-error corrections is able to compensate this and will pick up all ANONCES in range to calculate a valid message pair.
e.g.: NC = 2 will pick up this ANONCES
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb7
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb8
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb9
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbba

NC = 8 will try this ones:
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb1
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb2
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb3
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb4
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb5
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb6
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb7
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb8
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbb9
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbba
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbbb
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbbc
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbbd
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbbe
f234f3ed8a9809958579bec6f87b04eadba43cebdabc389ea9904d7fda53cbbf

Running hcxdumptool in active attack mode, you don't need to set NC, because this will be done by automatic. There should be no packet loss to be compensated by NC.

hcxhastool will show you this.
AP attack vector:
```
KEY VERSION: WPA2
REPLAYCOUNT: 1
RC INFO....: NC not required
MP M2M3 E2.: authorized
```
CLIENT attack vector:
```
KEY VERSION: WPA2
REPLAYCOUNT: 64164
RC INFO....: NC not required
RC INFO....: ROGUE attack / NC not required
MP M1M2 E2.: challenge
```

conf11s3d, 2020-12-10:
Perfect excellent explanation as always I treasure it and study baggage of your explanations now everything is clear to me thanks ...

ZerBea, 2020-12-10:
The automatic is close to 100% to detect a valid PMKID and/or a valid EAPOL message pair.
`hcxdumptool -> hcxpcangtool -> hashcat
`
But it is only close to 100% and sometimes it require manual action (hcxdumptool passive mode, target at the edge of RF range, signal reflection). In this case we have to expand the workflow:
`hcxdumptool -> hcxpcapngtool (--all) -> hcxhastool -> hashcat
`

conf11s3d, 2020-12-10:
target at the edge of RF range, signal reflection) is one of many reasons why I prefer --all thanks...