ZerBea/hcxdumptool #155: Troubles installing hcxdumptool and hcxtools

CiprianDistrugatorul, 2020-12-17:
Hi there! I have been struggling with this problem for a few hours, I've read a lot of posts and none have got me anywhere, so I'm directly asking for your help!

Initially I cloned and tried to make and make install the files, but it gave me an error that told me I have issues with libssl. I somehow managed to solve that, and now I am getting another problem. After cloning the repos, I try the make command, and the terminal is giving me this:

`make: pkg-config: No such file or directory
make: pkg-config: No such file or directory
make: pkg-config: No such file or directory
make: pkg-config: No such file or directory
cc -O3 -Wall -Wextra -std=gnu99    -MMD -MF .deps/hcxpcapngtool.d -o hcxpcapngtool hcxpcapngtool.c    -DVERSION_TAG=\"6.1.4-21-g395bd29\" -DVERSION_YEAR=\"2020\"
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `getpmkid':
hcxpcapngtool.c:(.text+0x1ab7): undefined reference to `PKCS5_PBKDF2_HMAC_SHA1'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x1b0d): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x1b37): undefined reference to `HMAC'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `initlists':
hcxpcapngtool.c:(.text+0x2e35): undefined reference to `OpenSSL_version_num'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `outputeapmschapv2hashlist':
hcxpcapngtool.c:(.text+0x4775): undefined reference to `SHA1_Init'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x478a): undefined reference to `SHA1_Update'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x479f): undefined reference to `SHA1_Update'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x47b5): undefined reference to `SHA1_Update'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x47c8): undefined reference to `SHA1_Final'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `addpmkid':
hcxpcapngtool.c:(.text+0x4c31): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x4c5f): undefined reference to `HMAC'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `decompressgz':
hcxpcapngtool.c:(.text+0x6210): undefined reference to `inflateInit2_'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x62d4): undefined reference to `inflate'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x62e6): undefined reference to `inflateEnd'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6324): undefined reference to `inflateEnd'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6359): undefined reference to `inflateEnd'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `omac1_aes_128_vector':
hcxpcapngtool.c:(.text+0x64d8): undefined reference to `CMAC_CTX_new'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x64ed): undefined reference to `EVP_aes_128_cbc'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6507): undefined reference to `CMAC_Init'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6550): undefined reference to `CMAC_Update'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6560): undefined reference to `CMAC_CTX_free'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x658a): undefined reference to `CMAC_Final'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `testeapolpmk':
hcxpcapngtool.c:(.text+0x67af): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x67db): undefined reference to `HMAC'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6801): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x682c): undefined reference to `HMAC'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x69eb): undefined reference to `EVP_sha256'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6a15): undefined reference to `HMAC'
/usr/bin/ld: hcxpcapngtool.c:(.text+0x6a91): undefined reference to `EVP_md5'
/usr/bin/ld: /tmp/ccqc3DOg.o: in function `gethandshake':
hcxpcapngtool.c:(.text+0x82ce): undefined reference to `PKCS5_PBKDF2_HMAC_SHA1'
collect2: error: ld returned 1 exit status
make: *** No rule to make target 'hcxpcapngtool', needed by 'build'.  Stop.`

What should I do now?

ZerBea, 2020-12-17:
You have to solve the dependencies as mentioned in README.md:
```
Requirements
--------------

* Linux (recommended Arch Linux, but other distros should work, too (no support for other distributions).

* gcc 10 recommended (deprecated versions are not supported: https://gcc.gnu.org/)

* libopenssl and openssl-dev installed

* librt and librt-dev installed (should be installed by default)

* zlib and zlib-dev installed (for gzip compressed cap/pcap/pcapng files)

* libcurl and curl-dev installed (used by whoismac and wlancap2wpasec)

* libpthread and pthread-dev installed (used by hcxhashcattool)

* pkg-config installed

To install requirements on Kali use the following 'apt-get install pkg-config libcurl4-openssl-dev libssl-dev zlib1g-dev'
```
Otherwise you will not be able to compile hcxdumptool/hcxtools.
At least you are missing the development files of openSSL and pkg-config. It looks like this files are not installed on your distribution - so you have to do this.

Closed this, because it isn't hcxtools issue, but you can still ask your questions, here.

closed 2020-12-17 by ZerBea

ZerBea, 2020-12-17:
I also recommend to read this (especially if you're running KALI Linux):
https://github.com/ZerBea/hcxtools/issues/35