ZerBea/hcxdumptool #156: Can't compile hcxdumptool 6.1.4 on Kali

gril002, 2020-12-19:
I am running Kali kernel version 5.9.0-kali4-amd64 in a VM. I have installed both the dependencies, did apt-get update and upgrade and tried getting it through git and manual zip download. When I try to make in the hcxdumptool directory I get the output below. I manually downloaded the ZIP of version 6.1.3 and used the same steps and I was able to successfully compile and install the software. Sorry if I am missing anything obvious/easily googlable. Let me know if you need any more info
```
$ make          
make: pkg-config: No such file or directory
make: pkg-config: No such file or directory
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c   -DVERSION_TAG=\"6.1.4-6-ge49098e\" -DVERSION_YEAR=\"2020\"
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `globalclose':
hcxdumptool.c:(.text+0x2393): undefined reference to `SSL_free'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `detectweakwpa':
hcxdumptool.c:(.text+0x3d0c): undefined reference to `PKCS5_PBKDF2_HMAC_SHA1'
/usr/bin/ld: hcxdumptool.c:(.text+0x3e69): undefined reference to `EVP_sha256'
/usr/bin/ld: hcxdumptool.c:(.text+0x3e93): undefined reference to `HMAC'
/usr/bin/ld: hcxdumptool.c:(.text+0x3ea6): undefined reference to `CMAC_CTX_new'
/usr/bin/ld: hcxdumptool.c:(.text+0x3eb9): undefined reference to `EVP_aes_128_cbc'
/usr/bin/ld: hcxdumptool.c:(.text+0x3ed7): undefined reference to `CMAC_Init'
/usr/bin/ld: hcxdumptool.c:(.text+0x3f0c): undefined reference to `CMAC_Update'
/usr/bin/ld: hcxdumptool.c:(.text+0x3f1c): undefined reference to `CMAC_CTX_free'
/usr/bin/ld: hcxdumptool.c:(.text+0x408e): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxdumptool.c:(.text+0x40ba): undefined reference to `HMAC'
/usr/bin/ld: hcxdumptool.c:(.text+0x40f4): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxdumptool.c:(.text+0x411b): undefined reference to `HMAC'
/usr/bin/ld: hcxdumptool.c:(.text+0x4279): undefined reference to `EVP_md5'
/usr/bin/ld: hcxdumptool.c:(.text+0x429e): undefined reference to `CMAC_Final'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `globalinit':
hcxdumptool.c:(.text+0x4e03): undefined reference to `OpenSSL_version_num'
/usr/bin/ld: hcxdumptool.c:(.text+0x52bf): undefined reference to `OPENSSL_init_ssl'
/usr/bin/ld: hcxdumptool.c:(.text+0x52c8): undefined reference to `OPENSSL_init_ssl'
/usr/bin/ld: hcxdumptool.c:(.text+0x52cd): undefined reference to `TLS_server_method'
/usr/bin/ld: hcxdumptool.c:(.text+0x52d5): undefined reference to `SSL_CTX_new'
/usr/bin/ld: hcxdumptool.c:(.text+0x52f9): undefined reference to `SSL_CTX_use_certificate_file'
/usr/bin/ld: hcxdumptool.c:(.text+0x5319): undefined reference to `SSL_CTX_use_PrivateKey_file'
/usr/bin/ld: hcxdumptool.c:(.text+0x5355): undefined reference to `SSL_CTX_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x536d): undefined reference to `SSL_CTX_set_verify'
/usr/bin/ld: hcxdumptool.c:(.text+0x5385): undefined reference to `SSL_CTX_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x539d): undefined reference to `SSL_CTX_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x53ab): undefined reference to `SSL_CTX_set_quiet_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x54b7): undefined reference to `ERR_print_errors_fp'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `printpmkid':
hcxdumptool.c:(.text+0x5a2c): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxdumptool.c:(.text+0x5a58): undefined reference to `HMAC'
/usr/bin/ld: hcxdumptool.c:(.text+0x5bdd): undefined reference to `PKCS5_PBKDF2_HMAC_SHA1'
/usr/bin/ld: hcxdumptool.c:(.text+0x5ede): undefined reference to `EVP_sha1'
/usr/bin/ld: hcxdumptool.c:(.text+0x5f08): undefined reference to `HMAC'
/usr/bin/ld: hcxdumptool.c:(.text+0x6018): undefined reference to `PKCS5_PBKDF2_HMAC_SHA1'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `send_eap_tls':
hcxdumptool.c:(.text+0x63b1): undefined reference to `SSL_write'
/usr/bin/ld: hcxdumptool.c:(.text+0x63d0): undefined reference to `BIO_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x6412): undefined reference to `BIO_read'
/usr/bin/ld: hcxdumptool.c:(.text+0x661c): undefined reference to `BIO_read'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `send_eap_tls_eap':
hcxdumptool.c:(.text+0x7492): undefined reference to `SSL_write'
/usr/bin/ld: hcxdumptool.c:(.text+0x74b1): undefined reference to `BIO_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x74f3): undefined reference to `BIO_read'
/usr/bin/ld: hcxdumptool.c:(.text+0x7744): undefined reference to `BIO_read'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `process80211exteap_resp_tls':
hcxdumptool.c:(.text+0x10186): undefined reference to `SSL_free'
/usr/bin/ld: hcxdumptool.c:(.text+0x101c9): undefined reference to `SSL_is_init_finished'
/usr/bin/ld: hcxdumptool.c:(.text+0x1021c): undefined reference to `SSL_in_init'
/usr/bin/ld: hcxdumptool.c:(.text+0x102e1): undefined reference to `BIO_write'
/usr/bin/ld: hcxdumptool.c:(.text+0x10306): undefined reference to `SSL_get_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x10321): undefined reference to `SSL_is_init_finished'
/usr/bin/ld: hcxdumptool.c:(.text+0x1032e): undefined reference to `ERR_clear_error'
/usr/bin/ld: hcxdumptool.c:(.text+0x1033d): undefined reference to `SSL_accept'
/usr/bin/ld: hcxdumptool.c:(.text+0x1035d): undefined reference to `SSL_get_error'
/usr/bin/ld: hcxdumptool.c:(.text+0x10375): undefined reference to `ERR_get_error'
/usr/bin/ld: hcxdumptool.c:(.text+0x103b4): undefined reference to `SSL_free'
/usr/bin/ld: hcxdumptool.c:(.text+0x107fc): undefined reference to `SSL_is_init_finished'
/usr/bin/ld: hcxdumptool.c:(.text+0x10813): undefined reference to `SSL_do_handshake'
/usr/bin/ld: hcxdumptool.c:(.text+0x10824): undefined reference to `SSL_new'
/usr/bin/ld: hcxdumptool.c:(.text+0x10837): undefined reference to `SSL_set_accept_state'
/usr/bin/ld: hcxdumptool.c:(.text+0x1083c): undefined reference to `BIO_s_mem'
/usr/bin/ld: hcxdumptool.c:(.text+0x1084b): undefined reference to `BIO_new'
/usr/bin/ld: hcxdumptool.c:(.text+0x10854): undefined reference to `BIO_s_mem'
/usr/bin/ld: hcxdumptool.c:(.text+0x10863): undefined reference to `BIO_new'
/usr/bin/ld: hcxdumptool.c:(.text+0x1088b): undefined reference to `SSL_set_bio'
/usr/bin/ld: hcxdumptool.c:(.text+0x109b4): undefined reference to `BIO_write'
/usr/bin/ld: hcxdumptool.c:(.text+0x10a01): undefined reference to `SSL_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x10a1a): undefined reference to `SSL_free'
/usr/bin/ld: hcxdumptool.c:(.text+0x10a96): undefined reference to `SSL_read'
/usr/bin/ld: hcxdumptool.c:(.text+0x10d94): undefined reference to `ERR_reason_error_string'
/usr/bin/ld: hcxdumptool.c:(.text+0x10ee7): undefined reference to `SSL_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x10eff): undefined reference to `SSL_write'
/usr/bin/ld: hcxdumptool.c:(.text+0x10f1e): undefined reference to `BIO_ctrl'
/usr/bin/ld: hcxdumptool.c:(.text+0x10f60): undefined reference to `BIO_read'
/usr/bin/ld: hcxdumptool.c:(.text+0x11153): undefined reference to `BIO_read'
/usr/bin/ld: hcxdumptool.c:(.text+0x11532): undefined reference to `SSL_shutdown'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `process80211exteap_resp':
hcxdumptool.c:(.text+0x11cd3): undefined reference to `SSL_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x11cf2): undefined reference to `SSL_free'
/usr/bin/ld: /tmp/ccEmCmhK.o: in function `process_fd':
hcxdumptool.c:(.text+0x14c49): undefined reference to `SSL_shutdown'
/usr/bin/ld: hcxdumptool.c:(.text+0x14c58): undefined reference to `SSL_free'
collect2: error: ld returned 1 exit status
make: *** No rule to make target 'hcxdumptool', needed by 'build'.  Stop.

```

gril002, 2020-12-19:
I read the previous closed issue which was the same. This is my fault for just skimming through the requirements , although I did check the "Solve dependencies (Debian based distributions: KALI, UBUNTU, ...)" where pkg-config is not listed. So I would suggest to add it there as well to lower the number od people who can't read and create the same Issue over and over again

closed 2020-12-19 by gril002

ZerBea, 2020-12-19:
Already done by this commit (hcxdumptool):
https://github.com/ZerBea/hcxdumptool/commit/5c5b19fdabe2ffc7b158c470dc402011e84c8fb8
and this commit (hcxtools):
https://github.com/ZerBea/hcxtools/commit/ea20e144fca39584a52c408fe64ae78c4f385f1e

```
Requirements
--------------
* Operatingsystem: Arch Linux (strict), Kernel >= 5.4 (strict). It may work on other Linux systems (notebooks, desktops) and distributions, too (no support for other distributions, no support for other operating systems).
* Chipset must be able to run in monitor mode and driver must support monitor mode as well as full packet injection. Recommended: MEDIATEK (MT7601) or RALINK (RT2870, RT3070, RT5370) chipset 
* gcc 10 recommended (deprecated versions are not supported: https://gcc.gnu.org/)
* libopenssl and openssl-dev installed
* pkg-config installed
* Raspberry Pi A, B, A+, B+, Zero (WH). (Recommended: Zero (WH) or A+, because of a very low power consumption), but notebooks and desktops may work, too.
* GPIO hardware mod recommended (push button and LED).
```

gril002, 2020-12-19:
I meant to also add it here like:
Solve dependencies (Debian based distributions: KALI, UBUNTU, ...)
You need to install missing dependencies before running make:

sudo apt-get install libcurl4-openssl-dev
sudo apt-get install libssl-dev
sudo apt-get install pkg-config

Since I and most people probably just see "Dependencies for Kali" and just read and install what is listed here. I realise that people being lazy readers isn't your fault but it would make it more obvious.

ZerBea, 2020-12-19:
Ok, I understand. Done by latest commit.

ZerBea, 2020-12-19:
But I don't think the users are lazy. I assume that they are unexperienced and think they will be "hackers" after installing KALI.