ZerBea/hcxdumptool #157: wigle.net compatible GPS format

Czechball, 2020-12-25:
Hello,
would it be possible to somehow turn the .nmea or .gpx file I get from using a GPS module with hcxdumptool into a Wigle.net compatible format?

ZerBea, 2020-12-25:
That can be done via bashtools (e.g.: cat, awk), python, perl, php or any other scripting language, because the formats are standard (CSV):
NMEA format is described here:
https://wiki.openstreetmap.org/wiki/NMEA

Wigle format is described here:
https://api.wigle.net/csvFormat.html

A php example is here
https://github.com/RealEnder/dwpa/blob/master/web/wigle.php

Czechball, 2020-12-25:
I know how the wigle CSV format works, but if I understand it correctly, the NMEA from hcxdumptool doesn't actually contain any WiFi info? Only GPS output, right? So I'd also need to somehow parse the pcapng file... But doesn't the pcapng file also contain location data? I'm a little confused here

ZerBea, 2020-12-25:
Ok, I understand. hcxdumptool/hcxpcapngtool only provide a PGPS trackpoint: MAC_AP + coordinate + time in GPS track format.
To correlate this with capture WiFI data like channel and RSSI you need tshark.
Using t shark you can retrieve WiFI parameters from the pcapng file.
On overview over all filter can be retrieved by running:
`$ tshark -G fields`
or here:
https://www.wireshark.org/docs/dfref/r/radiotap.html

But let me think awhile how to add this feature to hcxpcapngtool.

ZerBea, 2020-12-25:
This is a quick and dirty example, how to retrieve WiFi information via tshark:
$ tshark -r dump.pcapng -T fields -e frame.number -e frame.time -e wlan.bssid -e wlan.ssid -e wlan_radio.channel -e wlan_radio.signal_dbm
as described here
Overview:
https://www.wireshark.org/docs/dfref/
WLAN filters:
https://www.wireshark.org/docs/dfref/w/wlan.html
RADIO filters:
https://www.wireshark.org/docs/dfref/w/wlan_radio.html

Now correlation with NMEA coordinate should be possible via the timestamp and/or MAC_AP (BSSID).

Czechball, 2020-12-25:
Thanks for the info, I'll try to play with it a bit and maybe write a script for generating wigle compatible CSVs. Adding the possibility to directly output Wigle CSV from hcxdumptool would be great tho.

ZerBea, 2020-12-25:
Background information:

https://api.wigle.net/csvFormat.html

Description
[BSSID],[SSID],[Capabilities],[First timestamp seen],[Channel],[RSSI],[Latitude],[Longitude],[Altitude],[Accuracy],[Type]
Example
1a:9f:ee:5c:71:c6,Scampoodle,[WPA2-EAP-CCMP][ESS],2018-08-01 13:08:27,161,-43,37.76578028,-123.45919439,67,3.2160000801086426,WIFI

Please do not close this issue report. I'll take a closer look at the formats and a possible hcxpcapngtool implementation.
https://wigle.net/
https://3wifi.stascorp.com/home

Czechball, 2020-12-25:
Actuall, now that I think about it, maybe dwpa (wpa-sec) could detect if pcapng file contains location data and automatically (or optionally) submit them to Wigle on upload... @RealEnder what do you think?

RealEnder, 2020-12-25:
Hi guys, I can implement submission to wigle from wpa-sec, if ZeroBeat add it to hcxpcapngtool as parser. But submissions will account to wpa-sec account, which may not be desirable for some users. But it will be useful, because we'll get the AP's coordinates right away, without later wigle query.

Czechball, 2020-12-25:
I think it would work nicely for people who don't use Wigle or don't want to use it for some reason. Although I admit that the percentage of people who use a GPS module with hcxdumptool is probably pretty low.

ZerBea, 2020-12-25:
After analyzing some pcapng dumpfiles I'm not sure about the implementation due to this major reasons:

Nearly all MACs used/generated/calculated by hcxdumptool are cloned or randomized. This will produce many, many false networks. Hcxdumptool is CLIENT oriented (in contrast to kismet, aircrack-ng and other tools). That means that the AP to which a CLIENT will connect mostly is not real. Neither wigle nor 3WiFi will like this.

Hcxdumptool doesn't use RSSI values. The RSSI value is a "one way" value how the interface sees the transmitter. But we are packet oriented and it is important how the target receive hcxdumptool. So we measure this. I mean we can have a position on which we receive an AP, but from this position the AP doesn't receive us.

@RealEnder you know what I mean with CLIENT oriented - same MAC on different NETWORKs using the same ESSID or different MACs on the same NETWORK). As far as I know, only hcxdumptool/hcxtools follows this philosophy.
 
To add this feature, both tools must receive massive changes.

Czechball, 2020-12-25:
But hcxdumptool marks rouge proberequests, so it's possible to only extract real station Mac addresses, right?

Czechball, 2020-12-25:
I mean, everything we really care about when importing to Wigle are beacons...

ZerBea, 2020-12-25:
Unfortunately not. Example: 
CLIENT A search AP HOME by directed proberequest on position A:
11:11:11:11:11:11 <- AA:AA:AA:AA:AA:AA HOME
hcxdumptool will store this combination (MAC_AP/ESSID) for in internal AP list: 11:11:11:11:11:11 HOME

CLIENT B search AP HOME by undirected proberequest on position B:
11:11:11:11:11:11 <- AA:AA:AA:AA:AA:AA HOME
now hcxdumptool will use the previous stored MAC_AP/ESSID to retreive M2 and or PMKID from the CLIENT. 

We can only use the BEACON or the PROBERESPONSE from a real AP to convert to wigle.
All other values are randomized and only hcxdumptool will know this (to prevent countermeasures against us).

ZerBea, 2020-12-25:
"I mean, everything we really care about when importing to Wigle are beacons..."
Correct, but hcxpcapngtool doesn't know this. So hcxdumptool must notify hcxpcapngtool (via comment field) that this packet came from a real AP.

Czechball, 2020-12-25:
Okay, I think I understand. Yeah, hcxdumptool would need to mark real mac addresses somehow.

ZerBea, 2020-12-25:
Correct. Unfortunately we disregard real MACs at the moment, because we don't need them to get the PSK/PMK.
This stuff is really hard to understand, because hcxdumptool/hcxtools running a completely different attack vector than all other tools and that makes hcxdumptool/hcxtools completely different to other tools
I explained it (dynamic/static) here, too:
https://github.com/s3inlc/hashtopolis/issues/678

ZerBea, 2020-12-25:
examples:
https://wpa-sec.stanev.org/?search=Fibertel+WiFi264+2.4GHz
https://wpa-sec.stanev.org/?search=Fibertel+WiFi370+2.4GHz
I assume, wigle will not be "amused".

Czechball, 2020-12-25:
But what you linked are all real distinct APs, they just all have the same SSID... Or am I missing something?

ZerBea, 2020-12-25:
No. Some of them is the same network (same ESSID, same PSK, but different MACs - generated by hcxdumptool)
another example (same network - all MACs randomized):
https://wpa-sec.stanev.org/?search=client%3A68c63a931a2d
https://wpa-sec.stanev.org/?search=client%3Ac0e8629b572f
https://wpa-sec.stanev.org/?search=client%3A442c05e3970a

Czechball, 2020-12-25:
Still, all of these BSSIDs are on Wigle and have a pretty high QOS value as well as their lifetime dates... They look like real APs

Czechball, 2020-12-25:
If they were randomly generated by hcxdumptool, their records would only have a few minutes of lifetime on Wigle if at all

RealEnder, 2020-12-25:
Once submitted to wigle, it stays there. The only way I see is as ZeroBeat wrote is to mark the real mac address, which in case of attacking the client is simply not known. Also the RSSI value - wigle uses it to triangulate the AP more precise, based on several beacon captures. We'll need this also stored in pcapng, if happen to attack the AP.

ZerBea, 2020-12-25:
> If they were randomly generated by hcxdumptool, their records would only have a few minutes of lifetime on Wigle if at all
That is the "magic" of hcxdumptool attack vector. It looks like it's real (to prevent countermeasures against us).

Czechball, 2020-12-25:
> Once submitted to wigle, it stays there. The only way I see is as ZeroBeat wrote is to mark the real mac address, which in case of attacking the client is simply not known. Also the RSSI value - [...]

Yeah I understand this... But Kismet somehow has a way to convert its logs to Kismet CSVs from what I presume are also beacon frames, so does it get the RSSI value in a similar way to the Wigle app or do they calculate it in another way?

> That is the "magic" of hcxdumptool attack vector. It looks like it's real (to prevent countermeasures against us).
A CLIENT "sees" a real AP!
An AP "sees" a real CLIENT!

Yeah, I think I'm pretty familiar with how fake proberesponses work in hcxdumptool. But in the records you sent (for example https://wigle.net/search?netid=02%3Af0%3A21%3A04%3A5c%3Aeb), you can see in the detail view that some of the recorded sightings are from the year 2014, 2017 and even 2019. How could this possibly be a result of hcxdumptool's rogue proberesponses? You'd need to run it constantly on one place for multiple years, and somehow don't increment the random mac for every new client to get this kind of data on Wigle.

ZerBea, 2020-12-25:
This is a better example:
https://wpa-sec.stanev.org/?search=000eef625691
```
$ whoismac -m 000eef625691
VENDOR: Private (UAA), unicast
```
coming from hcxdumptool private MAC pool
https://github.com/ZerBea/hcxdumptool/blob/master/include/ieee80211.h#L880

BTW:
The tools exist since several years, before they become public:
wlandump -> wlandump-ng -> hcxdumptool

If the target CLIENT is located at the same place, hcxdumptool will get the position of the CLIENT place. It must not be identical with the AP position.
Also please notice: At the moment no tool exist that convert hcxdumptool (older wlandump and wlandump-ng) coordinates to wigle format. A user must have done this by hand.

Czechball, 2020-12-25:
Okay, I think I finally understand what you mean. Thank you for your patience and sorry if some of my questions seem stupid.

ZerBea, 2020-12-25:
No, your questions ar not stupid. Wigle support is a nice feature. If possible without changing major parts of the code, I'll try to add it.

ZerBea, 2020-12-27:
By latest commit I started to add a simple radiotap parser to get the RSSI from the device:
https://github.com/ZerBea/hcxtools/commit/f6b2e904b82e23baa9a662de3dd680a7479bd39a
Please notice that, in contrast to kismet, hcxdumptool store only one BEACON. So we will get only one wigle CSV line each network. That isn't enough for wigle to triangulate the network exactly.
It will still take awhile until wiggle support is fully functional, because the code changes are enormous.

ZerBea, 2020-12-27:
@RealEnder I decided to implement wigle output to hcxpcapngtool due to this reasons:
- it will not slow down hcxdumptool (no additional log required, no additional functions required, everything is stored to pcapng file)
- wpa-sec will be able to retrive wigle csv line by hcxpcapngtool option

ZerBea, 2020-12-27:
By latest commit, I added a new option to hcxpcapngtool (--csv)
```
--csv=<file>                       : output network information in CSV format
```
A tabulator separated standard file will be created, containing the following columns:
`YYYY-MM-DD	HH:MM:SS	MAC_AP	ESSID	CHANNEL	RSSI`
Network information means that we only use BEACONs and PROBERESPONSEs.
This function is still under development, so order and columns can change.
Also some fields are still missing (cypher suite, akm, longitude, latitude, altitude, ....). I'll add them, soon.

Nevertheless I decided not to to provide a special format, expected by wigle, 3WiFi or any other service. 
This have to be done either by the user or by the service.

ZerBea, 2020-12-30:
I currently have no GPS device available to continue the work. It would be very nice if someone can add a short hcxdumptool pcapng file with one or 2 included NMEA comment fields.

Czechball, 2020-12-30:
I can provide you some, but would it be possible to send them to you privately?

ZerBea, 2020-12-31:
That would we fine. You can get my mail address from git API:
https://api.github.com/users/ZerBea/events/public
Press Ctrl+F and search for “email”.

ZerBea, 2020-12-31:
Thanks for the example pcapng file. It looks like we have a few small differences on how the devices provide NMEA sentences. I was able to fix several other issues regarding NMEA handling.
The CSV file now contain NMEA positions, too.

ZerBea, 2021-01-01:
Some words about the difference hcxpcapngtool NMEA output versus hcxpcapngtool CSV (output):

```
NMEA
Three different lines in NMEA format (comma separated with additional checksum):
$GPRMC (coordinate)
$GPGGA (coordinate)
$GPWPL (trackpoint: MAC_AP + coordinate)
output can be used by GPS Babel to convert it to various formats accepted by nearly all tools, devices)
every MAC is logged!

CSV
One single line (tabulator separated columns)
Additional information:
ESSID
CIPHER
AUTHENTICATION MANAGEMENT
CHANNEL
RSSI
COUNTRY INFORMATION (if transmitted by the ACCESS POINT)
only real ACCESS POINTs are logged (BEACON and PROBERESPONSE frames)
output can be used by own bash scripts to be converted to user defined formats (e.g.: accepted by wigle)
```

Czechball, 2021-06-03:
Hello again. So I finally got to testing and I've noticed that the --csv option doesn't provide gps info in the csv file. Is it possible to somehow add that info so it can be used for Wigle?

ZerBea, 2021-06-03:
Should be no problem to add some more columns that contain GPS data.
I'll check this.

ZerBea, 2021-06-03:
Fixed by latest commit:
https://github.com/ZerBea/hcxtools/commit/19e848503b290f080b0ea0ca4d5c3829d23fc1f6

Now, --csv output doesn't depend any longer on --nmea option.
If GPS data is available, it will be stored to CSV file without using --nmea option.

Please test.

Czechball, 2021-06-03:
Hey, I can not thank you enough for your quick responses. Even tho I responsed to something that was inactive for months, it only took you 30 minutes. I know that I was the one suggesting this feature, but I was too busy with other things shortly after and only got back to it now. Again, seriously, thanks for all your work on hcxdumptool.

It seems like it's working, the GPS values are now in the last 2 (4) columns. I only noticed one weird thing: The decimal point in GPS values is shifted by 2 places (for example 1234.567890 instead of 12.34567890), is this on purpose?

ZerBea, 2021-06-03:
That is correct and wanted, because we take the NMEA coodinate from the device and do no math on it.
According to the specs, NMEA specifies latitude as ddmm.mmmm and longitude as dddmm.mmmm.
To convert these to decimal degrees, we need to simply divide the minute (mm.mmmm) by 60.
(d)dd + (mm.mmmm / 60)
The first "d" is optional and only pertains to longitudinal coordinates.
https://www.heystephenwood.com/2014/05/converting-nmea-latitude-and-longitude.html

Czechball, 2021-06-03:
Right. I didn't realize that Wigle accepts decimal and the csv option outputs NMEA.

ZerBea, 2021-06-03:
Correct, we do not do the conversion to decimal degrees.
BTW:
gpsbabel can do this much better than I could ever do

ZerBea, 2021-06-03:
Nearly every GPS device provide/use NMEA 0183 standard.
https://www.nmea.org/content/standards/nmea_0183_standard
https://en.wikipedia.org/wiki/NMEA_0183

Is there any reason, why wigle doesn't accept NMEA sentences/coordinates?

Czechball, 2021-06-03:
I just noticed another thing when inspecting CSVs from my older pcapng files, some records don't have a channel associated to them (the column is just empty). How is this possible?
When I looked up the APs in question on Wigle, their channels were 48, 52, 62, 100, 108, 112 and some others. (I only checked 5GHz captures)

Czechball, 2021-06-03:
> Is there any reason, why wigle doesn't accept NMEA sentences/coordinates?

Not sure.. Maybe I can ask them on their forums

ZerBea, 2021-06-03:
There are two ways to get the information about the channel:

1) from the tagged parameters of a BEACON or PROBERESPONSE
Tag: DS Parameter set: Current Channel: 7
    Tag Number: DS Parameter set (3)
    Tag length: 1
    Current Channel: 7

2) from the RADIOTAP header
Channel frequency: 2442 [BG 7]

I decided to use way 1 - channel information from BEACON or PROBERESPONSE, because this is the channel that the AP use
Way 2 is the channel on which the interface receive. Due to overlapping of channels, way 1 and way 2 may not be identical.

If an AP doesn't transmit channel information, this field may be empty.

BTW:
Can you send me such a cap file with an empty channel information, please?

ZerBea, 2021-06-03:
The same applies to 5GHz channels:

1) BEACON / PROBERESPONSE
Tag: DS Parameter set: Current Channel: 36
    Tag Number: DS Parameter set (3)
    Tag length: 1
    Current Channel: 36

2) RADIOTAP header
Channel frequency: 5180 [A 36]

Czechball, 2021-06-03:
> Can you send me such a cap file with an empty channel information, please?

Done

ZerBea, 2021-06-03:
Received it, thanks.

Czechball, 2021-06-05:
Due to the recent changes to hcxtools, it is now possible to create Wigle imports using hcxpcapngtool and [my script](https://github.com/Czechball/hcx-to-wigle) from a pcapng file. I think we can close this issue.

closed 2021-06-05 by Czechball

ZerBea, 2021-06-06:
Your script is working fine. I think this solution (one to for one purpose) is the best way.
hcxpcapngtool will convert everything from a pcapng file to a standard CSV and your tool is doing the rest.
If something changed (wigle API) you (as a wigle user) can add it much better than I ever can do.