ZerBea/hcxdumptool #160: fatal error

sai1122966, 2021-02-01:
mkdir -p .deps
make: pkg-config: No such file or directory
make: pkg-config: No such file or directory
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c   -DVERSION_TAG=\"6.1.5-30-g2ee5d91\" -DVERSION_YEAR=\"2021\"
hcxdumptool.c:49:10: fatal error: openssl/opensslv.h: No such file or directory
   49 | #include <openssl/opensslv.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:73: hcxdumptool] Error 1
help me out bro

ZerBea, 2021-02-01:
You are missing some dependencies:

for hcxdumptool:
```
* libopenssl and openssl-dev installed
* pkg-config installed
```

for hcxtools:
```
* libopenssl and openssl-dev installed
* librt and librt-dev installed (should be installed by default)
* zlib and zlib-dev installed (for gzip compressed cap/pcap/pcapng files)
* libcurl and curl-dev installed (used by whoismac and wlancap2wpasec)
* libpthread and pthread-dev installed (used by hcxhashcattool)
* pkg-config installed
```

According to README.md:
To install requirements on Kali use the following 'apt-get install pkg-config libcurl4-openssl-dev libssl-dev zlib1g-dev'

ZerBea, 2021-02-01:
Closed, because issue is related to missing dependencies, but you can still ask your questions here.

closed 2021-02-01 by ZerBea