ZerBea/hcxdumptool #161: Interface is not on expected channel

sharmaViks, 2021-02-03:
Why I am getting this error : 
[image]

upto 100 times ...

My channels are:
[image]

ZerBea, 2021-02-03:
hcxdumptool print an information:
"interface is already in monitor mode"
How did you set this monitor mode?

Please do the following steps:
remove WLAN adapter
insert WLAN adaper
stop all services that take access to the adaper (NetworkManager and wpa_supplicant).
Do not use iw to set monitor mode
run the following commands and comment output
$ sudo hcxdumptool -i wlan0 --check_driver
$ sudo hcxdumptool -i wlan0 --check_injection

If you receive that warning, hcxumptool is not able to set a channel.
This happens if NETLINK (done by iw) is in use or if another tool take access to the device.

Please notice:
The rtl88xxau driver supplied by the kernel doesn't support monitor mode and packet injection. KALI is supplying a third party kernel that depend on NETLINK. I assume it is the aircrack-ng driver from here:
https://github.com/aircrack-ng/rtl8812au
Thiss driver depend on NETLINK. It will not work with hcxdumptool:
https://github.com/aircrack-ng/rtl8812au/issues/587

ZerBea, 2021-02-03:
Are you running airodump-ng in parallel with hcxdumptool? airodump-ng use an internal channel hopper that interfere with hcxdumptool's internal channel hopper. That will not work.

ZerBea, 2021-02-03:
By latest commit, I added an information which tool try to take access to the device while hcxdumptool is running.

sharmaViks, 2021-02-04:
> Are you running airodump-ng in parallel with hcxdumptool? airodump-ng use an internal channel hopper that interfere with hcxdumptool's internal channel hopper. That will not work.

sharmaViks, 2021-02-04:
[image]

I got this  after running 

$ sudo hcxdumptool -i wlan0 --check_driver
$ sudo hcxdumptool -i wlan0 --check_injection

ZerBea, 2021-02-04:
Ok, thanks for your answers.

First of all some remarks:
airodump-ng is a passive dumper that receive packets. It doesn't transmit, but it will hop on all channels as default.
hcxdumptool is an active attack tool. It transmit and it receive packets. Also it hops on channels (depending on channel options).
Therefore after tranmitting a packet it must stay on the channel until it received the answer from the target. If another tool
changed the channel in this time, hcxdumptool will print an ERROR and terminate after 100 ERRORs, because I can't allow to spam this channels (because we will never get an answer due to third party channel change).

Your command line will not do what you expected, because you're missing some important options.
$ hcxdumptool -i wlan0 -o galleria.pcapng -c 1
will run without any status messages. If you would like to see whats going on, you must enable status messages as described in  help
```
--enable_status=<digit>            : enable real-time display (waterfall)
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
                                     bitmask:
                                         0: no status (default)
                                         1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
                                        32: GPS (once a minute)
                                        64: internal status (once a minute)
                                       128: run as server
                                       256: run as client
                                       512: EAP
                                      1024: EAP NAK
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```
$ hcxdumptool -i wlan0 -o galleria.pcapng -c 1 --enable_status=31
will show you what's going on.

You received 2 warnings that other services take access to the device and hcxdumptool is not able to control it:
warning: NetworkManager is running with pid 941
warning: wpa_supplicant is running with pid 1548

You have to stop them before you start hcxdumptool:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service

Don't forget to restart the 2 services when hcxdumptool terminates, otherwise KALI will not have access to the devices.
$ sudo systemctl start NetworkManager.service
$ sudo systemctl start wpa_supplicant.service

Please notice:
hcxdumptool and hcxtools are analysis tools and totally different to aircrack-ng suite.
I strongly recommend to forget all what you know from aircrack-ng, because that procedure will not work on hcxdumptool/hcxtools/hashcat and/or john!
Also don't expect that the driver is working as expected, mostly due to its NETLINK dependency but also several issues:
https://github.com/aircrack-ng/rtl8812au/issues
https://github.com/aircrack-ng/rtl8814au/issues
https://github.com/aircrack-ng/rtl8188eus/issues
Christian is doing a great job on this drivers, but they are outdated. Successor will be this one after leaving alpha status:
https://github.com/kimocoder/realtek_rtwifi

ZerBea, 2021-02-04:
Closed this issue because it is not related to hcxdumptool, but you can still ask your questions here.
airodump-ng is running in parallel with hcxdumptool,  took access to the device and changed channels.
It is recommended in help not to do this:
```
usage  : hcxdumptool <options>
         press ctrl+c to terminate hcxdumptool
         press GPIO button to terminate hcxdumptool
         hardware modification is necessary, read more:
         https://github.com/ZerBea/hcxdumptool/tree/master/docs
         do not set monitor mode by third party tools
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon)
         do not run hcxdumptool in combination with tools (channel hopper), that take access to the interface (except: tshark, wireshark, tcpdump)
```

closed 2021-02-04 by ZerBea

sharmaViks, 2021-02-04:
After Performing the commands as said. I got this driver error now.

ZerBea, 2021-02-04:
Can you please comment the driver ERROR message?

thomasf1392, 2022-02-16:
Stop airopdump-ng before you start hcxdumptool.

ZerBea, 2022-02-16:
Aircrack-ng suite is really a great suite, especially for beginners and all included tools work perfect with each other.
But if you run hcxtools/hcxdumptool in combination with them it will lead to unexpected results.