ZerBea/hcxdumptool #162: hashcat potfile, discard found items - suggest

hbith, 2021-02-10:
Hi!
my suggest is to include an option in hcxdumptool to discard APs that are already found and we have the pass. I believe that the hashcat potfile could be used.

thanks!

ZerBea, 2021-02-10:
hcxdumptool has 2 options to discard targets:
filterlists in combination with filter options
and
BPF code

Filterlists are working as attack filters (in transmit branch).
BPF code is working in transmit branch and in receive branch.

Please notice:
If the target activated MAC randomization, filterlists are useless.

Is there any reason to add another list (which would slow down hcxdumptool in a huge way)?

ZerBea, 2021-02-10:
I can add a filtermode (hcashcat potfile) to hcxpcapngtool and hashashtool, too.
Than this targets will not longer converted to a has file.

ZerBea, 2021-02-10:
Please notice: 
We are online and all functions inside hcxdumptool are time-critical. We are counting in cpucycles/frame and usec values!

Here is an example:
$ tshark -r authentication.pcapng -T fields -E header=y -e frame.number -e frame.time 
1	Nov 10, 2017 09:54:51.054512000 CET <- AUTHENTICATIONREQUEST
2	Nov 10, 2017 09:54:51.054880000 CET <- ACK
3	Nov 10, 2017 09:54:51.056408000 CET <- AUTHENTICATIONRESPONSE
4	Nov 10, 2017 09:54:51.056630000 CET <- ACK

If we receive frame 1, we must evaluate it and decide what to do before the next frame arrived!
Do you really think we can browse a hashcat potfile for ESSIDs, search the matching MAC(s) for this ESSID and respond in time without massive packet loss?

hbith, 2021-02-10:
I've made a mistake, I'm trying to say the hascat outputfile, -o option, that includes client and AP mac addresses.
If hcxdumptool tests the passwords online and have success  could drop the search for that AP.

ZerBea, 2021-02-10:
PBKDF2 is an extreme slow process. We can't do that on the fly.
Also hcxdumptool doesn't search for a target. hcxdumptool receive a frame, evaluate an incoming frame and decide what to to.
What you described is extactly the function of the MAC filter lists and the BPF code. This functiona are implemented:
If you don't want to attack an AP, put the MAC in the filter list AP  and use filter mode 2.
The AP will not be attacked
If you don't want to attack a CLIENT, put the MAC in the filter list CLIENT and use filter mode 2.
The CLIENT will not be attacked (as long as he use this MAC and not a randomized one).
Please note that this filter lists are limited (256 entries) to avoid packet loss.

As an alternative you can use BPF code. MACs from this code are ignored completey during receive time.
Why do we need another filterlist? You can use simple bash code to convert the output of hashcat -o to a filterlist.

ZerBea, 2021-02-11:
I started to add PSK/MPK check to hcxhashtool by options --psk and --pmk.
We only test for a single PMK or a single PSK, because this calculations is slow on CPU.
If I find a way to increase speed I'll add pmk check from a potfile, too.
```
--psk=<PSK>                  : pre-shared key to test
                               due to PBKDF2 calculation this is a very slow process
                               no nonce error corrections
--pmk=<PMK>                  : plain master key to test
                               no nonce error corrections
```
Please notice, we are neither hashcat nor john and doing this on CPU, only.

ZerBea, 2021-02-11:
This is an example about the speed impact, when calculating PBKDF2 on a hashfile on CPU:
```
$ time hcxhashtool -i test.22000 --psk=ABCDEFGH

OUI information file...: /home/zerobeat/.hcxtools/oui.txt
OUI entires............: 29053
total lines read.......: 18378
valid hash lines.......: 18378
PMKID hash lines.......: 3268
EAPOL hash lines.......: 15110

real	0m28,495s
user	0m28,478s
sys	0m0,007s
```
The speed impact is huge. And we are only calculating a single PSK, not more.
Now you can imagine what would happen, if we try to calculate all PSKs from a hashcat potfile.
There is no more time to evaluate incoming frames and to respond the target.

hbith, 2021-02-12:
My proposal is related to improve hcxdumptool performance by discarding frames of bssid  already known in a previous session and of  the pair "client mac address-ssid" that we already have the password.
If I run multiple sessions of hcxdumptool-hashcat  in a dense wifi area this way is like an autolearn from session to session that could move the capture to the new ones forgetting the old known ones.

ZerBea, 2021-02-12:
Running incoming frames through a filter list will decrease performance.
Please notice:
In every case hcxdumptool must receive a frame. And it is much, much faster to write this frame directly to a pcang file than running it through a MAC filter list. We simply have not the time to do this until the next frame arrived.
On the other hand, it looks totally different when transmitting a frame. Here we have much more time and decide by filterlist to attack or not to attack.
In other words:
If we analyze every incoming frame (and verify it against a filter list), we will loose many, many incoming packets.

BTW:
auto reload of filter lists is allready implemented since this PR from 12.07.2020:
https://github.com/ZerBea/hcxdumptool/pull/123

ZerBea, 2021-02-12:
The idea behind hcxtools:

hcxdumptool (time sensitive)
receive every thing on the channel and save it to pcapng, regardless what it is (that is ultra fast)
attack everything except it is in a filterlist
you can make hcxdumptool complete silent in transmit branch, but you can't prevent that it doesn't receive if it is running

hcxpcapngtool (we have all the time in the world)
convert everything to a hashfile that hashcat/jtr accept

hcxhashtool (we have all the time in the world)
filter the hash file to that what you want or to increase cracking speed (due to reuse PBKDF2 for example)

worflow:
hcxdumptool -> hcxpcapngtool -> hcxhashtool  -> hashcat/john

ZerBea, 2021-02-12:
To prevent countermeasures against us, we are fully randomized on every session. That is the reason why I didn't implemented to restore a session.
Example (same AP, same ESSID, but different MACs):
https://wpa-sec.stanev.org/?search=Fibertel+WiFi298+2.4GHz
https://wpa-sec.stanev.org/?search=Hausboot

BTW:
Most of the CLIENTs do the same to prevent tracking

Implementing your feature request would slow hcxdumptool down.

hbith, 2021-02-12:
You are the creator and the expert and I believe you....
but my intuition is that computers write now could process 10 Gigabit ethernet  -with throughput of near 4Gigabits- and Wifi is so much slow..... I will make some experiments
;)

ZerBea, 2021-02-20:
I added a new repository:
https://github.com/ZerBea/wifi_laboratory
It contain a skeleton that allow several tests. It also include a "lesson learned".
Feel free to use it and to add your own ideas.

Your intuition about computers is absolutely correct and I fully aggree
.
But we are talking about 802.11 protocol (packet oriented) that has its own rules. 
That include several EAPOL timers:
```
EAP-Identity-Request Timeout
EAP-Identity-Request Max Retries
EAP-Request Timeout (seconds)
EAP-Request Max Retries
EAPOL-Key Timeout
EAPOL-Key Max Retries
``` 
If we receive a frame, we must (mandatory) respond within a specified period of time. Otherwise the target discard us.

Also we are talking about the  specific characteristics of the RF spectrum (2.4GHz and 5/6 Ghz).
And we are talking about a small system like a Raspberry Pi or an OpenWRT device.

If we start an attack, we absolutely know nothing about the target. 

For your tests, I recommend to use a spectrum analyzer, to monitor what's going on on the RF channel.
Also I recommend to run Wireshark or tshark in parallel to monitor the traffic on the interface (request - respond times).

ZerBea, 2021-02-20:
I'll don't close this feature request. Maybe you/we find a solution.
I started some tests, using the WiFi laboratory skeleton.

ZerBea, 2021-02-20:
Please also notice that this feature is partly added. To use it:
create an ap_filter list with some initial entries

run hcxdumptool with options --filtermode=1 --filterlist_ap=filter.list

start hashcat and monitor -o output and get networks of which the PSK is recovered:
catnewmac | awk 'BEGIN { FS = ":" } ; { print $3F }' | sort | uniq > filter.list

send SGIHUB to hcxdumptool to reload the filterlist:
https://bash.cyberciti.biz/guide/Sending_signal_to_Processes

ZerBea, 2021-03-26:
After several tests, I don't got a solution to verify hashcat potfile entries online.
Please notice:
Right now, we verify the if the PMK is zeored or not. That is the only (slow) hash calculation we can do.
Every additional calculation will end in a huge packet loss.

closed 2021-03-26 by ZerBea