ZerBea/hcxdumptool #167: --essid-part Partial ESSID filterlist - request

ChemicalNRG, 2021-03-17:
Can you add a partial essid filterlist like hcxhashtool has?
I want to combine hcxdumptool with attacks on essid's (on open networks, so i can and have to make a connection) with partial known name like TP-LINK_
And because i am using macchanger i cant put the mac of my wireless card on the client filter.
So now i have to disable client attacks which results in far less PMKID's.

My suggestion would be:
--essid-part=partialESSIDfilterlist.txt 
example: TP-LINK_

ZerBea, 2021-03-17:
Unfortunately this is not feasible because:
Only a few 802.11 frames contain an ESSID (BEACON, PROBERESPONSE, PROBEREQUEST, ASSOCIATIONREQUEST, REASSOCIATIONREQUEST and some ACTION frames).
If we add an ESSID filter, we have to add an additional ring buffer.
If we receive a frame that doesn't contain an ESSID, we have to walk through this ring buffer to get the ESSID. Than we have to search for this ESSID in the ESSID filter list.
That will result in a huge packet loss.

ChemicalNRG, 2021-03-17:
a partial mac filter list or a vendor filter would do the job too, if that could be done?

JuniorJPDJ, 2021-03-17:
> a partial mac filter list or a vendor filter would do the job too, if that could be done?
> I think a vendor filter would make sense because then its also possible to only attack/monitor vendors with weak default passwords or skip the vendors with strong default passwords

This is already implemented, you can write BPF for it

ZerBea, 2021-03-18:
As @JuniorJPDJ mentioned, this feature is already implemented by BPF option.
http://www.tcpdump.org/manpages/pcap-filter.7.html
The BPF is running in kernel space. It is ultra fast. You can filter nearly everything.

Please notice:
- hcxdumptool is running randomized MACs (on both attack vectors: as AP and as CLIENT).
- nearly every CLIENT is running randomized MACs.
In that case, do not expect that a filter is working as expected. If a CLIENT is probing using a randomized MAC which is not in the list, hcxdumptool will respond.

BTW:
You mentioned macchanger. Do not use macchanger in combination with hcxdumptool, because hcxdumptool is using an own MAC address space.
Also do not use other tools (except tshark and Wireshark) which take access to the attack device, while hcxdumptool is running on it. That will result in a packet loss.

But let me think about a soft coded VENDOR filter which is fast enough to handle this. I have an idea.

ZerBea, 2021-03-18:
Added VENDOR filter options by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/ad855d225f60c733ec3b6a11e8515a5d6a29e97e

closed 2021-03-18 by ZerBea