ZerBea/hcxdumptool #171: compile error with musl libc

rofl0r, 2021-05-02:
```
hcxdumptool.c: In function 'main':
hcxdumptool.c:8800:11: error: assignment of read-only variable 'stdout'
    stdout = fopen("/dev/null", "w");
           ^
make: *** [hcxdumptool] Error 1
```
according to C standard, stdout pointer is read only and may not be changed.

i suggest using `FILE* out` instead of stdout and assign it to stdout on startup.

ZerBea, 2021-05-02:
Please give me some time to check it.

n0w1re, 2021-05-02:
Ok, that was totally my fault. I already sorta fixed that issue in #168, but left the same construct at another place at it was. It was also pointless at that place ...

@rofl0r Please try compiling again with your setup with PR #174.

ZerBea, 2021-05-02:
Merged, thanks.

PR #174 (2021-05-02): fixed reassignment of stdout in client mode
- removed pointlessly doubled stdout reassignment that used non-portable way of assinment as already solved in #168
- fixed writing pcapng to stdout in client mode

fixes #171
closed 2021-05-02 by ZerBea