ZerBea/hcxdumptool #172: doesn't work with libressl 3.1.4

rofl0r, 2021-05-02:
```
hcxdumptool.c: In function 'detectweakwpa':
hcxdumptool.c:3802:9: warning: implicit declaration of function 'EVP_PKEY_new_CMAC_key' [-Wimplicit-function-declaration]
  pkey = EVP_PKEY_new_CMAC_key(NULL, testptk, 16, EVP_aes_128_cbc());
....
/tmp/cccLFHAN.o: In function `detectweakwpa':
hcxdumptool.c:(.text.unlikely+0x171a): undefined reference to `EVP_PKEY_new_CMAC_key'
collect2: error: ld returned 1 exit status
make: *** [hcxdumptool] Error 1
```

rofl0r, 2021-05-02:
the function seems to be also removed in latest openssl: https://github.com/openssl/openssl/issues/13628

ZerBea, 2021-05-02:
This function is missing in libressl:
EVP_PKEY_new_CMAC_key

There is an issue report here:
https://github.com/libressl-portable/portable/issues/651
which received a fix here:
https://github.com/libressl-portable/portable/issues/651#issuecomment-822630295

more information available here:
https://github.com/ZerBea/hcxtools/issues/184

So we can close this issue report.

rofl0r, 2021-05-02:
did you see my [last comment](https://github.com/ZerBea/hcxdumptool/issues/172#issuecomment-830818030) ? this needs to be fixed anyway as soon as you upgrade openssl

ZerBea, 2021-05-02:
OK, I'll check the replacement.

ZerBea, 2021-05-02:
They only write that the function is deprecated:
https://www.openssl.org/docs/manmaster/man3/EVP_PKEY_new.html
But I don't fin the replacement, yet.

ZerBea, 2021-05-02:
Ok,  I'm going to move all functions to new EVP_MAC API:
https://www.openssl.org/docs/manmaster/man3/EVP_MAC.html

That will take a while. At least until OpenSSL 3.0 release date:
https://wiki.openssl.org/index.php/OpenSSL_3.0_Release_Schedule

Until than, I recommend to use the current development release of libressl, because it is confirmed to let hcxtools/hcxdumptool compile without errors:
https://github.com/libressl-portable/portable/issues/651#issuecomment-822630295

ZerBea, 2021-05-03:
You may have noticed, that I changed the topic, because the origin issue is fixed by LibreSSL developers and will find the way into the next relase. 
Next step is to move all HMAC/CMAC functions to the new EVP_API. Doing this, hcxdumptool/hcxtools will not work on older versions of the crypto library any more. So it is mandatory to wait until the final release of OpenSSL 3.0.

ZerBea, 2021-05-05:
If you would like to use hcxdumptool on a small device, I suggest to take a look at
https://github.com/ZerBea/wifi_laboratory

Remove the debug option from Makefile:
CFLAGS	= -O3 -Wall -Wextra
Than compile / install hcxlabseries.

Create a BPF (either to attack or to protect) and run thew desired tool:

1. hcxlabgetm1 -> request PMKIDs
2. hcxlabgetm2 -> request M2 from CLIENTs
3. hcxlabgetm1234 -> get PMKIDs M1...M4 from connected CLIENTs
4. hcxlabgetmall -> do attack vector 1,2 and 3
5. hcxlabdumpall -> dump all to pcapng

The lab series is a little bit more reacitve, a little bit faster and much more aggressive.
The tools are designed to run on smaller devices. They have less dependencies.

Please read more here:
https://github.com/ZerBea/hcxdumptool/issues/169

rofl0r, 2021-05-05:
> The lab series is a little bit more reacitve, a little bit faster and much more aggressive.
> The tools are designed to run on smaller devices. They have less dependencies.

thanks, good to know. however i do run both musl and libressl as my libc and ssl lib of choice on my desktop because they're technically superior, not because of hardware constraints. my distro is http://sabo.xyz btw.

ZerBea, 2021-05-06:
Thanks for the information about your distribution.
I'm running Arch Linux on all my systems:
https://archlinux.org/
https://archlinuxarm.org/packages
https://archlinux32.org/

Arch Linux is a Rolling Release distribution and all hcxtools, which are part of this distribution, follow this policy:
https://archlinux.org/packages/?q=hcx
https://archlinuxarm.org/packages/aarch64/hcxdumptool
https://archlinuxarm.org/packages/aarch64/hcxtools
https://archlinux32.org/packages/?q=hcx
They depend on standard libraries, supplied with Arch Linux. This implies that, it is mandatory to update to latest versions of drivers and libraries if you decide to use latest git head of hcxdumptool/hcxtools.

ZerBea, 2021-05-06:
The same applies to the development of the Linux kernel.
Let me give you an example:
A few days ago, we noticed that a driver ignore hcxdumptool/hcxlabtool timestamp settings.
We contacted the maintainer of the driver and he applied a patch for the upcoming kernel:
https://patchwork.kernel.org/project/linux-wireless/list/?series=476707

If you use a Ralink device (rt2800usb driver), it is mandatory to update to the latest kernel, to get benefit of this fix.

rofl0r, 2021-05-06:
thanks for info.

> This implies that, it is mandatory to update to latest versions of drivers and libraries if you decide to use latest git head of hcxdumptool/hcxtools.

well, my distro is rolling too, but i dont have neither the resources nor the will to always have bleeding edge of everything. i'd think there's little cost of additionally supporting older library versions. arch linux is a pretty decent distro, but the beauty of linux is that there's a lot of diversity and a gazillion of options to setup your userspace.

> A few days ago, we noticed that a driver ignore hcxdumptool/hcxlabtool timestamp settings.

good find, and thanks for raising my awareness of this. in sabotage, we usually backport fixes like that to the long-time-support kernels we support, which are currently 4.4, 4.9, 4.14, and 5.4. so just because uname -a tells you that the kernel isn't the latest and greatest, it doesn't actually mean that a specific issue isn't fixed using a patch (or maybe the issue didn't even exist in older versions to begin with).

ZerBea, 2021-05-06:
It shouldn't be a big problem, because latest relase of hcxdumptoo/hcxtools will work on latest release of the distribution and libraries/drivers.

BTW:
LibreSSL 3.3.3 was released:
https://github.com/libressl-portable/portable/releases/tag/v3.3.3
hcxdumptool should compile on this version

Now I wait for release of hashcat 6.2.0 and gcc 11.1. Than it's time for v6.2.0 of hcxdumptool/hcxtools, too.

BTW:
> ...there's little cost of additionally supporting older library versions

For sure, it is - but I don't want do code complete crypto functions for old libraries that doesn't include them by default.

ZerBea, 2021-05-06:
The recommended system for hcxdumptool is:
```
a cheap RaspBerry Pi Zero,
2 resistors,
1 push button,
1 low power LED,
20 cm wire,
a cheap WiFi adapter (Mediatek or Ralink chipset)
and a case
```
all in all less than 15,- Euro.

ZerBea, 2021-05-06:
I'm sure, LibreSSL will use EVP_MAC in future times, too. The developer of LibreSSL added the missing function (EVP_PKEY_new_CMAC_key) shortly after the issue report.
I don't expect a showstopper in future times (on LibreSSL and musl libc).

ZerBea, 2022-11-15:
Done by this commits:
EAPOL calculation:
https://github.com/ZerBea/hcxdumptool/commit/87fcd83f2bf109bd8a924209cc9af338ed50169a
PMKID calculation:
https://github.com/ZerBea/hcxdumptool/commit/29c1743486c34ae7f8382597505a2d0f68d7dd4a

closed 2022-11-15 by ZerBea