ZerBea/hcxdumptool #177: problems with check_injection

romeueff, 2021-06-18:
I've been trying for a few days to make hcxdumptool work, I noticed that in the command
sudo hcxdumptool -i wlan0 --check_injection
doesn't bring the ratio

└─# sudo hcxdumptool -i wlan0 --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!

terminating...
-----------------------

# lsusb                                                                                                          127 ⨯
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
---------------------
└─# hcxdumptool -I
wlan interfaces:
00e2b01118de wlan0 (mt7601u)
---------------------------

└─# sudo hcxdumptool -i wlan0 --check_driver                                                                         1 ⨯
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
-----------------
└─#  uname -r
5.10.0-kali7-amd64
--------------------

I'm using virtualbox with kali
what could be happening?

ZerBea, 2021-06-18:
Please comment output of hcxdumptool -v
```
$ hcxdumptool -v
hcxdumptool 6.2.0-7-ga124c1a (C) 2021 ZeroBeat
```
Your status output looks like your're running an old version of hcxdumptool. Which version of hxcdumptool do you use?
Running this version, output should look like this  (on mt76 device;
```
$ hcxdumptool -I
wlan interfaces:
74da38f2038e wlp5s0f4u2 (mt7601u)

$ sudo hcxdumptool -i wlp5s0f4u2 --check_injection
initialization...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13
packet injection is working on 2.4GHz!
injection ratio: 44% (BEACON: 296 PROBERESPONSE: 131)
your injection ratio is average, but there is still room for improvement
antenna ratio: 75% (NETWORK: 16 PROBERESPONSE: 12)
your antenna ratio is excellent, let's ride!

terminating...
```

ZerBea, 2021-06-18:
You may have noticed that your (old version status output looking so::
initialization...
starting packet injection test (that can take up to two minutes)...

while latest version looking this way:
initialization...
starting antenna test and packet injection test (that can take up to two minutes)...

romeueff, 2021-06-24:
Good morning friend, I'm sorry for the delay I didn't see your answer,

it really was with the old version 6.0

romeueff, 2021-06-24:
# sudo hcxdumptool -i wlan0 --check_injection                                                                 1 ⨯
initialization...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14
packet injection is working on 2.4GHz!
injection ratio: 48% (BEACON: 37 PROBERESPONSE: 18)
your injection ratio is average, but there is still room for improvement
antenna ratio: 100% (NETWORK: 3 PROBERESPONSE: 3)
your antenna ratio is huge - say kids what time is it?

ZerBea, 2021-06-24:
Looking pretty good, so far.
Indeed, v6.0 is very old.