ZerBea/hcxdumptool #178: What am i doing wrong?

awesomedonald, 2021-06-23:
I cant figure it out. Can you teach me?
[image]

ZerBea, 2021-06-23:
By default, hcxdumptool doesn't show a real time status.
If you want to see what's going on, you have to enable the waterfall status by option "enable_status":
```
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
                                     bitmask:
                                         0: no status (default)
                                         1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
                                        32: GPS (once a minute)
                                        64: internal status (once a minute)
                                       128: run as server
                                       256: run as client
                                       512: EAP
                                      1024: EAP NAK
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```
Just add --enable_status=15 to your command line and you'll see what hcxdumptool is doing.
Also I recommend to do a rca scan first to determine the AP working channel.
Than restart your attack and set AP working channel by -c option.

Remove option -f 79, because it is not mandatory for your attack. Using default settings, hcxdumptool will save all frames to required by hcxpcapngtool to convert to hashcat format.

example command line:
$ hcxpcapngtool -i wlp2s0f0 -o dump.pcapng --filterlist_ap=myfilter.txt --filtermode=2 --enable_status=31

ZerBea, 2021-06-23:
Now wait until you see a message like this:
```
20:04:08  11 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [EAPOL:M1M2 EAPOLTIME:2214 RC:8 KDV:2]
20:04:08  11 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [EAPOL:M2M3 EAPOLTIME:9680 RC:9 KDV:2]
20:04:08  11 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [EAPOL:M3M4ZEROED EAPOLTIME:7079 RC:9 KDV:2]
or
20:04:08  11 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [EAPOL:M1M2ROGUE EAPOLTIME:1214 RC:8 KDV:2]
or
20:07:48   6 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [PMKIDROGUE:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx KDV:2]
or
20:07:48   6 xxxxxxxxxxxx xxxxxxxxxxxx TARGET_ESSID [PMKID:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx KDV:2]
```

ZerBea, 2021-06-23:
Closed this report, because it is not an issue, but you can still ask your questions here.

closed 2021-06-23 by ZerBea

awesomedonald, 2021-06-24:
Now I am getting this

08:49:01   6 ffffffffffff 7a83c23a58c6 KD-145834 [BEACON]
08:49:03   6 7acb66e09b89 7a83c23a58c6 KD-145834 [PROBERESPONSE]

ZerBea, 2021-06-24:
This is the result, we expected to see.
Now wait for  a PMKID if the AP will suport this.
If the AP doesn't support PMKIDs, you have to wait for a CLIENT to get a 4way handshake.

BTW:
Have you checked that packet_injection is working?
$ sudo hcxdumptool -i wlp2s0f0 --check_injection

awesomedonald, 2021-06-24:
Just updated

09:34:35   6 e6c19793a6f3 7483c23a58c6 THBB [PMKID:e5502c0dc8017b155476e4a1cfdd71d6 KDV:2]
09:34:35   6 e6c19793a6f3 7483c23a58c6 THBB [EAPOL:M1M2 EAPOLTIME:18458 RC:1 KDV:2]
09:34:35   6 e6c19793a6f3 7483c23a58c6 THBB [EAPOL:M2M3 EAPOLTIME:132 RC:2 KDV:2]
09:34:35   6 e6c19793a6f3 7483c23a58c6 THBB [EAPOL:M3M4ZEROED EAPOLTIME:1405 RC:2 KDV:2]

ZerBea, 2021-06-24:
Great, you got a PMKID and a full 4way handshake.

Now you can terminate hcxdumptool and try to recover the PSK.

by wpa-sec:
$ wlancap2wpasec yourdumpfile.pcapng
or
$ hcxpcapngtool -o hash.22000 -E wordlist yourdumpfile.pcapng
$ hashcat m- 22000 hash.22000 wordlist
or
$ hashcat m- 22000 hash.22000 your wordlist
or
$ hcxpsktool -c hash.22000 | hashcat m- 22000 hash.22000
or using masks, rules, other hcxpsktool options...

awesomedonald, 2021-06-24:
What do you mean attack filter?

My new output

awesome@awesome-HP-Pavilion-11-x360-PC:~$ sudo wlancap2wpasec dump.pcapng
connecting to https://wpa-sec.stanev.org
uploading dump.pcapng to https://wpa-sec.stanev.org
reading from phpk25HDa...

summary capture file
--------------------
file name................................: phpk25HDa
version (pcapng).........................: 1.0
operating system.........................: Linux 5.8.0-59-generic
application..............................: hcxdumptool 6.2.0-14-g2261bfc
interface name...........................: wlp2s0f0
interface vendor.........................: 2c337a
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 68e1661dfc59 (incremented on every new client)
MAC CLIENT...............................: d85dfbf095c6
REPLAYCOUNT..............................: 62345
ANONCE...................................: f902c7ded4daed543106a449e6c9672747f0c8d89a810f8ef4f8d1ecb2f7127c
SNONCE...................................: eb8f9ec84a64db851f39bd53efbedbe1290fd01d3308d5bf141ac239b69a205a
timestamp minimum (GMT)..................: 23.06.2021 17:52:19
timestamp maximum (GMT)..................: 23.06.2021 17:54:50
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 378
BEACON (total)...........................: 41
PROBEREQUEST.............................: 6
PROBEREQUEST (directed)..................: 1
PROBERESPONSE............................: 36
AUTHENTICATION (total)...................: 13
AUTHENTICATION (OPEN SYSTEM).............: 13
IPv4 (total).............................: 31
UDP (total)..............................: 31
EAPOL messages (total)...................: 242
EAPOL RSN messages.......................: 242
ESSID (total unique).....................: 37
EAPOL ANONCE error corrections (NC)......: not detected
EAPOL M1 messages (total)................: 242
PMKID (total)............................: 1
PMKID (best).............................: 1
PMKID ROGUE..............................: 1
PMKID written to old format (1680x)......: 1

session summary
---------------
processed pcapng files................: 1

awesomedonald, 2021-06-24:
Downloading cracked.txt.gz
and rockyou.txt

working on a slow connection.

awesomedonald, 2021-06-24:
hashcat m- 22000 hash.22000 cracked.txt.gz

Can you break this down?

Sorry i know a lot of questions. Learning or at least trying to.

ZerBea, 2021-06-24:
hcxdumptool is able to use 2 different kinds of filters.
attack/protect filter in combination with filter mode:
All traffic is displayed
Attack on targets depend on entries of the filter lists

Berkeley Packet Filter (BPF)
attack/protect and display filter
Only traffic depending on the BPFC (Berkeley Packet Filter Code) is displayed
Attack on targets depend on BPFC.

hashcat has a new mode to work on EAPOL and PMKIDs. This mode is called 22000 and it replace deprecated modes 2500 and 16800

ZerBea, 2021-06-24:
BTW:
There is no need to run hcxtools as super user (sudo).
It is mandatory only for hcxdumptool to get access to the device.

awesomedonald, 2021-06-24:
Hash 'hash.22000': Separator unmatched
No hashes loaded.

???

ZerBea, 2021-06-24:
That is a hashcat error message. Make sure you're running latest hashcat version. 
Please comment your commandlines. They should look like this:

Convert:
$ hcxpapngtool -o hash.22000 -E wordlist dumpfile.pcapng
$ hashcat -m 22000 hash.22000 wordlist

awesomedonald, 2021-06-25:
Finally getting somewhere.

[image]

ZerBea, 2021-06-25:
Glad to hear this.
But don't be disappointed, if you are not able to recover the PSK by the rockyou list.
Rockyou is a very (very) old list and not related to WiFi.
Please read more here:
https://hashcat.net/forum/thread-10151-post-52834.html#pid52834

awesomedonald, 2021-06-25:
Which one have you had the most luck with?

ZerBea, 2021-06-25:
That depend on the target.
If you really want to try some lists, take a look at this ones:
https://wpa-sec.stanev.org/dict/cracked.txt.gz (updated on every new finding)
https://wpa-sec.stanev.org/dict/rkg.txt.gz (updated on every new finding)
https://3wifi.stascorp.com/ (3WiFi list updated every 3 month)
and
https://github.com/routerkeygen/routerkeygenPC
and
hcxtools