ZerBea/hcxdumptool #179: Alfa AWUS036ACM takes an awful long time to start up

ukscone, 2021-06-23:
I picked it an Alfa AWUS036ACM as it was on the "thought to work list" as my v1 TL-WN722 v1's are now troublesome and it works but it takes an awful long time to start up. A couple of minutes at least from pressing enter on the command line to getting the settings list plus start of scan. Is this a known thing for everyone, something wrong with the linux distro, i use raspbian/pios or something i'm doing wrong

ZerBea, 2021-06-24:
Yes, the driver will take a long time for the init process.
Additional hcxdumptool will take some additional time, too, because the interface support many channels and we check every channel on start up.
There is nothing I can do to speed up this init process.

$ time sudo hcxdumptool -i wlp5s0f4u2
initialization...
real	0m49,808s
user	0m0,028s
sys	0m0,526s

ukscone, 2021-06-24:
ah ok. it's a nice adapter and seemed to work well it was just I was a little concerned at the startup time. 

thanks

closed 2021-06-24 by ukscone

ZerBea, 2021-06-25:
Yes, this adapter is a "beast".
`$ hcxdumptool -i wlp5s0f4u2 -C`
will give you a long list with available channels. beside calibration, this will take a while, too.

Injection works perfectly (on both bands) if you set wireless regulatory to IN.
either pemanent:
```
$ cat /etc/conf.d/wireless-regdom | grep IN
WIRELESS_REGDOM="IN"
```

or via iw reg set
```
phy#0
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (N/A)
	(5470 - 5725 @ 160), (N/A, 24), (N/A)
	(5725 - 5875 @ 80), (N/A, 30), (N/A)
```

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.