ZerBea/hcxdumptool #180: Is broadcom not supported or its some other error? thx

kocoman1, 2021-06-24:
PHY	Interface	Driver		Chipset

phy0	wlp4s0		wl		Broadcom Inc. and subsidiaries BCM4331 802.11a/b/g/n (rev 02)

		(experimental wl monitor mode vif already enabled for [phy0]wlp4s0 on [phy0]prism0)
------
airmon-ng works fine
---

 hcxdumptool --check_driver -i prism0

initialization...
--
failed to detect wlan interface: Operation not supported
warning: failed to init socket
 
terminating...
1 driver error encountered
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device

ZerBea, 2021-06-24:
None of the system calls (which we need to perform fast attacks) is supported by the "experimental" BROADCOM driver
The same applies to packet injection:
"Packet injection (with radiotap; no FCS injection currently though hardware supports it - a radiotap flag is being discussed for this) "
http://linuxwireless.sipsolutions.net/en/users/Drivers/b43/__v266.html

Additional it looks like you're running a virtual interface (prism0) instead of the pyh interface wlp4s0.

From --help:
         do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon) created by airmon-ng and iw
 
and from README.md:
Not recommended WiFi chipsets:
Intel PRO/Wireless
Broadcom
Realtek RTL8811AU, RTL8812AU, RTL 8814AU

Just run hcxdumptool on the phy interface and you'll get several warnings, that the driver doesn't support the modes we need for our attakcs:
$ sudo hcxdumptool -i wlp4s0 --check_driver
$ sudo hcxdumptool -i wlp4s0 --check_injection

ZerBea, 2021-06-24:
But anyway, by latest commit I added an additional warning if a virtual prism interface is detected:
https://github.com/ZerBea/hcxdumptool/commit/a42fc84de3875150760ffeeab163f05b9bccf353

BTW:
I interested in the ERROR messages and WARNINGs, you'll receive if you try to set phy monitor mode by hcxdumptool or if you try to inject packtets.
Please comment output of (please do not use a third party tool to set monitor mode, before):
$ sudo hcxdumptool -i wlp4s0 --check_driver
$ sudo hcxdumptool -i wlp4s0 --check_injection

ZerBea, 2021-07-05:
According to README.md and help, virtual(logical interfaces are not supported (and will not be supported in future times) because the are unusable for our purpose.

closed 2021-07-05 by ZerBea