ZerBea/hcxdumptool #181: NO WORK

romeueff, 2021-07-05:
I am not able to capture correctly, only receive EAPOL return will be that the nearby routers do not transmit the keys

07:33:48  10 8cf71074495c b0be76d1f37c G********3 [EAPOL:M3M4ZEROED EAPOLTIME:1270 RC:2 KDV:2]
07:35:41  10 6432a81c4440 b0be76d1f37c G********3 [EAPOL:M1M2 EAPOLTIME:2035 RC:1 KDV:2]
07:35:41  10 6432a81c4440 b0be76d1f37c G********3 [EAPOL:M2M3 EAPOLTIME:2458 RC:2 KDV:2]
07:36:22   6 b499baf97863 e01954e0bf46 G*******1 [EAPOL:M1M2 EAPOLTIME:19056 RC:1 KDV:2]
07:36:22   6 b499baf97863 e01954e0bf46 G*******1 [EAPOL:M2M3 EAPOLTIME:13381 RC:2 KDV:2]
07:37:19   6 b499baf97863 e01954e0bf46 G******* 1 [EAPOL:M1M2 EAPOLTIME:13512 RC:1 KDV:2]
07:37:19   6 b499baf97863 e01954e0bf46 G*******1 [EAPOL:M2M3 EAPOLTIME:60 RC:2 KDV:2]
07:37:33  10 6432a81c4440 b0be76d1f37c G********3 [EAPOL:M1M2 EAPOLTIME:1947 RC:1 KDV:2]
07:37:38  11 e85a8b27b8b6 b0be76d1f37c G********3 [EAPOL:M1M2ROGUE EAPOLTIME:3169 RC:61939 KDV:2]

-------------
└─# sudo hcxdumptool -i wlan0 --check_driver
initialization...
starting driver test...

driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

-------

└─# sudo hcxdumptool -i wlan0 --check_injection                                                                 1 ⨯
initialization...

starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14
packet injection is working on 2.4GHz!
injection ratio: 7% (BEACON: 90 PROBERESPONSE: 7)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 100% (NETWORK: 2 PROBERESPONSE: 2)
your antenna ratio is huge - say kids what time is it?

terminating...

ZerBea, 2021-07-05:
I'm not sure what you mean with "nearby routers do not transmit the keys".
If you mean PMKID:
The ACCESS POINTs in range don't transmit a PMKID. In that case you will not receive only M1 and M3 EAPOL frames from them.
In addition you will recevie M2 and M4 EAPOL frames from connected CLIENTs and M2 EAPOL frames from not connected CLIENTs.

That is not an hcxdumptool issue.

ZerBea, 2021-07-05:
I recommend to read som background information about 802.11i, 802.11r and OKC:
https://en.wikipedia.org/wiki/IEEE_802.11r-2008
https://community.arubanetworks.com/community-home/digestviewer/viewthread?MID=4386
https://documentation.meraki.com/MR/Other_Topics/PMKID_Vulnerability_FAQ_-_WPA%2F%2FWPA2-PSK_and_802.11r

romeueff, 2021-07-05:
that's right, that's what I meant, that the nearby routers didn't transmit PMKID
so most likely my settings are correct,
I will read the websites you indicated, thank you very much for your attention

ZerBea, 2021-07-05:
BTW:
Your injection ratio is fantastic and the response time (e.g.: EAPOLTIME:60) is pretty good.

There are many older routers. Usually they don't transmit PMKIDs.
On some newer routers, PMKID caching is disabled. So they also don't transmit a PMKID.

But I noticed that a CLIENT tried to connect to hcxdumptool:
07:37:38 11 e85a8b27b8b6 b0be76d1f37c G********3 [EAPOL:M1M2ROGUE EAPOLTIME:3169 RC:61939 KDV:2]

Maybe this CLIENT will transmit the PSK. In that case, hcxpcapngtool -E will extract the key.
An example is here:
https://github.com/evilsocket/pwnagotchi/issues/835#issuecomment-598597214

ZerBea, 2021-07-05:
Closed this issue report, because we are not able to force an AP to transmit a PMKID if this feature is not implemented in its firmware or if it is deactived by default / by user, but you can still ask your questions here
or here:
https://hashcat.net/forum/thread-6661.html

closed 2021-07-05 by ZerBea