ZerBea/hcxdumptool #182: ALFA AWUS036NHA  do not retrieve PMKID and keep factory MAC

algreen702021, 2021-07-24:
**Hi,
testing on ALFA AWUS036NHA (and TP-Link TL-WN722N), TP-link APA-25, Kali 2021, router Technicolor tg789vac v2. 
ISP does not advise customer about IDS or updates at all.
Macchanger display not spoofed MAC.**
This is the output from AWUS036NHA:

**(kali㉿kali)-[~/hcxdumptool/hcxtools]
└─$ sudo hcxdumptool -o PMKIDcaptured.pcapng -i wlan1 --enable_status 15  –disable_deauthentications –filterlist="listMAC.txt" -filtermode=2
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan1
INTERFACE HARDWARE MAC....: *******79753
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.10.0-kali9-amd64
DRIVER FIRMWARE VERSION...: 1.4
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 000e17d84594 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 000e17004595 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 000e17d84596 (incremented on every new client)
ROGUE (CLIENT)............: dc70144b5f17
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 63246
ANONCE....................: cef0cf4af7419347ade1d1302a2764c195caae1ec68f2e1d0d4035e676e38e0d
SNONCE....................: 95187bb93169cdc335e43587850500b233e494265fd9d75bfedef0de0e5aa524

00:26:32  11 ffffffffffff a491b18b154c Voda-****6508 [BEACON]
00:29:22  10 f47def6224e8 a491b16d263c Voda-****6508 [PROBERESPONSE]**

ZerBea, 2021-07-24:
First of all, your command line is faulty
` –disable_deauthentications –filterlist="listMAC.txt" -filtermode=2
`
and hcxdumptool all switches about filtering:
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused

About your reported PMKID issue:
Not every AP transmit a PMKID.
If the AP doesn't transmit a PMKID (and it looks like the Technicolor doesn't do this), hcxdumptool will not retrieve one.

About your reported MAC issue:
This are the MACs used by hcxdumptool for its attacks during runtime:
```
ROGUE (ACCESS POINT)......: 000e17d84594 (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 000e17004595 (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 000e17d84596 (incremented on every new client)
ROGUE (CLIENT)............: dc70144b5f17
```
In addition to that, hcxdumptool will show the origin MAC of the ADAPTER (regardless if you use a tool to spoof your MAC):
`INTERFACE HARDWARE MAC....: *******79753
`
but it will not use it for its attacks.

From --help:
`do not run hcxdumptool in combination with tools (channel hopper), that take access to the interface (except: tshark, wireshark, tcpdump)
`
macchanger is a tool that take access to the device and hcxdumptool will ignore all the changes of this tool, because it runs its own MAC space.

By this commit:
https://github.com/ZerBea/hcxdumptool/commit/982861952914a906839d2d02279e6a65b8b94973
I added a warning about macchanger:
`do not use tools like machcanger, because hcxdumptool run its own MAC space
`

Closed this report, because it isn't an issue:
reported issue 1: hcxdumptool doesn't receive a PMIKD -> AP doesn't transmit a PMKID
reported issue 2: hcxdumptool keep factory MAC -> hcxdumptool use its own MAC space
but you can still ask your questions here.

closed 2021-07-24 by ZerBea

ZerBea, 2021-07-24:
Regarding your command line, I assume that you are running an old version of hcxdumptool.
Please comment output of
`$ hcxdumptool -v`

I strongly recommend to update hcxdumptool to latest version:
```
$ hcxdumptool -v
hcxdumptool 6.2.0-17-g9828619 (C) 2021 ZeroBeat
```
because filter options are improved and changed:
```
--bpfc=<file>                      : input kernel space Berkeley Packet Filter (BPF) code
                                     affected: incoming and outgoing traffic - that include rca scan
                                     steps to create a BPF (it only has to be done once):
                                      set hcxdumptool monitormode
                                       $ hcxumptool -m <interface>
                                      create BPF to protect a MAC
                                       $ tcpdump -i <interface> not wlan addr1 11:22:33:44:55:66 and not wlan addr2 11:22:33:44:55:66 -ddd > protect.bpf
                                       recommended to protect own devices
                                      or create BPF to attack a MAC
                                       $ tcpdump -i <interface> wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 -ddd > attack.bpf
                                       it is strongly recommended to allow all PROBEREQUEST frames (wlan_type mgt && wlan_subtype probe-req)
                                       see man pcap-filter for a list of all filter options
                                      to use the BPF code
                                       $ hcxumptool -i <interface> --bpfc=attack.bpf ...
                                     notice: this is a protect/attack, a capture and a display filter
--filtermode=<digit>               : user space filter mode for filter list
                                     mandatory in combination with --filterlist_ap and/or --filterlist_client
                                     affected: only outgoing traffic
                                     notice: hcxdumptool act as passive dumper and it will capture the whole traffic on the channel
                                     0: ignore filter list (default)
                                     1: use filter list as protection list
                                        do not interact with ACCESS POINTs and CLIENTs from this list
                                     2: use filter list as target list
                                        only interact with ACCESS POINTs and CLIENTs from this list
                                        not recommended, because some useful frames could be filtered out
                                     using a filter list doesn't have an affect on rca scan
                                     only for testing usefull - devices to be protected should be added to BPF
                                     notice: this filter option will let hcxdumptool protect or attack a target - it neither a capture nor a display filter
--filterlist_ap=<file>             : ACCESS POINT MAC filter list
                                     format: 112233445566, 11:22:33:44:55:66, 11-22-33-44-55-66 # comment
                                     maximum entries 256
                                     run first --do_rcascan to retrieve information about the target
--filterlist_ap_vendor=<file>      : ACCESS POINT MAC filter list by VENDOR
                                     format: 112233, 11:22:33, 11-22-33 # comment
                                     maximum entries 256
                                     run first --do_rcascan to retrieve information about the target
--filterlist_client=<file>         : CLIENT MAC filter list
                                     format: 112233445566, 11:22:33:44:55:66, 11-22-33-44-55-66 # comment
                                     maximum entries 256
                                     due to MAC randomization of the CLIENT, it does not always work!
--filterlist_client_VENDOR=<file>  : CLIENT MAC filter list
                                     format: 112233, 11:22:33, 11-22-33 # comment
                                     maximum entries 256
                                     due to MAC randomization of the CLIENT, it does not always work!
```

ZerBea, 2021-07-24:
BTW:
Don't feel safe if you run a tool like macchanger.
As you can see, hcxdumptool is able to read the origin MAC of your adapter!

ZerBea, 2021-07-24:
BTW 2:
You should know that there are several (unfixed) bugs regarding the ath9k_htc driver:

[](https://bugzilla.kernel.org/buglist.cgi?bug_status=NEW&bug_status=ASSIGNED&bug_status=NEEDINFO&bug_status=REOPENED&field0-0-0=product&field0-0-1=component&field0-0-2=alias&field0-0-3=short_desc&field0-0-4=status_whiteboard&field0-0-5=content&order=changeddate%20DESC%2Cbug_status%2Cpriority%2Cassigned_to%2Cbug_id&query_format=advanced&type0-0-0=substring&type0-0-1=substring&type0-0-2=substring&type0-0-3=substring&type0-0-4=substring&type0-0-5=matches&value0-0-0=ath9k_htc&value0-0-1=ath9k_htc&value0-0-2=ath9k_htc&value0-0-3=ath9k_htc&value0-0-4=ath9k_htc&value0-0-5=%22ath9k_htc%22l)

ZerBea, 2021-07-24:
By the latest commit:
https://github.com/ZerBea/hcxdumptool/commit/9e987c9a1bae7499c97d0c6945e16ee1e62c0d31
I added some additional information to the status display:
```
$ sudo hcxdumptool -i wlp39s0f3u1u5 --enable_status=15
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u5
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE HARDWARE MAC....: 1ab3ce7beb4f (not used for the attack)
INTERFACE VIRTUAL MAC.....: 1ab3ceffffff (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.13.4-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 1cf4ca349439 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 1cf4ca34943a (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 1cf4ca34943b (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: c02250d473ac
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63402
ANONCE....................: 0993608d79007d689708add3ac9bef92e97c2c31a46b6f058b9ede038eccf277
SNONCE....................: 140306671bc43f63b717d1ca9cc0d478a5c90aa5b02253070db8728b9e7fdedc
```

Now the VIRTUAL INTERFACE MAC is shown, too
Additional I added some information, which MACs are in use and which MACs are not in use.

For this example I changed the origin MAC from 1ab3ce7beb4f to 1ab3ceffffff
but I strongly recommend to not do this, because it is useless.